CadQuery/cadquery #150: Installation: No module named 'OCC.Core'

isimonsm, 2019-06-14:
I set up a new conda environment and installed the package over the channels provided in the documentation:
`conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ`

When I tried to import the cadquery module, following error was thrown:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\IvoSimonsmeier\Anaconda3\envs\pre\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix
  File "C:\Users\IvoSimonsmeier\Anaconda3\envs\pre\lib\site-packages\cadquery\occ_impl\geom.py", line 4, in <module>
    from OCC.Core.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp, gp_XYZ
ModuleNotFoundError: No module named 'OCC.Core'`

I quickly tested the OCC package and it worked on its own. I tried to use other versions of the OCC package, but nothing worked for cadquery.

adam-urbanczyk, 2019-06-14:
Thanks for trying, looks like the docs are outdated. Try this
'''
conda install -c conda-forge -c cadquery cadquery
'''

adam-urbanczyk, 2019-06-15:
Actually it should be this:
```
conda install -c conda-forge -c cadquery cadquery=2
```

There is still a cadquery 1 package in conda-forge.

closed 2019-06-15 by adam-urbanczyk

isimonsm, 2019-06-16:
Thank you for your help, everything works fine now!

adam-urbanczyk, 2019-06-16:
Great, glad to hear that!

rdgraham, 2019-10-28:
I am trying to install cadquery 2 and am getting the same import error:

 `ModuleNotFoundError: No module named 'OCC.Core'`

I am trying to install the package directly, I really would rather not use anaconda on my system.

I am wondering what the underlying issue with the module naming is. It seems there is no `.Core` module within OCC (18.1), and if I go through and remove all the .Core from the imports then I can import cadquery without error and at least run basic operations (still haven't tested with cq-editor).

It would be helpful if the install instruction could also support installing the package directly from git. For example, with pip install --user.

PR #151 (2019-06-15): Update installation instructions
This will resolve #150