CadQuery/cadquery #152: Expose  outerWire and innerWires in the fluent api

adam-urbanczyk, 2019-06-15:
Currently `outerWire` and `innerWires` are only in the occ_impl.

CC: @yguel

marcus7070, 2021-05-08:
This seems like it would be quite useful when used with `Workplane.toPending` and then extruding.

Should it be another method added to Workplane? I think it could also be implemented as an optional keyword argument to `Workplane.wires`, which would be my preference.

thirtythreeforty, 2025-03-13:
I needed this and was able to get something working:

```
def outerWires(self):
    return (
        self
        .filter(lambda x: isinstance(x, cq.Face))
        .each(lambda face: face.outerWire(), combine=False)
    )
cq.Workplane.outerWires = outerWires
```