CadQuery/cadquery #154: extruding into a curved surface

mattvilim, 2019-06-19:
What is the recommended way to extrude into a curved surface without passing through? I see a note on future enhancements here: https://cadquery.readthedocs.io/en/latest/roadmap.html

> allow a cut or extrude to terminate at another surface, rather than either through all or a fixed distance

Is there a way to accomplish this using existing primitives?

adam-urbanczyk, 2019-06-20:
Thanks for trying cadquery @mattvilim . This is currently not implemented. If you are willing to get your hand dirty it requires exposing more 'Perfrom` methods from (https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_b_rep_feat___make_d_prism.html) in here: 

https://github.com/CadQuery/cadquery/blob/ef0808f4104e7c82da5df0e66355da0f3bbd24af/cadquery/occ_impl/shapes.py#L1476

Another (suboptimal) option would be to use csg approach:
1) (over)extrude the base solid
2) extrude the curved limiting face in the same direction
3) subtract (2) from (1)

mattvilim, 2019-06-20:
Thanks for the references; I will try implementing it and put in a pull request.

It doesn’t look like the OpenCascade API does any validity checking. If the user specifies something non-sensical (e.g. extrude from a face into a face that lies parallel to the path of extrusion), is it ok to crash or do you expect CadQuery to use other geometry primitives to figure out if the operation is invalid and warn the user?

adam-urbanczyk, 2019-06-20:
I think it is OK to crash. Looking forward to your PR!