CadQuery/cadquery #157: Provide docker image

adam-urbanczyk, 2019-06-25:
It would be nice to have a docker image for users who cannot use conda.

Relevant resources
* https://github.com/ContinuumIO/docker-images
* https://github.com/jozo/docker-pyqt5

NB: this is a continuation of #153

dcowden, 2019-06-25:
I can own this one

adam-urbanczyk, 2019-06-25:
Great!

Peque, 2019-09-22:
Maybe not what you were looking for (I guess you were not talking about a Jupyter-enabled Docker image), but I managed to create this:

https://github.com/Peque/cadquery-binder

If you are interested, I can transfer the repo to CadQuery organization. I think it makes more sense there.

jmwright, 2020-05-06:
Referencing pull request #176

jmwright, 2021-02-05:
Referencing the Dockerfile created by @RubenRubens [here](https://github.com/RubenRubens/cq-testing/blob/master/Dockerfile)

RubenRubens, 2021-04-22:
I have just created a repo that may help with this.
https://github.com/RubenRubens/cq-containers

jmwright, 2021-04-23:
@RubenRubens Very cool, thanks for putting this together. Two questions:

1. Will these be available on Dockerhub (or similar)?
2. What's the difference between your jypyter-cadquery image and the one from the jypyter-cadquery project?

RubenRubens, 2021-04-23:
**1.** Dockerhub is a container registry. Even if it is the most popular it's not the only one, there are other public container registries such as:

+ [Amazon ECR](https://gallery.ecr.aws/)
+ Dockerhub
+ Github Container Registry (ghcr, for short)
+ And many more...

The images are publicly available on ghcr, can be found on the repo as a package. A pipeline is missing to update these images automatically (it's work in progress).  In the near future I have no plans to move to Dockerhub.

**2.** cq-jupyter is just a link to @bernhard-42 repo.

Any suggestion, PR or criticism is welcome.

jmwright, 2021-04-23:
> 2. cq-jupyter is just a link to the @bernhard-42 repo.

Sorry, copied the base URL without the header ID. Here's [the link](https://github.com/bernhard-42/jupyter-cadquery#b-using-a-docker-image) I meant to post.

> The images are publicly available on ghcr

That's great, I'll read up on ghcr.

I spent some time integrating cq-cli into a CadHub Docker image so they can support CadQuery scripts. You've got a cq-cli image already, but just in case you're interested: https://github.com/Irev-Dev/cadhub/blob/0485f34ee1621860c92feb58a05a3576e3d1c67f/api/src/docker/cadquery/Dockerfile

jmwright, 2021-04-29:
@RubenRubens Would you be willing to add a description with some usage instructions to your cq-containers repo? I'd like to add a Docker section to the installation section of the readme that points users to your repo. I expect I'll just mirror or paraphrase the description you have in your repo when providing a link.

RubenRubens, 2021-05-02:
The repo follows this structure:

```
.
├── cq-cli
│   ├── Dockerfile
│   └── readme.md 
├── cq-conda
│   ├── Dockerfile
│   └── readme.md 
└── readme.md 
```
So, the instructions are in `./cq-cli/readme.md` and `./cq-conda/readme.md`. The file `./readme.md` is simply a table that contains information about cq-cli,  cq-conda and cq-jupyter. I hope it is clear enough.

closed 2021-05-08 by marcus7070