CadQuery/cadquery #158: Implement rounded slot feature

adam-urbanczyk, 2019-06-25:
[image]

bweissinger, 2019-08-26:
What kind of slot commands should we implement? Commonly used slots seem to be:

- 	Center to Center
- 	Center Point
- 	Three Point Arc Slot
- 	Center Point Arc Slot

A 'Slot Along Spline' type may be useful as well.

Also, should these features be implemented as 2d sketch objects or as 3d operations? i.e. should they behave like circle() or hole()?

adam-urbanczyk, 2019-08-26:
I'd propose to have `center`, `length` and `dir` as inputs to follow current conventions. My preference would go to a 2D sketch.

bweissinger, 2019-08-29:
Agreed on both counts. I think it would be an excellent addition. 

I think the ability to use existing sketch geometry to slot along would cover any advanced uses of slots, and be a good future addition. I imagine that would be closely related to offsetting sketch geometry, which would also be a good feature to have.

dcowden, 2019-08-29:
@adam-urbanczyk @bsilvereagle I know this is off topic somewhat, but I've been desperate for many years ( since about the first month of CQ development actually ) to integrate parametric sketches.  I was stuck for a long time because to do that you need a constraint solver.    CQ does a good job considering design intent when going from 2d to 3d, but when constructing sketches you can't convey intent as easily as you can when you draw in a CAD tool.  IE: I want a slot, and i want it _centered_ with respect to these two sides.

adam-urbanczyk, 2019-09-17:
Solved by #186

closed 2019-09-17 by adam-urbanczyk