CadQuery/cadquery #160: makeWedge broken

rkapl, 2019-07-08:
The makeWedge prototype does not match the actual OCC function.
```
import cadquery as cq

# Bogus values
cq.Solid.makeWedge(
   0, 0, 0, 0, 0,
   0, 0, 0, 0, 0)
```

```
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    0, 0, 0, 0, 0)
  File "C:\Users\Roman\.conda\envs\cadquery\lib\site-packages\cadquery\occ_impl\shapes.py", line 1224, in makeWedge
    x2max).Solid())
  File "C:\Users\Roman\.conda\envs\cadquery\lib\site-packages\OCC\Core\BRepPrimAPI.py", line 629, in __init__
    _BRepPrimAPI.BRepPrimAPI_MakeWedge_swiginit(self, _BRepPrimAPI.new_BRepPrimAPI_MakeWedge(*args))
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_BRepPrimAPI_MakeWedge'.
  Possible C/C++ prototypes are:
    BRepPrimAPI_MakeWedge::BRepPrimAPI_MakeWedge(Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const)
    BRepPrimAPI_MakeWedge::BRepPrimAPI_MakeWedge(gp_Ax2 const &,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const)
    BRepPrimAPI_MakeWedge::BRepPrimAPI_MakeWedge(Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const)
    BRepPrimAPI_MakeWedge::BRepPrimAPI_MakeWedge(gp_Ax2 const &,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const)
```

adam-urbanczyk, 2019-07-08:
Confirmed, thanks for reporting.

jmwright, 2019-12-01:
@rkapl Once the associated PR is finalized and merged, this will be fixed. I've also added the higher-level wedge primitive in CQ so that it's possible to do things like this:
```python
import cadquery as cq
s = (cq.Workplane("XY").rect(20.0, 20.0, forConstruction=True).vertices()
                       .wedge(10, 10, 10, 5, 5, 5, 5, centered=(False, False, False), dir=(0, 1, 0)))
show_object(s)
```

jmwright, 2019-12-05:
Fixed in #228

closed 2019-12-05 by jmwright