CadQuery/cadquery #161: loft with a single wire crashes

rkapl, 2019-07-08:
```
import cadquery as cq
cq.Workplane('XY').rect(10, 10).loft()
```

Crashes (segfault) in the OCC layer.

```
 # Child-SP          RetAddr           Call Site
00 0000001c`33fe79d0 00007ffd`4288d81e TKernel!Storage_StreamUnknownTypeError::DynamicType+0x65867
01 0000001c`33fe7a00 00007ffd`38e8d696 TKGeomAlgo!Law_S_Type_+0x2d1630
02 0000001c`33fe7a70 00007ffd`38e88ea9 TKOffset!BRepOffset_DataMapOfShapeMapOfShape::`default constructor closure'+0x2932e
03 0000001c`33fe7da0 00007ffd`38e873a0 TKOffset!BRepOffset_DataMapOfShapeMapOfShape::`default constructor closure'+0x24b41
04 0000001c`33fe8630 00007ffd`42d964ed TKOffset!BRepOffset_DataMapOfShapeMapOfShape::`default constructor closure'+0x23038
05 0000001c`33fe8760 00000000`6907d4f9 _BRepBuilderAPI!gp_Ax3::XReverse+0x2439a
```

adam-urbanczyk, 2019-07-08:
Segfault is obviously never nice, but out of curiosity what result were you trying to get by lofting one wire?

rkapl, 2019-07-08:
Nothing, I've forgot to add the second wire. However, it is a PITA to find what is crashing with that segfault.

adam-urbanczyk, 2019-07-09:
Agreed, we need to add some input validation.

Hatatister, 2019-08-14:
I think this issue can be closed now

closed 2019-08-14 by jmwright