CadQuery/cadquery #166: 3D operation to make point, line etc...

dimonbavly, 2019-07-11:
While there is a construction of objects in 2D - there are no constructions of the same objects in 3D?.

jmwright, 2019-07-12:
CadQuery typically expects geometry creation to be on a workplane, which kind of ties it to a 2D representation. It should be possible to create objects in 3D, but I think you'd have to access the levels below the top-level fluent API to make it happen.

adam-urbanczyk, 2019-07-12:
@dimonbavly in the fluent API you need to create a workplane to create most of the objects, but they are 3D objects in the end. If your needs are not covered by this you could always use the `occ_impl` layer. There you can specify e.g. a spline or a poly-line with a list of arbitrary 3D points. It would help if you shared your use case in more detail, maybe then w could offer more detailed advice.