CadQuery/cadquery #168: Error while installing CadQuery with Conda

roipoussiere, 2019-08-07:
I try to install CadQuery with Conda. I'm not used to Conda so it maybe a trivial issue.

As explained in [the documentation](https://cadquery.readthedocs.io/en/latest/installation.html), I did the following:

```bash
conda env create --name cadquery2
conda activate cadquery2
conda install -c conda-forge -c cadquery cadquery=2
```

But I get this error: `ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /home/cadquery`.

<details><summary>full stacktrace</summary>

```
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done

Downloading and Extracting Packages
freetype-2.10.0      | 885 KB    | ########## | 100% 
libpng-1.6.37        | 343 KB    | ########## | 100% 
setuptools-41.0.1    | 612 KB    | ########## | 100% 
openssl-1.1.1c       | 2.1 MB    | ########## | 100% 
readline-8.0         | 441 KB    | ########## | 100% 
oce-0.18.2           | 192.8 MB  | ########## | 100% 
six-1.12.0           | 22 KB     | ########## | 100% 
tbb-2019.7           | 1.4 MB    | ########## | 100% 
sqlite-3.29.0        | 2.0 MB    | ########## | 100% 
xz-5.2.4             | 366 KB    | ########## | 100% 
pythonocc-core-0.18. | 46.4 MB   | ########## | 100% 
pip-19.2.1           | 1.9 MB    | ########## | 100% 
python-3.6.7         | 34.6 MB   | ########## | 100% 
zlib-1.2.11          | 105 KB    | ########## | 100% 
pyparsing-2.4.2      | 57 KB     | ########## | 100% 
ca-certificates-2019 | 145 KB    | ########## | 100% 
freeimageplus-3.17.0 | 5.4 MB    | ########## | 100% 
tk-8.6.9             | 3.2 MB    | ########## | 100% 
wheel-0.33.4         | 34 KB     | ########## | 100% 
ncurses-6.1          | 1.3 MB    | ########## | 100% 
certifi-2019.6.16    | 149 KB    | ########## | 100% 
libffi-3.2.1         | 46 KB     | ########## | 100% 
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
Ran pip subprocess with arguments:
[u'/home/cadquery/.conda/envs/cadquery2/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/cadquery/tmpDqddbx.requirements.txt']
Pip subprocess output:

Pip subprocess error:
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /home/cadquery

CondaEnvException: Pip failed
```
</details>

I then tried to copy [the setup.py from this repository](https://github.com/CadQuery/cadquery/blob/master/setup.py) on the current folder, but I get an other error: `OSError: [Errno 17] File exists: '/home/cadquery/.conda/pkgs/cache'`.

<details><summary>full stacktrace</summary>

```
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
Collecting package metadata (repodata.json): ...working... failed

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/opt/conda/lib/python2.7/site-packages/conda/exceptions.py", line 1062, in __call__
        return func(*args, **kwargs)
      File "/opt/conda/lib/python2.7/site-packages/conda_env/cli/main.py", line 73, in do_call
        exit_code = getattr(module, func_name)(args, parser)
      File "/opt/conda/lib/python2.7/site-packages/conda_env/cli/main_create.py", line 101, in execute
        installer.install(prefix, pkg_specs, args, env)
      File "/opt/conda/lib/python2.7/site-packages/conda_env/installers/conda.py", line 27, in install
        unlink_link_transaction = solver.solve_for_transaction(prune=getattr(args, 'prune', False))
      File "/opt/conda/lib/python2.7/site-packages/conda/core/solve.py", line 112, in solve_for_transaction
        force_remove, force_reinstall)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/solve.py", line 150, in solve_for_diff
        force_remove)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/solve.py", line 249, in solve_final_state
        ssc = self._collect_all_metadata(ssc)
      File "/opt/conda/lib/python2.7/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/solve.py", line 389, in _collect_all_metadata
        index, r = self._prepare(prepared_specs)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/solve.py", line 974, in _prepare
        self.subdirs, prepared_specs, self._repodata_fn)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/index.py", line 214, in get_reduced_index
        repodata_fn=repodata_fn)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 91, in query_all
        result = tuple(concat(executor.map(subdir_query, channel_urls)))
      File "/opt/conda/lib/python2.7/site-packages/concurrent/futures/_base.py", line 641, in result_iterator
        yield fs.pop().result()
      File "/opt/conda/lib/python2.7/site-packages/concurrent/futures/_base.py", line 455, in result
        return self.__get_result()
      File "/opt/conda/lib/python2.7/site-packages/concurrent/futures/thread.py", line 63, in run
        result = self.fn(*self.args, **self.kwargs)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 87, in <lambda>
        package_ref_or_match_spec))
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 96, in query
        self.load()
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 160, in load
        _internal_state = self._load()
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 188, in _load
        self.cache_path_json)
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 153, in cache_path_json
        return self.cache_path_base + '.json'
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 144, in cache_path_base
        create_cache_dir(),
      File "/opt/conda/lib/python2.7/site-packages/conda/core/subdir_data.py", line 646, in create_cache_dir
        mkdir_p_sudo_safe(cache_dir)
      File "/opt/conda/lib/python2.7/site-packages/conda/gateways/disk/__init__.py", line 84, in mkdir_p_sudo_safe
        os.mkdir(path)
    OSError: [Errno 17] File exists: '/home/cadquery/.conda/pkgs/cache'

`$ /opt/conda/bin/conda-env create --name cadquery2`

  environment variables:
                 CIO_TEST=<not set>
  CONDA_AUTO_UPDATE_CONDA=false
               CONDA_ROOT=/opt/conda
          LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:/usr/local/nvidia/li
                          b:/usr/local/nvidia/lib64:/usr/local/nvidia/lib:/usr/local/nvidia/lib6
                          4
                     PATH=/usr/local/nvidia/bin:/usr/local/cuda/bin:/usr/local/sbin:/usr/local/b
                          in:/usr/sbin:/usr/bin:/sbin:/bin:/opt/conda/bin/
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>

     active environment : None
       user config file : /home/cadquery/.condarc
 populated config files : 
          conda version : 4.7.10
    conda-build version : not installed
         python version : 2.7.16.final.0
       virtual packages : 
       base environment : /opt/conda  (read only)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /opt/conda/pkgs
                          /home/cadquery/.conda/pkgs
       envs directories : /home/cadquery/.conda/envs
                          /opt/conda/envs
               platform : linux-64
             user-agent : conda/4.7.10 requests/2.22.0 CPython/2.7.16 Linux/4.19.62-1-MANJARO ubuntu/18.04.3 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.
```
</details>

adam-urbanczyk, 2019-08-18:
I cannot reproduce this (though `conda env create --name cadquery2` should be changed to `conda create --name cadquery2`). Could you provide more details, e.g. the output of `conda info -a` ?

jmwright, 2021-02-05:
No answer from OP. Closing.

closed 2021-02-05 by jmwright