CadQuery/cadquery #169: CadQuery2 Dockerfile

roipoussiere, 2019-08-07:
I made a Dockerfile for CadQuery2:

- [docker-hub](https://hub.docker.com/r/roipoussiere/cadquery2/dockerfile)
- [source ](https://github.com/roipoussiere/cadquery/blob/master/docker/Dockerfile)

It is based on [nvidia's cudagl image](https://hub.docker.com/r/nvidia/cudagl) in order to use OpenGL in Docker.

For now, CadQuery2 is installed from sources and not with Conda, because I experienced [some issues](https://github.com/CadQuery/cadquery/issues/168) with this last.

I'm interested to switch to a Conda-based installation, because the compilation process is very long and reaches [Docker hub limitations](https://forums.docker.com/t/automated-build-resource-restrictions/1413/2), so I need to build the image locally then upload it to docker hub directly. The drawback is the dockerfile page on docker hub does not have many information (no Dockerfile, no description, etc.).

Usage:

```
echo "import cadquery; print(cadquery.Workplane('XY').box(1,2,3).toSvg())" > cadquery_script.py
cat cadquery_script.py | docker run --rm -i roipoussiere/cadquery2 > 3d_model.svg
```

You can read more on the [dedicated readme](https://github.com/roipoussiere/cadquery/blob/master/docker/readme.md).

I'm writing this issue before submitting a pull request in order to get some feedback about this.

In particular, I note that there is no CLI script in CadQuery2, like in [CadQuery vanilla](https://github.com/dcowden/cadquery/blob/master/cq_cmd.py). Should I create it, based on this one, or is there a specific reason to not include it?

There are many improvements to do, put it works enough for a proof of concept.

Things I would like to do before an eventual merge:
- get your feedback;
- Conda-based installation;
- make the container [CQGI-compliant](https://cadquery.readthedocs.io/en/latest/cqgi.html);
- allow input and output volumes to be mounted.

jmwright, 2019-08-08:
@roipoussiere Thanks for all your work on this. I'm not aware of any reason not to include the CLI. I expect we just forgot it in the transition.

roipoussiere, 2019-08-08:
> I'm not aware of any reason not to include the CLI. I expect we just forgot it in the transition.

Ok, I will include it!

Since there is now a Github organization for all CadQuery-related projects, it makes sense for me to do the same thing on docker hub.

`cadquery` name was available, so I just created this [CadQuery organization on dockerhub](https://hub.docker.com/u/cadquery/), and added my [CadQuery2 Dockerfile](https://hub.docker.com/r/cadquery/cadquery2) (with a full description! I just managed how to do it).

I can also add the [CadQuery1 dockerfile](https://hub.docker.com/r/dcowden/cadquery) as well.

If you give me a docker-hub user name, I can [add it to this organization](https://success.docker.com/article/how-do-i-add-a-member-to-the-owners-team-of-an-organization-on-the-docker-hub) and [transfer ownership](https://success.docker.com/article/how-do-i-transfer-ownership-of-an-organization-account). This will allow you to connect the docker org. to the github org., which is useful for autobuilds for example.

Is this ok for you?

jmwright, 2019-08-12:
@roipoussiere Thanks!

@dcowden You've probably got a Docker-Hub account from the work you've done with it before, right? Can @roipoussiere add you to the org and transfer ownership to you?

dcowden, 2019-08-12:
@roipoussiere thanks for the work!

My docker hub account is dcowden, and I'm happy to receive ownership for future work. Though I would say you should retain ownership also since you have done so much work on it! 

Thanks again for the contribution!

roipoussiere, 2019-08-12:
In fact there is not really an *transfer ownership* thing, because docker hub works with *teams*, and the repository is already owned by the *owners* team. The steps suggested [here](https://success.docker.com/article/how-do-i-transfer-ownership-of-an-organization-account) to transfer ownership is to add a new member to this team and remove myself to it, but I don't want to do it because I would like to continue to work on this. ;)

That being said, @dcowden, you are now a member of the docker hub cadquery's owners team, congratulations! Is there a adoubement ceremony to do or something similar?

jmwright, 2019-08-12:
@roipoussiere What if you you remove yourself to transfer ownership and then @dcowden adds you right back? Will that work?

roipoussiere, 2019-08-12:
> What if you you remove yourself to transfer ownership and then @dcowden adds you right back? Will that work?

It will be exactly the same as the initial situation. Since we are both in the *owner* team, we both have the ownership of the organization, with the exact same privileges.

If you think it's necessary, I can however create an other team on the organization with less privileges (read / write instead of admin) and move into it, but I will not be able to do some some actions useful for me at this point, like edit project description, configure builds, etc.

Here is a recap of the repository permissions:

- **read only**: Pull and view repository details and builds;
- **read & write**: Pull, push, and view a repository; view, cancel, retry or trigger builds;
- **admin**: Pull, push, view, edit, and delete a repository; edit build settings; update the repository description.

jmwright, 2019-08-13:
@roipoussiere It sounds like you already have what you need, I was just misunderstanding what was going on and was over-complicating things. Please keep the ownership privileges that you have now.

Thanks for setting the org up.

roipoussiere, 2019-08-13:
Closing this, the discussion can continue on [the pull request](https://github.com/CadQuery/cadquery/pull/176).

closed 2019-08-13 by roipoussiere