CadQuery/cadquery #170: pushPoints and sweep not working together

bragostin, 2019-08-08:
I am trying to pattern a solid on a plane. With cylinders the code below works perfectly and I get 5 cylinders:
```python
yz = [ (0, 0),(10, 0),(0, 10),(-10, 0),(0, -10) ]
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane("YZ")
pl = pl.pushPoints(yz)
pl =  pl.circle(2.0).extrude(5)
show_object(pl)
```
However, when I want to achieve the same with drumbells columns created with sweep like below, I get the error message: *AttributeError: 'Vector' object has no attribute 'geomType'*
```python
yz = [ (0, 0),(10, 0),(0, 10),(-10, 0),(0, -10) ]
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane("YZ")
pl = pl.pushPoints(yz)
pl = pl.workplane(offset=-2).circle(5).workplane(offset=2).circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True)
show_object(pl)
```
After 4 hours through the documentation I still have no idea what is wrong here.

jmwright, 2019-08-08:
@bragostin I think you want to use pushPoints on each new workplane you create. The first pushPoints call doesn't apply to all workplanes globally. I've experimented with your code a little bit, and have gotten close, but keep getting segfaults in CQ-editor. If I remove the combines in the code, I get errors that the faces are not planar. I'm not sure if I'm being naive about how combine works in this context, or if the kernel is just having trouble with the operation and resulting shape. I'm out of time for this morning, but I'll try to circle back to this later. Maybe another community member will have some thoughts as well.
```
import cadquery as cq
yz = [ (0, 0),(10, 0),(0, 10),(-10, 0),(0, -10) ]
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane("YZ")
#pl = pl.pushPoints(yz)
pl = pl.workplane(offset=-2).pushPoints(yz).circle(5).combine().workplane(offset=2).pushPoints(yz).circle(1).combine().sweep(path, multisection=True)#.workplane(offset=2).circle(5).sweep(path, multisection=True)
show_object(pl)
```

bragostin, 2019-08-08:
I have found this workaround below with nested loops. It works, but I don't think this is the best way to use CadQuery.
```python
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
pl0 = cq.Workplane("YZ")
pl0 = pl0.workplane(offset=-2).circle(5).workplane(offset=2).circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True)
y = [-10. ,0., 10.]
z = [-10. ,0., 10.]
pl = cq.Workplane("YZ")
for yi in y:
   for zi in z:
        pl = pl.union(pl0.translate((0.,yi,zi)))
show_object(pl)
```

bragostin, 2019-08-09:
Although the nested loop solution works it is very slow. I am trying with the function
```
Workplane.eachpoint(callbackFunction[, …])
```
like below, in the hope that pushPoints would be applied to the function *pillar()* as a whole:
```python
yz = [ (0, 0),(10, 0),(0, 10),(-10, 0),(0, -10) ]
pl = cq.Workplane("YZ").pushPoints(yz)
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
def pillar(pos):
    return pl.workplane(offset=-2).circle(5).workplane(offset=2).circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True)
pl = pl.each(pillar)
```
But I get the error message *Attribute error: 'Vector' object has attribute 'geomType'.*

adam-urbanczyk, 2019-08-09:
@bragostin the code below works for me. Note the differences in the `pillar` implementation:

1. It has to return a `occ_impl.Shape` or subclass (hence `val()` is called)
2. You were referencing the original `pl` Workplane

```python
path = cq.Workplane("XZ").moveTo(-2, 0).lineTo(2, 0)
def pillar(pos):
    return cq.Workplane("YZ",origin=pos).workplane(offset=-2).circle(5).workplane(offset=2)\
    .circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True).val()

yz = [ (0, 0),(10, 0),(0, 10),(-10, 0),(0, -10) ]
pl = cq.Workplane("YZ").pushPoints(yz).each(pillar)
```

bragostin, 2019-08-09:
I just tried, this solution works perfectly and is very fast even with >100 pillars! I started with CadQuery last Wednesday only so I still have a lot to learn obviously :) Until now I was using Openscad, which has been very useful to me, but I must say that CadQuery is way more efficient and elegant. Thank you!

closed 2019-08-09 by bragostin

adam-urbanczyk, 2019-08-09:
Good to hear, if speed is ever an issue you could try constructing the object once and just returning transformed copies.