CadQuery/cadquery #172: cutBlind makes whole extrusion disappear

bragostin, 2019-08-12:
I stumbled upon this strange behaviour. I first create a shell, then add an extrusion layer inside, then make a cut-out on the other side of the shell. But the cut-out does not appear (1). If I replace cutBlind() by extrude(), then I get my cut-out at the right location, but the first extrusion layer disappears completely (2). I still can't figure out what is wrong here.
(1)
```python
#SHELL
shl = cq.Workplane('XY').box(3, 100,150).faces('+Z').shell(1) 
# ADDITIONAL LAYER
shl  = shl.faces('<X').workplane(offset=-2).rect(99, 149).extrude(1)
# CUT OUT --> DOES NOT APPEAR
shl = shl.faces(">X").workplane(offset=0).center(-1,-74.5).moveTo(0,74).lineTo(0, 148).radiusArc((2, 148), 1).lineTo(2,1).radiusArc((0,1), 1).close().cutBlind(1)
# SHOW
show_object(shl)
```
(2)
```python
#SHELL
shl = cq.Workplane('XY').box(3, 100,150).faces('+Z').shell(1) 
# ADDITIONAL LAYER
shl  = shl.faces('<X').workplane(offset=-2).rect(99, 149).extrude(1)
# CUT OUT --> APPEARS BUT ADDITIONAL LAYER DISAPPEARS
shl = shl.faces(">X").workplane(offset=0).center(-1,-74.5).moveTo(0,74).lineTo(0, 148).radiusArc((2, 148), 1).lineTo(2,1).radiusArc((0,1), 1).close().extrude(1)
# SHOW
show_object(shl)
```

jmwright, 2019-08-12:
@bragostin I think the normal of the workplane you were trying to cut into was just flipped. Notice the `invert=True` I added in the following code. I think it gives the result you were looking for.
```python
import cadquery as cq
#SHELL
shl = cq.Workplane('XY').box(3, 100,150).faces('+Z').shell(1) 
# ADDITIONAL LAYER
shl  = shl.faces('<X').workplane(offset=-2).rect(99, 149).extrude(1)
# CUT OUT --> DOES NOT APPEAR
shl = shl.faces(">X").workplane(offset=0, invert=True).center(-1,-74.5).moveTo(0,74).lineTo(0, 148).radiusArc((2, 148), 1).lineTo(2,1).radiusArc((0,1), 1).close().cutBlind(1)
# SHOW
show_object(shl)
```

bragostin, 2019-08-12:
@jmwright I just tried it, but this is what I get:
[image]
Instead of a cut-out, I get a bump *outwards* the shell. If I replace `cutBlind(1)` with `cutBlind(-1)` I should get a cut-out *inwards* the shell, but nothing appears, even after export to a STEP file.

jmwright, 2019-08-12:
@bragostin I get a message when I click on the image link saying it contains errors.

How are you using CadQuery? Are you using CQ-editor? What version(s)? The code above works fine for me in release candidate 1 of CQ-editor, assuming that I've correctly understood what you're trying to achieve.

jmwright, 2019-08-12:
@bragostin Here are two screenshots of my result. Does it look correct?

[image]

[image]

bragostin, 2019-08-12:
@jmwright 
I compiled the sources of oce-master and pythonocc-core-master, then I downloaded CadQuery from https://github.com/CadQuery/cadquery and CQ-editor (candidate 1) from https://github.com/CadQuery/CQ-editor and installed them with python setup.py install. From your screenshot it is difficult to see if the cutBlind has created a hole in the shell wall (what I want to achieve). 
[image]

jmwright, 2019-08-12:
Ok, changing `cutBlind` to `cutThruAll` gives the wrong result as well. The extrusion along the X axis line in the screenshot is the cutThruAll operation.

[image]

@adam-urbanczyk Any ideas on why this might be happening?

bragostin, 2019-08-12:
Something funny, the code below works (using a separate cut() instead of a cutBlind() on the shell surface), but only if the additional layer inside the shell does NOT touch the shell (unfortunately I need to have it touching the internal wall of the shell). As soon this layer is contacting the shell (I tried to union() it separately too, to no avail), then the cut() yields an extruded solid instead of a cut-out.
```python
#SHELL
shl = cq.Workplane('XY').box(3, 100,150).faces('+Z').shell(1) 
# ADDITIONAL LAYER
shl  = shl.faces('<X').workplane(offset=-2, origin=(0,0,0)).rect(99, 149).extrude(0.75).combine()
# CUT OUT
cutout = cq.Workplane('YZ').workplane(offset=0.5, origin=(0,0,0)).center(-1,-74.5).moveTo(0,74).lineTo(0, 148).radiusArc((2, 148), 1).lineTo(2,1).radiusArc((0,1), 1).close().extrude(3)
shl = shl.cut(cutout)
# SHOW
show_object(shl)
```

Hatatister, 2019-08-12:
It seems as if something is wrong with the face orientation of the shell (somehow as if the faces are orientated to the inner of the solid and the "inner" of the solid is the "outer"). 
If you construct the shell like this:
`shl = cq.Workplane('XY').box(5, 100,150).faces('+Z').shell(-1)`

it will work. The only thing you have to do is flip the orientation of the second workplane:
`shl = shl.faces(">X").workplane(offset=0, invert=True).

Without flipping the face normal points away from the solid, which is ok for extrusion but will lead to a cut through air for cutBlind.

If you need the fillets you could add them manually.

bragostin, 2019-08-12:
@Hatatister I am happy to report that your solution works, thank you! But is this an expected behavior of the shell() function?

bragostin, 2019-08-13:
@Hatatister regarding the fillet ()
```python
shl = cq.Workplane('XY').box(5, 100,150).faces('+Z').shell(-1).edges().fillet(0.5)
```
actually does not work (BRep_API) error. How to fillet ALL edges outside AND inside?

Hatatister, 2019-08-13:
@bragostin 
Regarding the fillet. OpenCascade will fail to generate a fillet, ff the width(radius) of the fillet is equal or bigger than the width of the surfaces which are attached to the edge, which shall become a fillet. In your case, if you apply a fillet to an outer edge, than the remaing surface has a width of 0.5 which is to much for the inner edge fillet. If you change `fillet(0.5)` to `fillet(0.499)` it will work.

Hatatister, 2019-08-13:
Regarding the shell:
I think this is an OpenCascade issue since CQ delegates to pythonOCC. Anyhow, the generated shell looks optically ok regardless if you use -1 or 1, but for some reason the shell constructed with param `1` shows this strange behavior after fusion/cutting with another solid.

bragostin, 2019-08-13:
@Hatatister in this case maybe the solution below is better? It allows larger radius fillet, and by offsetting the cut out box I create my additional layer in the same operation.
```python
#SHELL
shl = cq.Workplane('XY').box(5, 100, 150).edges().fillet(1).cut(cq.Workplane("XY" ).center(0.5,0).box(3, 99, 149)).edges().fillet(1)
# CUT OUT
cutout = cq.Workplane('YZ').workplane(offset=0.5, origin=(0,0,0)).center(-1,-74.5).moveTo(0,74).lineTo(0, 148).radiusArc((2, 148), 1).lineTo(2,1).radiusArc((0,1), 1).close().extrude(3)
shl = shl.cut(cutout)
show_object(shl)
```

Hatatister, 2019-08-13:
I'm not sure which one is better. There are many ways to create a part

bragostin, 2019-08-13:
I started using CadQuery one week ago only, so I am still trying to figure out what are the "best" practices with CadQuery. Thank you for your help!

closed 2019-08-13 by bragostin