CadQuery/cadquery #175: select facse after a .revolve()

roipoussiere, 2019-08-12:
I designed this wonderful pulley (my first design with CadQuery!):

[image]

source:

```python
pulley_outer_diameter, pulley_thickness = 50, 10
pulley_shell_thickness, pulley_inner_diameter = 2, 10
shaft_diameter, shaft_diameter = 5, 4.5
pulley_split_thickness = pulley_thickness - pulley_shell_thickness*2

s = cq.Workplane('front').line(0, pulley_inner_diameter/2)\
    .line(-pulley_split_thickness/2, 0)\
    .lineTo(-pulley_split_thickness/2, pulley_outer_diameter/2 - pulley_shell_thickness)\
    .line(-pulley_shell_thickness, pulley_shell_thickness)\
    .line(0, -pulley_outer_diameter/2)\
    .close().mirrorY()\
    .revolve(360, [pulley_outer_diameter/2, -0.01])
```

Now I would like to do some operations on the pulley, like adding a hole, fillets, etc.

I tried adding `.faces("|X").workplane().hole(5)`, among many other solutions that doesn't work.

How to select, let's say, the discs, or any other face?

jmwright, 2019-08-13:
You might want to look at the combining selectors section of the docs.

https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors

You could also select the top face with >Z and then bottom face with <Z in separate operations.

roipoussiere, 2019-08-13:
> You could also select the top face with >Z and then bottom face with <Z in separate operations.

I suppose that you mean `>X` and `<X` (look at the axes)?

I already tried combining selectors, but I was not able to continue to add operations on the pulley. I don't know if it's my lack of knowledge at CadQuery scripting or an actual bug.

For instance, following your suggestion, with `.faces('>X and <X').workplane().hole(5)`, I get `list index out of range` error, and if I do `.faces('<X').faces('>X').workplane().hole(5)`, I get `Standard_ContructionError`.

jmwright, 2019-08-13:
> I suppose that you mean >X and <X (look at the axes)?

Yeah, I'm used to constructing things using the XY workplane and didn't pay attention.

```
.faces('<X').faces('>X').workplane().hole(5)
```

I don't think you can stack the `faces` calls like that, because you'd be selecting a face of a face. I have to invert the workplane normal and skip the hole operation to get things to work right, but this code works for me. Is it the result you were looking for?
```
pulley_outer_diameter, pulley_thickness = 50, 10
pulley_shell_thickness, pulley_inner_diameter = 2, 10
shaft_diameter, shaft_diameter = 5, 4.5
pulley_split_thickness = pulley_thickness - pulley_shell_thickness*2

s = cq.Workplane('front').line(0, pulley_inner_diameter/2)\
    .line(-pulley_split_thickness/2, 0)\
    .lineTo(-pulley_split_thickness/2, pulley_outer_diameter/2 - pulley_shell_thickness)\
    .line(-pulley_shell_thickness, pulley_shell_thickness)\
    .line(0, -pulley_outer_diameter/2)\
    .close().mirrorY()\
    .revolve(360, [pulley_outer_diameter/2, -0.01])\
    .faces('>X').workplane(invert=True).circle(5).cutBlind(2.0)\
    .faces('<X').workplane(invert=True).circle(5).cutBlind(2.0)
```

roipoussiere, 2019-08-13:
> Is it the result you were looking for?

Not really, because I was looking for a through hole, but I will try tonight. Thanks for your answer!

jmwright, 2019-08-13:
If you want a thru hole, just remove the `<X` face selector line and change cutBlind on the first to cutThruAll.
```
pulley_outer_diameter, pulley_thickness = 50, 10
pulley_shell_thickness, pulley_inner_diameter = 2, 10
shaft_diameter, shaft_diameter = 5, 4.5
pulley_split_thickness = pulley_thickness - pulley_shell_thickness*2

s = cq.Workplane('front').line(0, pulley_inner_diameter/2)\
    .line(-pulley_split_thickness/2, 0)\
    .lineTo(-pulley_split_thickness/2, pulley_outer_diameter/2 - pulley_shell_thickness)\
    .line(-pulley_shell_thickness, pulley_shell_thickness)\
    .line(0, -pulley_outer_diameter/2)\
    .close().mirrorY()\
    .revolve(360, [pulley_outer_diameter/2, -0.01])\
    .faces('>X').workplane(invert=True).circle(5).cutThruAll()
```

closed 2019-08-13 by roipoussiere