CadQuery/cadquery #177: Has any way install CDquery in python3.7?

Dustyposa, 2019-08-14:
I want to install cadquery in python37, has any support plane? :）

bragostin, 2019-08-14:
@Dustyposa  I have it running under python 3.7.3. I compiled the sources of oce-master and pythonocc-core-master, then I downloaded CadQuery from https://github.com/CadQuery/cadquery and CQ-editor (candidate 1) from https://github.com/CadQuery/CQ-editor and installed them with python setup.py install.

Dustyposa, 2019-08-14:
Oh thanks, the CadQuery branch is?
I want to use `asyncio.subprocess`, must in python37.
If ok, how about to push a docker images?:)

bragostin, 2019-08-14:
@bragostin I used the master branch. About Docker there a pull request opened: https://github.com/CadQuery/cadquery/pull/176

Dustyposa, 2019-08-14:
@bragostin OK, I got it. I will have a try.
The latest Docker Images's python version is `3.6.8`.
I hope have a 3.7 version. :)
The cadquery master branch's version is 1.2

bragostin, 2019-08-14:
@Dustyposa  you might encounter a Missing InitDriver call in viewer.py issue for which the solution can be found here https://github.com/CadQuery/CQ-editor/issues/52

roipoussiere, 2019-08-14:
Hello @Dustyposa 
While the docker PR is not merged yet, [the CadQuery image](https://cloud.docker.com/u/cadquery/repository/docker/cadquery/cadquery) hosted on docker hub works.

So if you have docker, just do:

```bash
echo "import cadquery; print(cadquery.Workplane('XY').box(1,2,3).toSvg())" > cadquery_script.py
cat cadquery_script.py | docker run --rm -i cadquery/cadquery > 3d_model.svg
```

Dustyposa, 2019-08-15:
@roipoussiere Oh, thanks it's work for me. But in docker , the python version is `3.6.8`, I hope an `3.7` version. `3.7` version can use `asynico.suprocess`.
If have an update python version plane or lift the conda python version limit. Please tell me. Thanks.

Dustyposa, 2019-08-15:
@bragostin I will have a look. Thanks.

bragostin, 2019-08-15:
@Dustyposa  I am working on Linux Debian 10,  oce-master and pythonocc-core-master are there easy to compile. I don't know under Windows though.

roipoussiere, 2019-08-15:
@Dustyposa I added a [`python3.7` tag](https://hub.docker.com/r/cadquery/cadquery/tags) on the Docker image:

```bash
 echo "import sys;print(sys.version)" | docker run --rm -i cadquery/cadquery:python3.7
3.7.4 (default, Jul  9 2019, 15:11:16) 
[GCC 7.4.0]
```

You should be able to run your program now.

Dustyposa, 2019-08-16:
@roipoussiere oh, thanks !!!! That's great!!  :)

closed 2019-08-16 by Dustyposa

Dustyposa, 2019-08-16:
@roipoussiere Sorry, the `cadquery/cadquery:python3.7` has a small problem.  
When I use `pip3 install xxx`. The xxx not in python37 version. Only in python3.6 version.

roipoussiere, 2019-08-16:
You should use `python -m pip install xxx`.

roipoussiere, 2019-08-16:
By the way how did you manage the use of pip install?

You make me realize that I haven't thought about that, because I directly run stdin inside the Python interpreter (ie. `docker run --rm -it cadquery/cadquery bash` will not work).

Dustyposa, 2019-08-16:
@roipoussiere I guess the dockerfile `ENTRYPOINT` broken the base cmd.
I use `docker run --rm --name test -itd cadquery/cadquery` make the container run in backends(this cmd will make --rm invalid).
Then use `docker exec -it test /bin/bash`,

roipoussiere, 2019-08-16:
> @roipoussiere I guess the dockerfile `ENTRYPOINT` broken the base cmd.

Yes, exactly.

> I use `docker run --rm --name test -itd cadquery/cadquery` make the container run in backends(this cmd will make --rm invalid).
> Then use `docker exec -it test /bin/bash`,

This is smart! :)

So I replaced the `ENTRYPOINT` by `CMD`, so you can do `docker run --rm -it cadquery/cadquery bash`, can you try again?

Also, I'm looking for a way to make it easier. What do you think about a syntax like:

```bash
cat cadquery_script.py | docker run -v $(pwd)/requirements.txt:/requirements.txt:ro --rm -i cadquery/cadquery > 3d_model.svg
```

So the script installs the pip dependencies at runtime if a `/requirements.txt` is found.

dcowden, 2019-08-16:
I really like the idea of a command line flow that allows running a script I side the container in one line. The trick to look for requirements.txt seems good.

Once you are mounting directories, you might consider looking for /data/requirements. txt rather than in root. That would allow the use case in which I want to have a script that reads other files from the filesystrm and or writes output to the file system.

As an example, suppose my script generates lots of step files as output. This is no longer good to stream to stdout. But if the script is run with /data as the working directory, I can throw requirements.txt there, and write output files there too.

Dustyposa, 2019-08-16:
@roipoussiere Oh, that's a good idea. This cmd can be a optional cmd who have a requirements.txt.
Because I have to run a server in this images. So I need exec container and do more cmd.

Dustyposa, 2019-08-16:
@roipoussiere If have a lot model need to install. Every time install requirement's too slow.

dcowden, 2019-08-16:
@Dustyposa @roipoussiere yes that's a good point!

I think we could solve those kinds of issue by making it very easy to make a derivative image.
That would be accomplished by using an ONBUILD command that will install requirements.txt.

Then, a one line dockerfile will allow making a derivative image that has the requirements installed:

`FROM cadquery:1.0`

 Combined with the above, you could then use that derivative image to run scripts in the same way.

roipoussiere, 2019-08-16:
I continue the discussion [here](https://github.com/CadQuery/cadquery/pull/176#issuecomment-522053475), since it's not related to the op's issue anymore. :)