CadQuery/cadquery #178: how to use Workplane.text

Zawarud, 2019-08-14:
I need to see an example for add text on 3d model with :
`Workplane.text(txt, fontsize, distance[, …])`
how to use this command?

jmwright, 2019-08-14:
There's an example in the test suite for CQ: https://github.com/CadQuery/cadquery/blob/master/tests/TestCadQuery.py#L1858

Documentation: https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=text#cadquery.Workplane.text

Zawarud, 2019-08-14:
Thanks. 
There are a metod to have always the text parallel to the view, if a rotate with mouse ?

jmwright, 2019-08-14:
If you're using CQ-editor, you can hold down on the middle mouse button and drag to pan the object around. That will keep the text parallel if it was parallel to the view plane when you started panning.

Zawarud, 2019-08-14:
im on jupyter lab. In this case how do this?

jmwright, 2019-08-14:
@Zawarud I'm assuming you are using the jupyter-cadquery project? https://github.com/bernhard-42/jupyter-cadquery

@bernhardBV @bernhard-42 Are you able to answer the question above?

Zawarud, 2019-08-14:
yes. jupyter-cadquery

lalebarde, 2020-05-23:
Hi, could you push back the example please, the [link](https://github.com/CadQuery/cadquery/blob/master/tests/TestCadQuery.py#L1858) is dead

nirvdrum, 2020-09-21:
A stable link is: https://github.com/CadQuery/cadquery/blob/2d721d0ff8a195a0902eb9c3add88d07546c05b1/tests/test_cadquery.py#L2701-L2750 I really wish GitHub didn't default to linking to "master".

closed 2021-01-04 by adam-urbanczyk