CadQuery/cadquery #179: Hole through drumbell fails

bragostin, 2019-08-15:
Something I don't understand. Why does this work
```python
path = cq.Workplane('XZ').moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane('YZ').workplane(offset=-2).circle(5).workplane(offset=2).circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True)
yzsh = [(1.25,1.25)]
pl = pl.faces('<X').workplane().pushPoints(yzsh).circle(5.5/2).extrude(-5).pushPoints(yzsh).hole(5.5-2*1.15, 4)
show_object(pl)
```
while this does not?
```python
path = cq.Workplane('XZ').moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane('YZ').workplane(offset=-2).circle(5).workplane(offset=2).circle(1).workplane(offset=2).circle(5).sweep(path, multisection=True)
yzsh = [(1.5,1.5)]
pl = pl.faces('<X').workplane().pushPoints(yzsh).circle(5.5/2).extrude(-5).pushPoints(yzsh).hole(5.5-2*1.15, 4)
show_object(pl)
```
The only difference being that the hole center if 0.25 mm further from the drumbell.

jmwright, 2019-08-17:
@bragostin I experimented a little bit, and it seems like it could be a CAD kernel thing, but I'm not sure. It's going to require some deeper investigation.

adam-urbanczyk, 2019-08-18:
It does seem like a CAD kernel issue. We should eventually update to OCCT 7.3, current kernel is really outdated.

bragostin, 2019-08-31:
Interestingly, replacing the circle with a polygon works:
```python
path = cq.Workplane('XZ').moveTo(-2, 0).lineTo(2, 0)
pl = cq.Workplane('YZ').workplane(offset=-2).polygon(12, 10).workplane(offset=2).polygon(12, 2).workplane(offset=2).polygon(12, 10).sweep(path, multisection=True)
yzsh = [(1.5,1.5)]
pl = pl.faces('<X').workplane().pushPoints(yzsh).circle(5.5/2).extrude(-5).pushPoints(yzsh).hole(5.5-2*1.15, 4)
show_object(pl) 
```

bragostin, 2020-06-02:
I am happy to report that OCP 7.4 solved this issue! 
The upgrade to OCP was also seamless: all my scripts still worked after the upgrade. Thank you for enabling OpenCascade 7.4 into CadQuery, a real work of art!

[image]

closed 2020-06-02 by bragostin