CadQuery/cadquery #180: Possible to use sweep and pushPoint together?

bragostin, 2019-08-16:
I want to to sweep a rectangle along a sinus path and have it repeated to fill an existing face. I was thinking to use something like 
```python
my_object  = my_object.faces('<Z').workplane().pushPoints(xy).rect(1,10).sweep(cq.Workplane('XZ').spline(pts))
```
can this work? Somehow I can't get it working.

bragostin, 2019-08-17:
This is a code that illustrate the purpose
```python
xy = [(0.,0.),(0.,10.),(0.,20.)]
my_object = cq.Workplane('XY').pushPoints(xy).rect(100.,3.).extrude(150.)
N = 100
pts = [(1.5 * sin(2.*3.14/10. * t * 150./N), t * 150./N) for t in range(N+1)]
path = cq.Workplane('XZ').spline(pts)
xyf = ([(0.,0.),(0.,10.),(0.,20.),(10.,0.),(10.,10.),(10.,20.)])
my_object = my_object.faces('<Z').workplane().pushPoints(xyf).rect(0.15, 10.).sweep(path)
show_object(my_object)
```
But the extruded wave does not appear at the xyf coordinates, while I would like to have 6.

jmwright, 2019-08-17:
I think that sweep is probably having trouble with the fact that you're creating 3 rectangles that don't define a single enclosed area, and trying to sweep them all at once. If you change the last sweep to extrude, it works ok since I think extrude is more forgiving in that way. I suspect that if you either create a single enclosed wire to extrude, or if you create the sweeps one at a time (i.e. in a loop), that you can probably achieve what you want. I'm still learning he way the CadQuery works now that it's based on PythonOCC though.

bragostin, 2019-08-17:
I think you mean something like this
```python
xy = [(0.,0.),(0.,10.),(0.,20.)]
my_object = cq.Workplane('XY').pushPoints(xy).rect(100.,3.).extrude(150.)
N = 50
pts = [(1.5 * sin(2.*3.14/10. * t * 150./N), t * 150./N) for t in range(N+1)]
xyf = [(0.,-5.), (0.,5.), (0.,15.), (10.,-5.), (10.,5.), (10.,15.)]
for xyi in xyf:
    xi, yi = xyi[0], xyi[1]
    path = cq.Workplane('XZ').workplane(origin=(xi,yi,0)).spline(pts)
    my_object = my_object.faces('<Z').workplane().moveTo(xi,yi).rect(0.15, 7.).sweep(path)
show_object(my_object)
```
I get the first row (x=0) of wavy sweep correctly, but the second row at x=10 is strangely deformed. Yet, I moved the sweep path to the new location with origin=(xi,yi,0) I would think.

adam-urbanczyk, 2019-08-18:
I don't think that you can use `sweep` and `pushPoints` reliably together. This is due to the fact that you are using an additional object (i.e. spine). For your example I could come up with the following solution:
```python
from math import sin

xy = [(0.,0.),(0.,10.),(0.,20.)]
my_object = cq.Workplane('XY').pushPoints(xy).rect(100.,3.).extrude(150.)
N = 100
pts = [(1.5 * sin(2.*3.14/10. * t * 150./N), t * 150./N) for t in range(N+1)]

xyf = [(0.,0.),(0.,10.),(0.,20.),(10.,0.),(10.,10.),(10.,20.)]

def wave(pt):
    path = cq.Workplane('XZ',origin=pt).spline(pts)
    return cq.Workplane('XY',origin=pt).rect(0.15, 10.).sweep(path).val()

waves =  cq.Workplane('XY').pushPoints(xyf).eachpoint(wave)

show_object(my_object.union(waves))
```

NB: I'm not sure what is your desired final geometry, but maybe you'd be better off using extrudes i.s.o. CSG

bragostin, 2019-08-18:
Thank you! I am travelling this week but I will try your solution as soon as I am back. I have one question. Is it not possible to use
```python
my_object =  my_object.faces('<Z').workplane().pushPoints(xyf).eachpoint(wave)
```
instead of the union?

bragostin, 2019-08-25:
@adam-urbanczyk The solution using union() works, thank you! The problem I have now is that in my model I actually have 300 of these waves to join to a solid made of 1800 elements. It takes 2h30. Do you think there is a way to accelerate this? I think this is similar to https://github.com/CadQuery/cadquery/issues/183
I was thinking to come back to using sweep and pushPoint together by writing a plugin. Do you think it could work? It seems that avoiding using unions yields better performances.
I made the same model with CREO 5, Openscad and now trying to finish it with Cadquery. I would be happy to share this comparison when it's done (I am comparing number of holes in STL file, rendering time, file size, in this order).

bragostin, 2019-08-26:
@adam-urbanczyk
With the code below, using 2 times union(), I am not able to obtain a result in a reasonable with 1 core used @ 3.3 GHz and 32 GB memory. Using a polyline() instead of spline() brings back the rendering time to 2-3 hours. That's why I thought there might be a solution where I can "grow" the pillars and waves on the faces of the shell instead of unioning them, and that it might be more efficient?
```python
from math import pi, sin
import numpy as np
import cadquery as cq

s_shell = cq.Workplane('XY').box(3, 96, 148).edges().fillet(1.45).cut(cq.Workplane("XY" ).box(2, 95, 147).edges().fillet(0.95))
    
path = cq.Workplane('XZ').moveTo(-1, 0).lineTo(1, 0)
spl = cq.Workplane('YZ').workplane(offset=-1).circle(5).workplane(offset=1).circle(0.5).workplane(offset=1).circle(5).sweep(path, multisection=True)

y_pl= np.linspace(-41, 41, 9)
z_pl= np.linspace(-66, 66, 14)
y_pl, z_pl = np.meshgrid(y_pl, z_pl)
yz_pl = [(y,z) for y,z in zip(y_pl.flatten(), z_pl.flatten())]

def s_pillar(pos):
    return spl.translate(pos).val()
pillars = cq.Workplane('YZ').pushPoints(yz_pl).each(s_pillar)

my_object =  pillars.union(s_shell).rotate((0,0,0),(0,0,1), -90)

y_cd= np.linspace(-49.5, 49.5, 11)
z_cd = np.zeros(len(y_cd))
yz_cd = [(y,z) for y,z in zip(y_cd.flatten(), z_cd.flatten())]

def cdf(pos):
    return my_object.translate(pos).val()
my_object = cq.Workplane('YZ').workplane().pushPoints(yz_cd).each(cdf).combine()

x_af = np.linspace(-45, 45, 27)
y_af = np.linspace(-44.5, 44.5, 10)
x_af, y_af = np.meshgrid(x_af, y_af)
xy_af = [(x,y) for x,y in zip(x_af.flatten(),y_af.flatten())]
Nz = int(148 / (10.8/5)) 
pts_af = [(1.5* sin(2*pi / 10.8 * t * 148/Nz), t * 148/Nz) for t in range(Nz+1)]
path_af = cq.Workplane('XZ').spline(pts_af)
#path_af = cq.Workplane('XZ').polyline(pts_af)
af = cq.Workplane('XY').rect(0.15, 7).sweep(path_af).translate((0, 0, -74))
def wave(pos):
    return af.translate(pos).val()
waveS = cq.Workplane('XY').pushPoints(xy_af).each(wave)
my_object = waveS.union(my_object)

my_object.val().exportStep('my_object.stp')
```

adam-urbanczyk, 2019-08-29:
Yes, this is what I meant with using extrudes. Once you have every solid you could try something like this:
https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_b_rep_feat___gluer.html
but this functionality is not exposed in CQ.

bragostin, 2019-08-30:
@adam-urbanczyk  thank you! I'll look into that

adam-urbanczyk, 2019-09-01:
I also found this interesting article: https://dev.opencascade.org/index.php?q=node/1179 . It does not seem to be available in OCC 6.9.1 though.

bragostin, 2019-09-02:
@adam-urbanczyk that's exactly spot on!  Often unions are merely used for assembly with no new surface creation. That's my case now. The short term workaround is to generate a separate STEP file and do the assembly in the 3D printer software for example.

adam-urbanczyk, 2019-09-02:
Stay tuned then for our own OCCT 7.3 (or maybe 7.4) bindings. There is quite some work done ( cf. 
https://github.com/CadQuery/pywrap ) but still stuff left to do.

bragostin, 2019-09-05:
@adam-urbanczyk thanks, I will!
Going through cq_utils.py in CQ-editor source code, I found another solution below, not using union() in https://github.com/CadQuery/cadquery/issues/180#issuecomment-524795480. It takes 1 min to generate instead of hours with union(). It does the assembly I need.
```python
waveS = cq.Workplane('XY').pushPoints(xy_af).each(wave)
vals = []
for o in my_object.all(): vals.extend(o.vals())
for o in waveS.all(): vals.extend(o.vals())
my_object = cq.Compound.makeCompound(vals)
my_object.exportStep('my_object.stp')
```

closed 2019-09-05 by bragostin

adam-urbanczyk, 2019-09-05:
Interesting, does it produce correct STL?

bragostin, 2019-09-05:
The connection is not real, it is just juxtaposed parts, so the STL has one shell per part. But the 3D printer software repairs them correctly. So it's a small price to pay for such a speed increase. At least in this case.