CadQuery/cadquery #182: No usable docs and examples on standalone cadquery usage

l29ah, 2019-08-21:
The ones at https://dcowden.github.io/cadquery/ suggest to use `show_object` or `exporters`, both of which aren't available after `import cadquery`:
```
  File "./a.py", line 14, in <module>
    exporters.exportShape(result)
NameError: name 'exporters' is not defined
```

jmwright, 2019-08-21:
@l29ah The latest docs are [here](https://cadquery.readthedocs.io/en/latest/).

adam-urbanczyk, 2019-08-21:
@l29ah if you use cadquery as a python module there is no magic happening in the background. `import cadquery` will only import the `cadquery` module, nothing less and nothing more. 

You could do something like this:
```python
import cadquery as cq

obj = cq.Workplane('XY').rect(4,4).extrude(1).faces('>Z').hole(3.5)
obj.val().exportStep('obj.step')
```

l29ah, 2019-08-21:
@jmwright same thing: no examples, no documentation on `exportStep`.
@adam-urbanczyk thanks!

Hatatister, 2019-09-11:
@l29ah 
Here is a minimal example using cadquerry directly and underlying pythonOCC for displaying. A prerequisite is that you have the pythonOCC package installed, either by using the conda package or if you are not using conda by building OCC / pythonOCC from source by yourself.

```
import cadquery as cq
from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()
dhl = cq.Workplane('XY').box(20, 20, 5)
display.DisplayShape(dhl.toOCC())
display.FitAll()
start_display()
```

SimLeek, 2020-02-02:
@HLevering I'm using that script as a quick display while editing in PyCharm.
@adam-urbanczyk I plan on using this to export from PyCharm.

Shouldn't we add this to the library with some examples? That way people don't have to install a different IDE to see output.

jmwright, 2020-02-02:
@SimLeek We've always tried to keep a separation of concerns between generating the model (cadquery library) and displaying the model (execution environment like CQ-editor). Many of CQ's users want to use CQ in a headless system (i.e. on a server), and they don't need the GUI infrastructure. The inclusion of `SimpleGui` breaks that separation, but that is probably just an artifact of the way PythonOCC is bundled. I don't know that we can guarantee that it will still be available after the upgrades to OCCT 7.3.x and 7.4.x. We could however discuss including a separate environment file to pull in GUI related packages if/when SimpleGui isn't available anymore.

We did include a command line tool in the CQ 1.0 repo that handled exporting models mostly. We could also discuss re-creating that tool for CQ 2.0 and wrapping this functionality into it. If passed a path on the command line it could watch for changes and update the SimpleGui instance when changes are made to the file.

In any case, I think an example showing how to do this could be added to the contrib repo via a PR if someone wants to take that on. https://github.com/CadQuery/cadquery-contrib

SimLeek, 2020-02-02:
@jmwright Hmm... not sure about that contrib repo. It includes some nice and more complicated examples, but this is more of a hack currently. I'm open to making a pull request, but I'd rather have something more correct that won't be broken in future releases.

I think the best solution would be to allow cq-editor to be launched from a python script. CQ-Editor's installation needs to be fixed though.

Another solution would be exporting a resulting solid and viewing that. I might want that if I make a program that can modify models by itself, but until then, it's much nicer to be able to see where all the wires and points are.

I think I should look into using CQ-Editor from PyCharm first, then into exporting models from PyCharm.

nirvdrum, 2020-09-20:
@SimLeek Were you able to make any progress on this? I'm brand new to CadQuery, so apologies if this was addressed somewhere already. I'd like to be able to develop in PyCharm and then launch CQ-Editor (or some other command) to visualize the object while retaining the ability to use my IDE.

adam-urbanczyk, 2020-09-20:
You can use CQ-editor with any external editor/IDE - just enable autoreloading:
[image]

nirvdrum, 2020-09-20:
@adam-urbanczyk Thank you! I saw the file view didn't refresh when it changed the file externally and I looked through the application preferences for a reload option, but didn't find anything. I didn't think to check the toolbar as well. You just made my day.

adam-urbanczyk, 2020-09-20:
Yes, also `File->Automatic reload and preview`

Tanneguydv, 2021-04-02:
Hello, I tried to set the "Automatic reload and preview" to the lowest possible, "1", but still it's quite long to update with my modifications from Pycharm (5 seconds or more, it reloads quicker when I click on the cq-editor window).
Is there a way to make it quicker? To see the change from my parametric functions instantly. I used to do that with Openscad/Muscad and it's really fast, but the problem is that there are less functions available than CadQuery.

The perfect solution would be to have the cq-editor visualizer available in a cadquery function (like "show(result)" for example), I cannot get the OCC display working on my computer.

And on another subject your documentations and videos are very well made, thanks a lot. It's just complicated for the beginners like me to make the bridge between Anaconda install and another IDE. I found the solution to make cadquery standalone work with Pycharm, it was very logical and you certainly know how to do it but I'm not used to deal with those issues and I think I'm not the only one, so :
- follow install instructions with miniconda prompt (to avoid installing Anaconda)
- have Freecad installed
In pycharm, select : 
- Add interpreter
-> Conda environment
-> Existing environment
-> Select the path of the python.exe from your cadquery-stable install
(mine = ‪C:\Users\'myname'\.conda\envs\cadquery-stable\python.exe)
-> Ok
And then you're in the correct environment to execute cadquery.

Note : I had to had 'Freecad 0.18/bin' to the file specified while "import freecad" to make it work (there was only up to 0.17 specified)

jmwright, 2021-04-02:
@Tanneguydv Are you running the old version of CadQuery (1.x)? CadQuery 2.x doesn't depend on FreeCAD anymore.

Tanneguydv, 2021-04-02:
> @Tanneguydv Are you running the old version of CadQuery (1.x)? CadQuery 2.x doesn't depend on FreeCAD anymore.

Ok, thank you for the precision! At first I've pip installed CadQuery and then see that it was no longer up to date, and as I had trouble with the conda environment CadQuery 1.x was been called while import. 
I've mixed my memories of code adjustments, sorry!

adam-urbanczyk, 2023-09-06:
See here: https://cadquery.readthedocs.io/en/latest/importexport.html

closed 2023-09-06 by adam-urbanczyk