CadQuery/cadquery #183: How to merge shape?

Dustyposa, 2019-08-23:
I know the function `union` can merge two shape. But when I use it in a loop.It's too slow.  
Have any other way to merge shape? Lilke `BRepAlgoAPI_Fuse` in `OCC`.  
Thanks in advance.

adam-urbanczyk, 2019-08-23:
`BRepAlgoAPI_Fuse` is used under the hood:
https://github.com/CadQuery/cadquery/blob/b0b99599173e0eec952ed2e8381822217ed949d4/cadquery/occ_impl/shapes.py#L572

dcowden, 2019-08-23:
@Dustyposa the speed is a cad kernel thing.  I've learned that the order of fusing and cutting can sometimes make a difference.  

For example, if you were adding 'spokes' onto a bicycle wheel, you'd probably find that first fusing all of the spokes together ( which would produce an OCC compound solid ), and then fusing that with the hub would perform better than fusing each spoke onto the hub one at a time.

Dustyposa, 2019-08-24:
@dcowden Thanks. I will hava a try.
But I use `BRepAlgoAPI_Fuse` like more speed than `union`, it's a illusion?

@adam-urbanczyk  Thanks.

adam-urbanczyk, 2019-08-25:
Could you share the two codes so that we can investigate it? Otherwise it'll hard to identify the issue.

Dustyposa, 2019-08-26:
@adam-urbanczyk 
cost time and code add import time 
---
below is OCC fuse code
```python
start_time = time.time()

from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Core.gp import gp_Trsf, gp_Vec

def location_from_vector(x, y, z):
    trsf = gp_Trsf()
    trsf.SetTranslation(gp_Vec(x, y, z))
    loc = TopLoc_Location(trsf)
    return loc

keyborad = BRepPrimAPI_MakeBox(keyborad_length, keyborad_weight, keyborad_height).Shape()

for k in range(9):
    for i in range(9):
        key = BRepPrimAPI_MakeBox(key_size, key_size, key_size_height).Shape()
        key.Move(
            location_from_vector(i + key_size * i + key_size // 2, key_size // 2 + k * key_size + k, keyborad_height))
        keyborad = BRepAlgoAPI_Fuse(keyborad, key).Shape()

print(f"OCC cost time is:{time.time() - start_time}")

>>> OCC cost time plus import time is:3.8429999351501465
```
below is cadquery by union
```python
start_time = time.time()
import cadquery as cq

keyboard = cq.Workplane("XY").box(keyborad_length, keyborad_weight, keyborad_height)
for k in range(9):
    for i in range(9):
        key = cq.Workplane("XY").workplane(offset=keyborad_height)\
            .center(i + key_size * i + key_size // 2, key_size // 2 + k * key_size + k)\
            .box(key_size, key_size, key_size_height)
        keyboard = keyboard.union(key)

print(f"cadquery union cost time plus import time is:{time.time() - start_time}")

>>> cadquery union cost time plus import time is:10.79200005531311
```
below is cadquery not use union
```python
start_time = time.time()
import cadquery as cq

keyboard = cq.Workplane("XY").box(keyborad_length, keyborad_weight, keyborad_height)
for k in range(9):
    for i in range(9):
        keyboard = keyboard.workplane(offset=keyborad_height).center(i + key_size * i + key_size // 2,
                                                                     key_size // 2 + k * key_size + k). \
            box(key_size, key_size, key_size_height)
print(f"cadquery no union cost time plus import time is:{time.time() - start_time}")

>>> cadquery no union cost time plus import time is:11.014000177383423
```

cost time and code no import time
---
```python
start_time = time.time()

def location_from_vector(x, y, z):
    trsf = gp_Trsf()
    trsf.SetTranslation(gp_Vec(x, y, z))
    loc = TopLoc_Location(trsf)
    return loc

keyborad = BRepPrimAPI_MakeBox(keyborad_length, keyborad_weight, keyborad_height).Shape()

for k in range(9):
    for i in range(9):
        key = BRepPrimAPI_MakeBox(key_size, key_size, key_size_height).Shape()
        key.Move(
            location_from_vector(i + key_size * i + key_size // 2, key_size // 2 + k * key_size + k, keyborad_height))
        keyborad = BRepAlgoAPI_Fuse(keyborad, key).Shape()

print(f"OCC cost time plus import time is:{time.time() - start_time}")

start_time = time.time()

keyboard = cq.Workplane("XY").box(keyborad_length, keyborad_weight, keyborad_height)
for k in range(9):
    for i in range(9):
        keyboard = keyboard.workplane(offset=keyborad_height).center(i + key_size * i + key_size // 2,
                                                                     key_size // 2 + k * key_size + k). \
            box(key_size, key_size, key_size_height)
print(f"cadquery no union cost time plus import time is:{time.time() - start_time}")

start_time = time.time()

keyboard = cq.Workplane("XY").box(keyborad_length, keyborad_weight, keyborad_height)
for k in range(9):
    for i in range(9):
        key = cq.Workplane("XY").workplane(offset=keyborad_height)\
            .center(i + key_size * i + key_size // 2, key_size // 2 + k * key_size + k)\
            .box(key_size, key_size, key_size_height)
        keyboard = keyboard.union(key)

print(f"cadquery union cost time plus import time is:{time.time() - start_time}")

>>> OCC cost time plus import time is:3.322999954223633
>>> cadquery no union cost time plus import time is:10.483999967575073
>>> cadquery union cost time plus import time is:10.317999839782715
```

Dustyposa, 2019-08-26:
and import cadquery need than 1s in my computer, below is cmd
`python3.7 -X importtime -c "import cadquery"`

adam-urbanczyk, 2019-08-29:
Thanks @Dustyposa. I'll run your code though a profiler to see what is going on.

adam-urbanczyk, 2019-09-13:
I did not have time yet to profile it yet, but just inspecting the code I can see that we are calling fuse in a loop:
https://github.com/CadQuery/cadquery/blob/b0b99599173e0eec952ed2e8381822217ed949d4/cadquery/cq.py#L2407
This is a bad idea and should be handled by the kernel.

CPME, 2019-10-27:
I too would be interested in an efficient method for making multi solid parts/assemblies. I ran into this problem modeling chainmail. Maybe I need to understand compounds better, but I feel like being able to pass cq.CQ.union() a list of CQ objects would be sensible. Thank you for this library btw. It is quite nifty.

asprionj, 2020-01-02:
The same (i.e. very long execution time) also applies when cutting / intersecting many objects from / with another one. First `union`ing all the objects to be cutted and then cutting this one does not help.

Andrei-Pozolotin, 2020-03-18:
is there a simple patch a user can apply to resolve this before CQ2.1 ?

adam-urbanczyk, 2020-03-18:
You could try with the OCP branch of  CQ + an OCP build pulled from azure: https://dev.azure.com/cadquery/pywrap/_build/results?buildId=379&view=results

Andrei-Pozolotin, 2020-03-18:
then question about future direction: does it mean OCP will replace OCC some time soon?

adam-urbanczyk, 2020-03-18:
That is possible, however should be transparent from the user point of view. Both  PythonOCC and OCP are rather thin warappers of OCCT. Ath the end of the day the goal is to have the latest OCCT version as a basis and be future proof w.r.t. coming OCCT releases.

Andrei-Pozolotin, 2020-03-19:
ok, then: OCC also went for OCCT 7.4:
https://github.com/tpaviot/pythonocc-core/releases
```
From release 7.4.0, pythonocc-core depends on the official OpenCascade-7.4.0 library (https://dev.opencascade.org)
```
what prevents CQ master from using that?

adam-urbanczyk, 2020-03-19:
What is your concern @Andrei-Pozolotin , do you want to contribute a PythonOCC7.4 test branch? Feel free then to continue the discussion here #226 .

adam-urbanczyk, 2020-04-06:
@Dustyposa took some tim, but on top of #312 switching off `clean` speeds things up in some situations:
```python
%prun result = cq.Workplane("XY").rect(20,20).extrude(h).faces('>Z').workplane().rarray(1,1,20,20).polygon(3,1).extrude(1,clean=False)
```
vs
```python
%prun result = cq.Workplane("XY").rect(20,20).extrude(h).faces('>Z').workplane().rarray(1,1,20,20).polygon(3,1).extrude(1,clean=True)  #true is defaul
```

Dustyposa, 2020-04-07:
@adam-urbanczyk Got it. Thanks! :smile:

adam-urbanczyk, 2020-07-19:
Solved by #312 , loops with bool ops were removed. Additional overhead comes from the optional (by default on) clean operation.

closed 2020-07-19 by adam-urbanczyk