CadQuery/cadquery #187: testText test case failing

bweissinger, 2019-09-04:
Running on a Fedora 30 VM and a conda env with CQ installed. Wanting to check if this is occurring for others as well.

obj2.val().Volume() appears to be returning 0.
```
======================================================================
FAIL: testText (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dev/Projects/cadquery/tests/TestCadQuery.py", line 1880, in testText
    self.assertLess(box.val().Volume(),obj2.val().Volume())
AssertionError: 7.999999999999998 not less than 0.0

----------------------------------------------------------------------
```

Using cq-editor to view the created text, it appears the dot in the center of the 0 in '2.0' is not a complete solid.

[image]

Interestingly the text has volume by itself, and does effect the volume of the box when cut from it.

```
box = cq.Workplane("XY" ).box(4, 4, 0.5)
obj1 = box.faces('>Z').workplane()\
            .text('CQ 2.0',0.5,-.05,cut=True,halign='left',valign='bottom', font='Sans')
obj2 = box.faces('>Z').workplane()\
            .text('CQ 2.0',0.5,.05,cut=False,combine=True, font='Sans')
text = cq.Workplane("XY").text('CQ 2.0',0.5,.05,cut=False,combine=False,font='Sans')

print(obj1.val().Volume())
print(obj2.val().Volume())
print(text.val().Volume())

OUTPUT:
7.99308664152082
0.0
0.006833732128143307
```

adam-urbanczyk, 2019-09-04:
I' m not sure if I get it. Running this
```python
box = cq.Workplane("XY" ).box(4, 4, 0.5)
obj1 = box.faces('>Z').workplane()\
            .text('CQ 2.0',0.5,-.05,cut=True,halign='left',valign='bottom', font='Sans')
obj2 = box.faces('>Z').workplane()\
            .text('CQ 2.0',0.5,.05,cut=False,combine=True, font='Sans')
text = cq.Workplane("XY").text('CQ 2.0',0.5,.05,cut=False,combine=False,font='Sans')

debug(obj1.val().Volume())
debug(obj2.val().Volume())
debug(text.val().Volume())
```

I get 

```
[2019-09-04 19:23:44.581956] INFO: Generic: 7.992428555797385
[2019-09-04 19:23:44.614385] INFO: Generic: 8.00757144420262
[2019-09-04 19:23:44.630339] INFO: Generic: 0.007571444202615158
```

bweissinger, 2019-09-05:
Running in an Ubuntu VM, the test passes. I am assuming it is an issue with the font that is used in the Fedora VM, since the rendered fonts are different between the two. 

Not sure what font is being used in either one. `fc-match Sans` returns `DejaVuSans.ttf: "DejaVu Sans" "Book"` in both VM's, but it most certainly is not the font used by CQ. I also seem unable to change the font using `font='<fontname>'`.

bweissinger, 2019-10-11:
I had a chance to look into this more and it does appear to be an issue with the font used in the Fedora 30 VM. 

Maybe a specific font (open source or otherwise distributable) should be packaged and used for the test (and any tests involving fonts) so that the test is the same across all systems? It would need functionality mentioned in #198