CadQuery/cadquery #188: How to save point and line data?

Dustyposa, 2019-09-06:
I have a code, below:
```python
import cadquery as cq

rec = cq.Workplane("XY").rect(50, 50)

show_object(chair)
```
It' can run and show a rect in display window.
But how can i exporter data?
I tried `AMF\STL\SVG`... But failed. 
Thanks in advance.:)

bragostin, 2019-09-06:
```python
rec.val().exportStep('rec.stp')
```

Dustyposa, 2019-09-06:
@bragostin Oh thanks, `stp` I can get .
But I need display it in front end, front end used `three.js`. 
Has any other format can export?

bragostin, 2019-09-06:
I also use STL format but I don't know if that works with ```three.js```
```
rec.val().exportStl('rec.stl', 0.1)
```

Dustyposa, 2019-09-06:
@bragostin STL can work on `three.js`, But below code  
`rec = cq.Workplane("XY").rect(50, 50)`  
can't get a stl data, the file data is None.

jmwright, 2019-09-06:
@Dustyposa You can use the `exporters` package and the `exporters.ExportTypes.TJS` (ThreeJS) type. Below are a couple of references on how to use it.

https://github.com/dcowden/cadquery/blob/master/cq_cmd.py#L35

https://github.com/CadQuery/cadquery/blob/master/tests/TestExporters.py#L50

The following works for me:
```python
import cadquery as cq
from cadquery import exporters

result = cq.Workplane("XY" ).box(3, 3, 0.5).edges("|Z").fillet(0.125)

file = open("/home/jwright/Downloads/test.tjs",'w')

exporters.exportShape(result, exporters.ExportTypes.TJS, file, 0.1)

file.flush()
file.close()
```

jmwright, 2021-02-05:
Stale issue. Closing.

closed 2021-02-05 by jmwright