CadQuery/cadquery #189: Are not these code snippets supposed to give the same output?

Peque, 2019-09-13:
```python
cq.Workplane('XY').moveTo(0.0, 0.0).lineTo(10.0, 0.0).lineTo(3.0, 5.0).close().extrude(2.0)
```
[image]

```python
points = [(0.0, 0.0), (10.0, 0.0), (3.0, 5.0), (0.0, 0.0)]
cq.Workplane('XY').polyline(points).extrude(2.0)
```

[image]

Using [`jupyter-cadquery`](https://github.com/bernhard-42/jupyter-cadquery#installation), so maybe it is not updated to latest CadQuery.

adam-urbanczyk, 2019-09-13:
I get the same error. In order to make it work I need to call `close` after `polyline`:

```python
points = [(0.0, 0.0), (10.0, 0.0), (3.0, 5.0), (0.0, 0.0)]
res2 = cq.Workplane('XY').polyline(points).close().extrude(2.0)
```

Peque, 2019-09-13:
@adam-urbanczyk Yeah, that works just fine thanks. :blush:

Feel free to close this issue if that is the expected way to extrude a polyline.

Peque, 2019-09-13:
Maybe a mention in the documentation could help. Either in the docstring or in the example, where `.close()` is not used:

https://cadquery.readthedocs.io/en/latest/examples.html#polylines

Notice that example may not be yielding the expected profile either (there is no i-beam shape generated). The result is not closed as a solid object, but rather a bent surface. Also, if you remove the `.mirrorY()` call, it would raise an exception as well. Why would you allow to extrude a non-closed polyline after mirroring but not before?

dcowden, 2019-09-13:
@adam-urbanczyk @Peque yes, .close() is designed to signal explicit intent to connect back to the starting point of the polyline.  It would also be ok to create a closed polyline without using close(), by simply providing the last point equal to the first point.  The rationale for allowing both forms was that sometimes it is not convenient to use .close()-- you actually want all the points to be defined in numeric form. for example using pushPoints.

Peque, 2019-09-13:
@dcowden What would be the call to generate the "i-beam shape" in the documentation examples then?

There should be an explicit `.close()` call, for what I understand, but it seems it cannot be called after `.mirrorY()` (?).

jmwright, 2019-09-13:
Not sure if this is totally related, but there has been some unexpected behavior in both CQ 1.0 and CQ 2.0 when mirroring polylines: https://github.com/CadQuery/cadquery/issues/125

adam-urbanczyk, 2019-09-13:
Good one, the example seems to be broken. The following works for me:
```
(L,H,W,t) = ( 100.0, 20.0, 20.0, 1.0)
pts = [
    (0,H/2.0),
    (W/2.0,H/2.0),
    (W/2.0,(H/2.0 - t)),
    (t/2.0,(H/2.0-t)),
    (t/2.0,(t - H/2.0)),
    (W/2.0,(t -H/2.0)),
    (W/2.0,H/-2.0),
    (0,H/-2.0)
]
result = cq.Workplane("front").polyline(pts).mirrorY().extrude(L)
```

[image]

adam-urbanczyk, 2019-09-13:
@Peque  `mirrorY` calls implicitely `wire` . This is indeed not very consistent. Maybe we should call it too in extrude?

https://github.com/CadQuery/cadquery/blob/b0b99599173e0eec952ed2e8381822217ed949d4/cadquery/cq.py#L1573