CadQuery/cadquery #191: Question: fitting a shape in a surface

Peque, 2019-09-15:
Imagine I have a surface like this:

```python
box = cq.Workplane("XY").box(10, 10, 4) \
    .faces(">Z").workplane().hole(3)
surface = box.faces(">Z")
surface
```

[image]

Now I want to place a cylinder on top of it, at a "random" position, but only if the whole cylinder's base sits on top of the surface. In example, this would be valid:

```python
position = (-2, 3)
surface.center(*position).circle(1).extrude(2)
```

[image]

But this would not:

```python
position = (-1, 1)
surface.center(*position).circle(1).extrude(2)
```

[image]

Is there a way to know if the circle (the cylinder's base) is fully contained in the surface?

The method should work for any surface shape (i.e.: not only for a perfectly squared surface with a perfect circle on the middle). The cylinder on top could also have a different shape in its base (like an irregular polygon).

I guess what I am looking for is a way to see if a face/polygon is contained in another face/polygon, right? The two faces/polygons are of course in the same plane.

adam-urbanczyk, 2019-09-16:
I don't think we have that in CQ. You could `union` the two faces and check if the area did not change. Probably there are also smarter methods.

dcowden, 2019-09-16:
@Peque yeah this sounds like something you'd have to do an intersection using OCE. This is possible in CQ, because CQ wraps all of the native objects. it'd be something like GeomAPI_IntCS or something similar, (not an OCE expert )

Peque, 2019-09-16:
Thanks for your feedback! :blush:

I will try the `.union()` method to see if it works. Otherwise, I will have a look at OCE lower-level operations for this task.

Peque, 2019-09-16:
It seems I cannot `.union()` faces:

[image]

So I guess I will have to go deeper... :sweat_smile: Any hints/advise would be really appreciated. :blush:

dcowden, 2019-09-16:
@Peque you can use .val() to access the underlying OCE object from a CQ wrapper.   then, once you have two faces, its time to dive into OCE.

Here's the (not super easy to navigate api docs for pythonOCC ):
https://cdn.rawgit.com/tpaviot/pythonocc-core/804f7f3/doc/apidoc/0.18.1/

In practice, when i was doing OCE i would usually just get the opencascade document package:
https://www.opencascade.com/content/documentation

which has the c++ level docs.

this part is really, really fuzzy, but I think one way to do it might be to use BRepExtrema_DistShapeShape (https://cdn.rawgit.com/tpaviot/pythonocc-core/804f7f3/doc/apidoc/0.18.1/OCC.BRepExtrema.html#OCC.BRepExtrema.BRepExtrema_DistShapeShape)

if two surfaces intersect, the distance will be zero.

Peque, 2019-09-18:
This is definitely more difficult than expected. :joy:

Is there a CadQuery-way to get the vertices of the surface in 2D coordinates? (relative to the workplane associated to that surface) I mean, for the example above it is easy, as you can just "remove" the Z coordinate. But if the surface was tilted, then the coordinates would be transformed. Not sure I am explaining myself well...

The idea would be to use these coordinates to create a [Shapely](https://shapely.readthedocs.io/en/latest/manual.html) polygon, which has a [`.contains()`](https://shapely.readthedocs.io/en/latest/manual.html#object.contains) method. Then I could use the resulted coordinates and go back to CadQuery.

dcowden, 2019-09-18:
Hi @Peque, 
There are <roughly> two ways to convert an arbitrary surface into '2d' coordinates.  One way is to parameterize the surface.  This is possible in OCE, but I suspect isn't what you are after, because you're working on surface intersections.

The other way is called projection-- you will project a surface onto another one. Most commonly, a complex surface is projected onto a plane-- the result is curves and points in the 2-d space of that plane. This is also possible in OCE.

Neither of these seem like the best way to compute surface intersections, however, if that's your intended outcome. 

In your problem definition, the face is planar, as is the bottom face of the cylinder you're positioning. Do you want a solution that will also consider non-planar surfaces? Or are you willing to accept the assumption that both are planar?  If you want to handle arbitrary surfaces, you will have no choice but to perform a surface intersection ( per suggestion above).  

If you're willing to accept that both surfaces are planar, then it might be possible to use some simpler methods ( projections or bounding boxes ) to  more quickly compute the result.  For example, you might consider using bounding boxes, and figuring out if one is within the other.

Peque, 2019-09-18:
@dcowden Yeah, faces are (co)planar always. I want to know if one surface is fully contained in another.

Peque, 2019-09-18:
An example:

```python
plane = cq.Workplane('XY').box(2, 2, 1).rotateAboutCenter(cq.Vector(-1, -1, 0), 30).faces(">Z")
plane
```

[image]

Now I would like to get the vertices of that surface in a way that they represent a square in 2D. Can CadQuery do that without diving into OCE? :innocent: 

Ideally, the 2D coordinates would match those of the "workplane". In example, if one vertex was at `(x, y)` position in the 2D plane, then this would extrude a cylinder on top of the vertex:

```python
plane.workplane().center(x, y).circle(.5).extrude(1)
```

Peque, 2019-09-18:
So far I have seen there are methods to project between local and global coordinates in the [`cq.Plane` class](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Plane).

I could access that method through the `plane` attribute, but I am still unable to find the proper object to find as parameter... :joy:

```python
plane = cq.Workplane('XY').box(2, 2, 1).rotateAboutCenter(cq.Vector(-1, -1, 0), 30).faces(">Z")
plane.plane.toLocalCoords(plane.vertices())
```

Leaving this here in case someone can shed some light. I will probably continue reading/trying during the weekend. :blush:

dcowden, 2019-09-18:
@Peque good idea on toLocalCoords()-- i didnt think of that.  You're pretty close. toLocalCoords() method needs a vector that represents a point.  plane.vertices() is a list, not a single point, so you could try just giving one point instead of a list.

Looking at other plane methods, you could also try isWireInside:

https://dcowden.github.io/cadquery/_modules/cadquery/freecad_impl/geom.html#Plane.isWireInside

it looks like it uses the transformGeometry method on wire. 
if you get a face reference ( faces(">Z").val() ), then you should have an OCE face. you should be able to call transformGeometry on that to translate it to a plane with a normal in the z direction.

Peque, 2019-09-18:
@dcowden Seems `toLocalCoords()` is not applying any transformation:

```python
>>> plane.plane.toWorldCoords(cq.Vector(0.3, -0.689, 1.14))
Vector: (0.3, -0.689, 1.14)
```

Neither is `toWorldCoords()`. Maybe I did not select the proper plane with the code above? :thinking: 

I will have a look at `isWireInside()`. :+1:

dcowden, 2019-09-18:
@Peque if toLocalCoords() gives back the same values, it means that somehow the plane selected is aligned with world coordinates, which is odd, and i wouldnt expect from your code above.

Peque, 2019-09-18:
@dcowden I was hoping I selected the wrong plane. That would be easier to fix. :joy:

dcowden, 2019-09-19:
@Peque can you print out the normal vector for the the plane in question, and see which way its pointing? that might provide guideance about whether the workplane is oriented , and thus whether its an issue with that, or with toLocalCoords()

Peque, 2019-09-20:
@dcowden Not sure this is how I should be printing the normal vector (please, tell me otherwise):

```python
solid = cq.Workplane('XY').box(2, 2, 1).rotateAboutCenter(cq.Vector(-1, -1, 0), 30)
plane = solid.faces(">Z")
plane.plane.__dict__
```

Which gives:

```python
{'zDir': Vector: (0.0, 0.0, 1.0),
 'xDir': Vector: (1.0, 0.0, 0.0),
 'yDir': Vector: (0.0, 1.0, 0.0),
 '_origin': Vector: (0.0, 0.0, 0.0),
 'lcs': class<'gp_Ax3'>,
 'rG': <cadquery.occ_impl.geom.Matrix at 0x7f308bfa2c50>,
 'fG': <cadquery.occ_impl.geom.Matrix at 0x7f308bfa2be0>}
```

dcowden, 2019-09-20:
Interesting. So the problem is definitely that the plane we have as a result is still essentially the xy plane.

Are we sure that the result of the rotate command is resulting in what we expect? 

What's the output of plane.vertices()

Peque, 2019-09-20:
```python
plane.vertices()
```

[image]

Also:

```python
for vertex in plane.vertices().objects:
    print(vertex.__dict__)
```

Gives:

```python
{..., 'X': -1.0428020990810754, 'Y': 1.0428020990810754, 'Z': -0.2740940792943281}
{..., 'X': -1.176776695296637, 'Y': -0.8232233047033631, 'Z': 0.43301270189221935}
{..., 'X': 0.8232233047033631, 'Y': 1.176776695296637, 'Z': 0.43301270189221935}
{..., 'X': 0.6892487084878017, 'Y': -0.6892487084878017, 'Z': 1.1401194830787669}
```

Which seems like coherent XYZ coordinates in the "world" reference.

Peque, 2019-09-20:
@dcowden In case you do not have an environment set up locally, I managed to create a notebook on "mybinder.org":

https://mybinder.org/v2/gh/peque/cadquery-jupyterlab-binder/master

You need to create a new notebook, though, but could be useful for quick tests. :blush: