CadQuery/cadquery #192: Hole through lofted surface succeeds if `clean=False`, fails if `clean=True`

marcus7070, 2019-09-16:
Simplified example:
```
"""
Make a simple pyramid shape with loft, cut a hole that breaks the outside edge
of the base.  If the cut is cleaned, this will fail. Lofted surface seems
essential, can't recreate with just a box.
"""

import cadquery as cq

broken = True  # change this to fail or succeed

do_clean = broken
part = (cq
    .Workplane("XY")
    .rect(10, 10)
    .workplane(offset=5)
    .rect(3, 3)
    .loft()
    .faces("<Z")
    .workplane()
    .moveTo(0, 4)
    .circle(1.5)
    .cutThruAll(clean=do_clean)
)
show_object(part)
```

Error:
```
Traceback (most recent call last):
  File "./circle_corner.py", line 22, in <module>
    .cutThruAll(clean=do_clean)
  File "/lib/python3.7/site-packages/cadquery/cq.py", line 2566, in cutThruAll
    s = s.clean()
  File "/lib/python3.7/site-packages/cadquery/occ_impl/shapes.py", line 207, in clean
    upgrader.Build()
RuntimeError: Standard_ConstructionError
Geom2d_TrimmedCurve:: C is null
```

I'm using CadQuery 2.0RC0 and pythonOCC compiled from latest commit https://github.com/CadQuery/pythonocc-core/commit/701e924ae40701cbe6f9992bcbdc2ef22aa9b5ab.

Probably closely related to #179, but I get a different error message and I can't fix #179 by adding `clean=False`.

Also, I'm not sure of the GitHub etiquette here. Pretty sure this is OCC's error, but I've made the issue here because perhaps CadQuery can fix this with a more recent version of OCC? Or perhaps I'm wrong about this being OCC's fault.

adam-urbanczyk, 2019-09-22:
Thanks for reporting this @marcus7070 . It looks like an OCC issue indeed. The only solution for this I can think of is to move to a newer OCC version (7.3+ vs current 6.9.* ). It is planned but still quite some work is needed.

marcus7070, 2019-09-23:
No worries. Thanks for confirming it.
I was messing around and wrote it up into test case you might be interested in including in a later version:
```python
class TestCadQuery(BaseTest):
    ...
    def testLoftWithHole(self):
        """
        A hole which breaks the base of a lofted pyramid shape (the clean version
        fails under OCC 6.9).
        """
        def loftedhole(do_clean=False):
            s = CQ.Workplane("XY").rect(10, 10).workplane(offset=5).rect(3, 3).loft()\
                .faces("<Z").workplane().moveTo(0, 4).circle(1.5).cutThruAll(clean=do_clean)
                self.saveModel(s)
                self.assertEqual(9, s.faces().size())
        loftedhole()
        loftedhole(do_clean=True)
```

In case anyone else runs into the same problem, I suggest you work around it by replacing the circular hole with a polygon. `clean=False` isn't really a solution, because many methods further along the chain could call `clean`.

adam-urbanczyk, 2019-09-23:
I prefer to keep it open in order to not forget.

BTW: If you need a workaround, this seems to result in a valid shape (NB: you might need to set the angle properly):
```python
import cadquery as cq

broken = True  # change this to fail or succeed

do_clean = broken
part = (cq
    .Workplane("XY")
    .rect(10, 10)
    .extrude(5,taper=35)
    .faces("<Z")
    .workplane()
    .moveTo(0, 5)
    .circle(1)
    .cutThruAll(clean=do_clean)
)
show_object(part)
```

adam-urbanczyk, 2020-06-02:
It works for me on the current master. Could you confirm @marcus7070 ?
[image]

marcus7070, 2020-06-02:
I'm still working on getting OCP installed through Nix instead of Conda so can't confirm yet. But I'm sure it will work for me too, feel free to close.

closed 2020-06-04 by adam-urbanczyk