CadQuery/cadquery #194: cutEach() cuts only one element

bragostin, 2019-09-17:
I am trying to use ```cutEach()``` to create cut-outs in a box like this:
```python
import numpy as np
import cadquery as cq

y = np.arange(-20., 20., 10.)[:-1]
z = np.arange(-20., 20., 10.)[:-1]
y, z = np.meshgrid(y, z)
yz = [[y,z] for y,z in zip(y.flatten(),z.flatten())]
my_box = cq.Workplane("XY").box(4., 40., 40.)
path = cq.Workplane("XZ").moveTo(-2., 0.).lineTo(2., 0.)
pl = cq.Workplane("YZ").workplane(offset=-2.).circle(5.).workplane(offset=2.).circle(0.5).workplane(offset=2.).circle(5.).sweep(path, multisection=True)
def s_pillar(pos):
    return pl.translate(pos).val()
my_box = my_box.pushPoints(yz).cutEach(s_pillar)
```
but I get the cuts in the wrong place instead of the intended locations showing correctly with
```python
my_box = cq.Workplane("YZ").rect(40,40, forConstruction=True).pushPoints(yz).each(s_pillar)
```
Is there something obviously wrong here?

bragostin, 2019-09-17:
My mistake, it works with
```python
my_box = my_box.faces('<X').workplane(offset=-2).pushPoints(yz).cutEach(s_pillar)
```

jmwright, 2019-09-17:
@bragostin Can this be closed then, or is there still a problem?

closed 2019-09-17 by bragostin