CadQuery/cadquery #197: rotateAndCopy() not implemented?

bragostin, 2019-10-03:
I wanted to use the rotateAndCopy() function like this:
```python
m =  [[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0]]
m = cq.Matrix(matrix=m).rotateX(45*pi/180)
ppl = cq.Workplane('YZ').workplane(offset=-2).polygon(4, 10).rotateAndCopy(m)
```
but I receive the following error message
```
Traceback (most recent call last):
  File "star_pillar.py", line 172, in <module>
    ppl = cq.Workplane('YZ').workplane(offset=-2).polygon(4, 10).rotateAndCopy(m)
  File "/usr/local/lib/python3.7/dist-packages/cadquery-0.5_SNAPSHOT-py3.7.egg/cadquery/cq.py", line 1584, in rotateAndCopy
    consolidated.wires().vals(), matrix)
  File "/usr/local/lib/python3.7/dist-packages/cadquery-0.5_SNAPSHOT-py3.7.egg/cadquery/occ_impl/geom.py", line 652, in rotateShapes
    raise NotImplementedError
NotImplementedError
```
It looks it is not implemented. Is that the case?

jmwright, 2019-10-03:
@bragostin `rotateShapes` has been short-circuited to raise a NotImplemented error: https://github.com/CadQuery/cadquery/blob/5c82cccdca1c1362995141384303a740eb8f6a86/cadquery/occ_impl/geom.py#L634

It should be easy enough to fix. @adam-urbanczyk do you know why rotateShapes has been disabled? I checked, and that's not a carry-over from the CQ 1.0 codebase.

bragostin, 2019-10-05:
I have enabled  ```rotateShape()``` in ```geom.py``` and played a bit with it but it looks broken, so it was disabled for good reason I guess. Anyway, what I really wanted to do was to rotate a polygon sketch, so I modified the ```polygon()``` function in ```cq.py``` in thiy way:
```python
    def polygon(self, nSides, diameter, rotation=0., forConstruction=False):
        """
        Creates a polygon inscribed in a circle of the specified diameter for each point on
        the stack

        The first vertex is always oriented in the x direction.

        :param nSides: number of sides, must be > 3
        :param diameter: the size of the circle the polygon is inscribed into
        :param rotation: rotation angle of the polygon
        :return: a polygon wire
        """
        rotation = math.pi / 180. * rotation
        def _makePolygon(center):
            # pnt is a vector in local coordinates
            angle = 2.0 * math.pi / nSides
            pnts = []
            for i in range(nSides + 1):
                pnts.append(center + Vector((diameter / 2.0 * math.cos(angle * i + rotation)),
                                            (diameter / 2.0 * math.sin(angle * i + rotation)), 0))
            return Wire.makePolygon(pnts, forConstruction)

        return self.eachpoint(_makePolygon, True)
```
by adding the lines
```python
rotation = math.pi / 180. * rotation
```
and modifying to add the rotation variable
```python
pnts.append(center + Vector((diameter / 2.0 * math.cos(angle * i + rotation)),
                                            (diameter / 2.0 * math.sin(angle * i + rotation)), 0))
```
I get my rotated polygon.
Other users might want to rotate polygons as well, do you think it is worth a pull request?

adam-urbanczyk, 2019-10-05:
@bragostin @jmwright  I don't remember what was the exact reasoning. It had something to do with the fact that `geom` should not be concerned with this kind of operations and that there were already `copy` and `rotate` methods in the `shapes` submodule:
https://github.com/CadQuery/cadquery/blob/1d5a5876175b057f9c65975375d1c5fd0bb2a7a9/cadquery/occ_impl/shapes.py#L481

Of course calling the fluent API should not result in a `NotImplementedError`.

bragostin, 2019-10-05:
I did not think about using ```rotateAboutCenter()``` to rotate my 2D sketch because in the documentation it is listed in 3D operations, so that I thought it applied only to solids, but it does work to rotate 2D sketches too. Problem solved, thank you! 
Maybe it could make sense to mention in the documentation that this method also works on sketches?

bragostin, 2019-10-05:
There seem to be an issue using ```rotateAboutCenter()``` together with ```cutThruAll()``` or ```extrude()``` though.
The following code
```python
xy = [[-10.,-10.],[-10.,10.],[10.,-10.],[10.,10.]]
bpl = cq.Workplane('YZ').box(30.,30.,4.)
ppl = bpl.faces('<X').workplane(offset=-2).pushPoints(xy).polygon(4, 8).rotateAboutCenter((1,0,0), 45).cutThruAll()
show_object(ppl)
```
produces rotated copies of the polygon in addition to the original polygon and cuts both though. Without ```cutThruAll()``` I get only the rotated polygons without the original (which is the desired outcome).

adam-urbanczyk, 2019-10-06:
Would the following code solve your issue?
```python
xy = [[-10.,-10.],[-10.,10.],[10.,-10.],[10.,10.]]
bpl = cq.Workplane('YZ').box(30.,30.,4.)
ppl = bpl.faces('<X').workplane(offset=-2)\
    .pushPoints(xy)\
    .polygon(4, 8,forConstruction=True)\
    .rotateAboutCenter((1,0,0), 45)\
    .cutThruAll()
```
In general all wires/edges on the stack are used for the next 3D operation (unless marked `forConstruction`)

[image]

bragostin, 2019-10-06:
@adam-urbanczyk Yes it works! I had not considered that ```rotateAboutCenter()``` was creating a new wire object that was added to the stack. Thank you very much!

closed 2019-10-06 by bragostin