CadQuery/cadquery #198: Ability to use font specified at path

bweissinger, 2019-10-11:
It would be handy to be able to use a font located at a specific file path. Right now you can specify what font to use with `text()` by font name. PythonOCC/OCE will search the registered system fonts to find the named font. As far as I can tell there is currently no way in CadQuery to use a non-system font.

`register_font` from `OCC.Core.Addons` does provide capability to register a font for use. I propose that we wrap `register_font` and either allow a file path to be passed to `text`, and then register and use that font, or create a separate function to allow users to register additional fonts.

adam-urbanczyk, 2019-10-13:
Agreed, I added this to the CQ2.1 issue list.

cactrot, 2020-09-18:
I was just working on doing this. Excellent that you are already planning on implementing it.

The use of the "typing" module threw me off for a bit, but this appears to load a font properly.

# register the font
font_mgr = cq.occ_impl.importers.OCP.Font.Font_FontMgr
font_aspect = cq.occ_impl.importers.OCP.Font.Font_FontAspect_Regular

font_mgr_instance = font_mgr.GetInstance_s()

font_list = cq.occ_impl.importers.OCP.Font.Font_NListOfSystemFont()

a_string = cq.occ_impl.shapes.TCollection_AsciiString

system_font = cq.occ_impl.importers.OCP.Font.Font_SystemFont(a_string("Liberation Sans"))

system_font.SetFontPath(font_aspect, a_string("""C:\\Windows\\fonts\\LiberationSans.ttf"""))

font_mgr_instance.RegisterFont(system_font, False)

#check if it worked
font_instance = font_mgr_instance.FindFont(a_string("Liberation Sans"), font_aspect)

font_instance.FontName().ToCString()

font_instance.FontPath(font_aspect).ToCString()

font_mgr_instance.AvailableFonts(font_list)

while font_list.Size():
    item = font_list.First()
    font_name = item.FontName()
    font_path = item.FontPath(font_aspect)
    print(font_name.ToCString(), font_path.ToCString())
    font_list.RemoveFirst()

jmwright, 2020-09-21:
@cactrot Would you be interested in putting together a PR for this?

cactrot, 2020-10-14:
I believe the merge of #453 resolves this issue.

closed 2020-10-14 by jmwright