CadQuery/cadquery #199: pushPoints missing half of the points

bragostin, 2019-10-11:
The code below makes only 8 holes out of the 16 in the list. There is this loop because there are 2 types of holes with different diameters. It seems to create the coordinate lists correctly. Any obvious reason for this behaviour?

```python
import numpy as np
import cadquery as cq

y_sh = np.array([14.5,15,15,14.5,29.5,62.3,70,70,119,119,125,125,129.8,131.8,132,132]) - 100./2.
z_sh = np.array([22.5,39.5,89.5,107.5,15.9,122.5,22.5,107.5,22.5,107.5,39.5,89.5,140.,15.9,108,126]) - 150./2.
dsh = np.array([7,5,7,7,5,5,7,7,7,7,7,7,5,5,5,5]) 

cq_object = cq.Workplane('XY').box(5, 100, 150)

for dh in np.unique(dsh):
    ysh = y_sh[dsh==dh]
    zsh = z_sh[dsh==dh]
    yzsh = [(y,z) for y,z in zip(ysh.flatten(), zsh.flatten())]
    print('dh, ysh, zsh = ', dh, ysh, zsh)
    print('yzsh = ', yzsh)
    cq_object = cq_object.faces('<X').workplane().pushPoints(yzsh).hole(dh)
    
show_object(cq_object)
```

Anthony-Gaudino, 2019-10-12:
@bragostin the holes aren't appearing because your box is too small, if you look at some of the point positions you code is generating, they are being placed outside of the box.

Remember that the box is in position (0, 0) when created and if it's size is of `100` as specified it goes from **-50** to **+50** in the **Y Axis**. Your code is generating for example point `(75.0, -35.5)` which is actually located outside of the box.

Setting you box like `cq_object = cq.Workplane('XY').box(5, 200, 150)` for example will display all 16 holes.

bragostin, 2019-10-12:
@Anthony-Gaudino, thank you! I had completely forgotten about the centered by default feature...

closed 2019-10-12 by bragostin