CadQuery/cadquery #201: Move to groups.io

adam-urbanczyk, 2019-10-18:
Shall we consider moving to https://groups.io/ , they seem to be free for open source projects and offer github integration. Google groups seem to be abandoned by google so maybe it is good to think about alternatives .

tolot27, 2019-10-18:
Did you ever use gitter.im? It is not a group, but a room for communication with github integration.

jmwright, 2019-10-18:
I've used Gitter for some other projects, but have never really liked it all that well. Most of the groups that I've been on seem to die out over time as well. Here's an example: https://gitter.im/tpaviot/pythonocc-core

In any case, I support moving away from Google Groups. I've been concerned for about the last year that Google is going to sunset it, maybe even without a lot of warning. It just isn't getting the updates and developer love that it should to be viable long-term. I just wasn't sure what other options would work. groups.io looks promising though.

tolot27, 2019-10-18:
I agree that Gitter has some drawbacks (i.e. crash on notifications, missing search, etc.). groups.io looks promising, indeed. But I wonder how they finance this free service and how long they keep alive.

dcowden, 2019-10-18:
i'm definitely in agreement with moving to _something_ else. i'm probably fine with gitter or groups.io. I'd _personally_ prefer slack.  It integrates with everything, wont be going away, and is free ( as long as we're ok with a 10k message archive limit, which i think we probably are).  But i'll admit the main reason for liking it is that i already use it daily, so i'd be able to provide support much more easily

tolot27, 2019-10-18:
The most annoying bug in slack are the rare notifications via email. Neither I use the desktop app nor the mobile app nor I have the slack website open in a tab. I already made the correct e-mail notification settings but it simply does not work as expected. Hence, I cannot recommand Slack for non-always-on users.

jmwright, 2019-10-18:
I use Slack for work and it's fine, but I've had trouble with Slack in larger open source groups that use it. I end up losing messages (including mentions), and it's hard to find some message threads that I want to refer back to. I also have trouble with email notifications, like what @tolot27 mentions.

dcowden, 2019-10-18:
I see.  I have not had those problems, but I don't doubt others have.  I'll go with the preference of the group.

jmwright, 2019-11-22:
Not to muddy the waters too much, but we could self-host Discourse.

https://discourse.org/

There don't seem to be any GitHub plugins that are currently maintained though. That would put it at a disadvantage compared to groups.io if GitHub integration is important to us. The advantage is we wouldn't have to worry about the groups.io service ever going away if we host Discourse ourselves.

jmwright, 2019-11-22:
I guess I have to take that back. It does have GitHub integration. https://discourse.org/integrations#github

techdragon, 2022-02-08:
Now that Github have added discussions https://docs.github.com/en/discussions ... this feels like it has an obvious answer. Just needs enabling by the repo owners.

dcowden, 2022-02-08:
I would support giving GitHub discussions as try.

jmwright, 2022-02-08:
Does/can GH Discussions replace a traditional email list like Groups.io? If you need a GitHub account to use it, I think there are people who use Google Groups who would be left out if there's no email option.