CadQuery/cadquery #203: cq.CQ.add() malfunction

CPME, 2019-10-27:
I'd like to add solids to the stack, so I can easily pass them to jupyter-cadquery.show().  Here is my attempt to add cq objects to the stack. I apologize if I am not understanding how this works.

`import cadquery as cq
from jupyter_cadquery.cadquery import (Assembly, Part, Edges, Faces, Vertices, show)
from jupyter_cadquery import set_sidecar
u.imperial.enable()

set_sidecar("CadQuery")

ring_od = 5
ring_id = 4
ring_t = 1
clearance = 1
ring_offset = ring_od+clearance
n_cells_x = 4
n_cells_y = 4

ring = cq.Workplane('XY').circle(ring_od).circle(ring_id).extrude(ring_t)
ringx = ring.rotate((0, 0, 0), (1, 0, 0), 90).translate((ring_offset, 0, 0))
ringy = ring.rotate((0, 0, 0), (0, 1, 0), 90).translate((0, ring_offset, 0))
ringy2 = ring.rotate((0, 0, 0), (0, 1, 0), 90).translate((0, 2*ring_offset, 0))
ring = cq.CQ.add(ringx)

show(ring, axes=True)`

Output:
`---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-35-0e25ce7447fe> in <module>
     21 #ringy.add([ringx, ringy])
     22 #print(ringy.all())
---> 23 ring = cq.CQ.add(ringx)
     24 
     25 # Creating many rings using looping union (very slow)

TypeError: add() missing 1 required positional argument: 'obj'`

I don't understand this error, given the only argument I pass to add() is a cq object. I replace "ringx" with "obj=ringx" and I get "1 required positional argument: 'self'". I add the kwarg "self=ringx" before "obj=ringx" and it runs, but does nothing useful.

adam-urbanczyk, 2019-10-27:
https://github.com/CadQuery/cadquery/blob/66d36b747487f88bdf7c325dce6791670764cc96/cadquery/cq.py#L221

You are using `add` as if it was a static method, while it is not.  Maybe you meant:
```python
ring = ring.add(ringx)
```

CPME, 2019-10-28:
Oh. That seems obvious now, thank you.

closed 2019-10-28 by adam-urbanczyk