CadQuery/cadquery #206: Bezier edge

fi0, 2019-11-04:
I can see there's an example for spline. Is there a helper function for bezier?

adam-urbanczyk, 2019-11-04:
Nope, in principle you can achieve the same with splines. Do you have any specific use case?

fi0, 2019-11-05:
I was trying to convert an SVG bezier path into CadQuery. I'll try splines.

adam-urbanczyk, 2019-11-05:
I'll reopen - we'd like to have SVG import eventually.

adam-urbanczyk, 2019-11-05:
Here is an example of using OCC api to prepare a 2D bezier:
https://github.com/CadQuery/rcad/blob/d814e0caee0d0ccc3ad4ec7fc92b9711a0c6ce95/ext/_rcad/_rcad.cpp#L920

adam-urbanczyk, 2020-07-09:
@fi0 we have preliminary DXF import though without splines. It will be eventually extended with splines. DXF will likely receive more attention than SVG.

adam-urbanczyk, 2021-01-04:
DXF import supports splines now - closing again.

closed 2021-01-04 by adam-urbanczyk