CadQuery/cadquery #207: shell crash

l4u, 2019-11-04:
example code 
```
pts = [
    (0, 7.05),
    (38.06, 7.05),
    (38.06, 2.3),
    (57.05, 2.3),
    (57.05, 0.05),
    (75.05, 0.05),
    (75.05, 2.3),
    (94.05, 2.3),
    (94.05, 7.07),
    (133.79, 7.07),
    (133.79, 67.6008157099696),
    (119.45896117526387, 92.56106913130021),
    (103.78992074042897, 83.54818611695411),
    (66.53717169362733, 78.01950957997295),
    (55.80788235294117, 63.31),
    (0, 63.31),

]
result = cq.Workplane("front").polyline(
    pts).close().extrude(10).faces("+Z").shell(0.1)
```

l4u, 2019-11-04:
version: cadquery-2.0RC1 py3.6 on macOS

l4u, 2019-11-04:
```
python(60142,0x108f74d40) malloc: *** error for object 0x7f9729ee8ba8: pointer being freed was not allocated
python(60142,0x108f74d40) malloc: *** set a breakpoint in malloc_error_break to debug
fish: 'python ./script.py' terminated by signal SIGABRT (Abort)
```

adam-urbanczyk, 2019-11-04:
It is similar to the following issue: https://github.com/CadQuery/CQ-editor/issues/94

I checked you code in linux and I get a `Standard_ConstructionError` exception. I see two issues:

1. On MacOS exception in oce cause a segfault for some reason.
2. CAD kernel does not work as expected for the model in question.

Regarding (1) I don't have access to a mac to debug this, so I will rely on the community help. It would be a good start if you could paste a backtrace from a debugger. Regarding (2) you'll have to wait till we wrap a newer CAD kernel (OCCT 7.3 or 7.4 - it is slowly coming).

adam-urbanczyk, 2019-11-04:
Similar issue: https://github.com/tpaviot/pythonocc-core/issues/655

michaelgale, 2019-11-04:
Does this help?  From faulthandler with macOS 10.14.6 / python 3.6:  
```
$ python crash.py
python(44019,0x115ffb5c0) malloc: *** error for object 0x7f98ba307d68: pointer being freed was not allocated
python(44019,0x115ffb5c0) malloc: *** set a breakpoint in malloc_error_break to debug
Fatal Python error: Aborted

Current thread 0x0000000115ffb5c0 (most recent call first):
  File "/Users/michaelgale/miniconda3/envs/cq2/lib/python3.6/site-packages/cadquery/occ_impl/shapes.py", line 1110 in shell
  File "/Users/michaelgale/miniconda3/envs/cq2/lib/python3.6/site-packages/cadquery/cq.py", line 848 in shell
  File "crash.py", line 36 in <module>
Abort trap: 6
```

As is already suspected, it is yet another example of OCCT 6.9 crashing.  I am all too familiar with these scenarios and often have to "outsource" the same operation to FreeCAD to perform without crashing (with its OCCT 7.x based CAD kernel)

michaelgale, 2019-11-05:
Interestingly, changing the code to this:
```
result = cq.Workplane("front").polyline(pts).close(). \
         extrude(10).edges("|Z").fillet(0.15).faces(">Z").shell(0.1)
```
seems to offer a bit of a workaround (assuming this is what you intended)...
[image]

michaelgale, 2019-11-05:
@adam-urbanczyk As a follow up here's the lldb backtrace with greater detail...
```
>>> exec(open("crash.py").read())
python(44865,0x1004895c0) malloc: *** error for object 0x133da9c78: pointer being freed was not allocated
python(44865,0x1004895c0) malloc: *** set a breakpoint in malloc_error_break to debug
Process 44865 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff75b412c6 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff75b412c6 <+10>: jae    0x7fff75b412d0            ; <+20>
    0x7fff75b412c8 <+12>: movq   %rax, %rdi
    0x7fff75b412cb <+15>: jmp    0x7fff75b3b457            ; cerror_nocancel
    0x7fff75b412d0 <+20>: retq
Target 0: (python) stopped.
(lldb) thread list
Process 44865 stopped
* thread #1: tid = 0x160b203, 0x00007fff75b412c6 libsystem_kernel.dylib`__pthread_kill + 10, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
(lldb) thread backtrace
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff75b412c6 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff75bfcbf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff75aab6a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff75bba077 libsystem_malloc.dylib`malloc_vreport + 545
    frame #4: 0x00007fff75bb9e38 libsystem_malloc.dylib`malloc_report + 151
    frame #5: 0x0000000104f9dd69 libc++abi.1.dylib`__cxa_end_catch + 137
    frame #6: 0x0000000106c3df11 _BRepBuilderAPI.so`_wrap_BRepBuilderAPI_MakeShape_Build(_object*, _object*) + 977
    frame #7: 0x0000000100089f95 python`_PyCFunction_FastCallDict + 245
    frame #8: 0x00000001000111d1 python`_PyObject_FastCallKeywords + 385
    frame #9: 0x000000010015f488 python`call_function + 392
    frame #10: 0x000000010015cef2 python`_PyEval_EvalFrameDefault + 46818
    frame #11: 0x0000000100150659 python`_PyEval_EvalCodeWithName + 425
    frame #12: 0x000000010015f7ea python`fast_function + 362
    frame #13: 0x000000010015f3ec python`call_function + 236
    frame #14: 0x000000010015cef2 python`_PyEval_EvalFrameDefault + 46818
    frame #15: 0x000000010015f73c python`fast_function + 188
    frame #16: 0x000000010015f3ec python`call_function + 236
    frame #17: 0x000000010015cef2 python`_PyEval_EvalFrameDefault + 46818
    frame #18: 0x0000000100150659 python`_PyEval_EvalCodeWithName + 425
    frame #19: 0x00000001001abde9 python`PyRun_StringFlags + 185
    frame #20: 0x000000010014aef4 python`builtin_exec + 532
    frame #21: 0x0000000100089f68 python`_PyCFunction_FastCallDict + 200
    frame #22: 0x000000010015f38f python`call_function + 143
    frame #23: 0x000000010015cef2 python`_PyEval_EvalFrameDefault + 46818
    frame #24: 0x0000000100150659 python`_PyEval_EvalCodeWithName + 425
    frame #25: 0x00000001001ab673 python`PyRun_InteractiveOneObjectEx + 1123
    frame #26: 0x00000001001a8341 python`PyRun_InteractiveLoopFlags + 433
    frame #27: 0x00000001001a816b python`PyRun_AnyFileExFlags + 139
    frame #28: 0x00000001001cf662 python`Py_Main + 4690
    frame #29: 0x0000000100001009 python`main + 313
    frame #30: 0x00007fff75a063d5 libdyld.dylib`start + 1
    frame #31: 0x00007fff75a063d5 libdyld.dylib`start + 1
(lldb)
```

adam-urbanczyk, 2019-11-05:
Thanks @michaelgale this is definitely helpful.

adam-urbanczyk, 2019-11-11:
@michaelgale I changed the pythonocc conda recipe and added a testcase triggering some random exception. Could you check if this solves the issue? It does seem to work on azure-ci .

https://anaconda.org/CadQuery/pythonocc-core/0.18.2/download/osx-64/pythonocc-core-0.18.2-py37h39e3cac_0.tar.bz2
or
https://anaconda.org/CadQuery/pythonocc-core/0.18.2/download/osx-64/pythonocc-core-0.18.2-py36h39e3cac_0.tar.bz2

adam-urbanczyk, 2019-11-12:
@l4u could you maybe check if the new builds solve the issue?

michaelgale, 2019-11-13:
@adam-urbanczyk I've not been able to build the Conda recipe above--even after editing the hardcoded Xcode v.9.4 in build.sh.  I'm not terribly familiar with Conda build workflows--so it could be my own fault.  Transcript below:

```
$ conda-build recipe/
No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
WARNING:conda_build.metadata:No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
Adding in variants from internal_defaults
INFO:conda_build.variants:Adding in variants from internal_defaults
Adding in variants from /Users/michaelgale/Downloads/pythonocc-core-0.18.2-py36h39e3cac_0/info/recipe/conda_build_config.yaml
INFO:conda_build.variants:Adding in variants from /Users/michaelgale/Downloads/pythonocc-core-0.18.2-py36h39e3cac_0/info/recipe/conda_build_config.yaml
/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/environ.py:418: UserWarning: The environment variable 'CC' is undefined.
  UserWarning
/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/environ.py:418: UserWarning: The environment variable 'CXX' is undefined.
  UserWarning
BUILD START: ['pythonocc-core-0.18.2-py36h39e3cac_0.tar.bz2']
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed

Leaving build/test directories:
  Work:
 /Users/michaelgale/miniconda3/conda-bld/work
  Test:
 /Users/michaelgale/miniconda3/conda-bld/test_tmp
Leaving build/test environments:
  Test:
source activate  /Users/michaelgale/miniconda3/conda-bld/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla
  Build:
source activate  /Users/michaelgale/miniconda3/conda-bld/_build_env

Traceback (most recent call last):
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/environ.py", line 757, in get_install_actions
    actions = install_actions(prefix, index, specs, force=True)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/common/io.py", line 88, in decorated
    return f(*args, **kwds)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/plan.py", line 474, in install_actions
    txn = solver.solve_for_transaction(prune=prune, ignore_pinned=not pinned)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/core/solve.py", line 117, in solve_for_transaction
    should_retry_solve)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/core/solve.py", line 158, in solve_for_diff
    force_remove, should_retry_solve)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/core/solve.py", line 275, in solve_final_state
    ssc = self._add_specs(ssc)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/core/solve.py", line 555, in _add_specs
    explicit_pool = ssc.r._get_package_pool(self.specs_to_add)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/resolve.py", line 523, in _get_package_pool
    pool = self.get_reduced_index(specs)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/common/io.py", line 88, in decorated
    return f(*args, **kwds)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/resolve.py", line 544, in get_reduced_index
    explicit_specs, features = self.verify_specs(explicit_specs)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda/resolve.py", line 280, in verify_specs
    raise ResolvePackageNotFound(bad_deps)
conda.exceptions.ResolvePackageNotFound:
  - libcurl==7.65.3=h16faf7d_0
  - openssl==1.1.1d=h0b31af3_0
  - ninja==1.8.2=h04f5b5a_1001
  - curl==7.65.3=h22ea746_0
  - oce==0.18.2=0
  - freeimageplus==3.17.0=0
  - ncurses==6.1=h0a44026_1002
  - libcxx==9.0.0=h89e68fa_1
  - bzip2==1.0.8=h01d97ff_1
  - libffi==3.2.1=h6de7cb9_1006
  - libuv==1.33.1=h01d97ff_0
  - ca-certificates==2019.9.11=hecc5488_0
  - cmake==3.12.2=h011004d_0
  - swig==3.0.12=h0a44026_1003
  - tbb==2018_20171205=0
  - expat==2.2.5=h4a8c4bd_1004
  - readline==8.0=hcfe32e1_0
[... 10 lines cut ...]
  - python==3.6.7=h4285619_1006
  - xz==5.2.4=h1de35cc_1001
  - libpng==1.6.37=h2573ce8_0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/michaelgale/miniconda3/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/cli/main_build.py", line 469, in main
    execute(sys.argv[1:])
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/cli/main_build.py", line 460, in execute
    verify=args.verify, variants=args.variants)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/api.py", line 209, in build
    notest=notest, need_source_download=need_source_download, variants=variants)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/build.py", line 2344, in build_tree
    notest=notest,
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/build.py", line 1408, in build
    create_build_envs(top_level_pkg, notest)
  File "/Users/michaelgale/miniconda3/lib/python3.6/site-packages/conda_build/build.py", line 1262, in create_build_envs
[... cut ...] 8c4bd_1004', 'libffi==3.2.1=h6de7cb9_1006', 'readline==8.0=hcfe32e1_0', 'swig==3.0.12=h0a44026_1003', 'setuptools==41.6.0=py36_1', 'rhash==1.3.6=h1de35cc_1001', 'libssh2==1.8.2=hcdc9a53_2', 'ninja==1.8.2=h04f5b5a_1001', 'libpng==1.6.37=h2573ce8_0', 'pcre==8.43=h4a8c4bd_0', 'libcurl==7.65.3=h16faf7d_0', 'ca-certificates==2019.9.11=hecc5488_0', 'bzip2==1.0.8=h01d97ff_1', 'libedit==3.1.20170329=hcfe32e1_1001', 'tk==8.6.9=h2573ce8_1003', 'cmake==3.12.2=h011004d_0', 'python==3.6.7=h4285619_1006'}
```

adam-urbanczyk, 2019-11-15:
@michaelgale  no need to build, just `conda install` it.

adam-urbanczyk, 2019-11-19:
Any luck with further testing @l4u @michaelgale ? I don't have a mac so I need to rely on your help with closing this issue.

l4u, 2019-11-20:
I ran `conda install https://anaconda.org/CadQuery/pythonocc-core/0.18.2/download/osx-64/pythonocc-core-0.18.2-py37h39e3cac_0.tar.bz2`

And I couldn't start cq-editor or import cadquery
https://gist.github.com/l4u/c219ead4c686bd38bc70a8e83978b941

adam-urbanczyk, 2019-11-21:
Thanks for trying @l4u . This is super weird, not sure how to continue. Could you share the output of `conda info` and `conda list`?

In the meantime I will add more tests to the conda package and see how a new build goes.

l4u, 2019-11-22:
https://gist.github.com/l4u/115506e8351bd952fc84060edafade6c

michaelgale, 2019-11-26:
@adam-urbanczyk I installed your conda recipe above (python 3.6).  The resulting conda environment works fine (can import cadquery, etc.) When executing the OP's code, it fails differently in that it drives the CPU 100% for around 2 minutes and then crashes with the following:

```
$ python crash.py
Fatal Python error: Segmentation fault

Current thread 0x0000000116b155c0 (most recent call first):
  File "/Users/michaelgale/miniconda3/envs/pycq3/lib/python3.6/site-packages/OCC/Core/BRepOffsetAPI.py", line 2507 in __init__
  File "/Users/michaelgale/miniconda3/envs/pycq3/lib/python3.6/site-packages/cadquery/occ_impl/shapes.py", line 1088 in shell
  File "/Users/michaelgale/miniconda3/envs/pycq3/lib/python3.6/site-packages/cadquery/cq.py", line 813 in shell
  File "crash.py", line 40 in <module>
Segmentation fault: 11
```

The lldb trace is as follows:

```
(lldb) process launch
Process 4341 launched: '/Users/michaelgale/miniconda3/envs/pycq3/bin/python' (x86_64)
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> exec(open("crash.py").read())
Process 4341 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
    frame #0: 0x0000000104b802a4 libTKG2d.11.dylib`Geom2d_BSplineCurve::Degree() const + 4
libTKG2d.11.dylib`Geom2d_BSplineCurve::Degree:
->  0x104b802a4 <+4>: movl   0x18(%rdi), %eax
    0x104b802a7 <+7>: popq   %rbp
    0x104b802a8 <+8>: retq
    0x104b802a9 <+9>: nopl   (%rax)
Target 0: (python) stopped.
(lldb) thread list
Process 4341 stopped
* thread #1: tid = 0x1ec60f8, 0x0000000104b802a4 libTKG2d.11.dylib`Geom2d_BSplineCurve::Degree() const + 4, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
(lldb) thread backtrace
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
  * frame #0: 0x0000000104b802a4 libTKG2d.11.dylib`Geom2d_BSplineCurve::Degree() const + 4
    frame #1: 0x00000001045ed51a libTKGeomBase.11.dylib`ProjLib_ProjectedCurve::Load(Handle_Adaptor3d_HCurve const&) + 5562
    frame #2: 0x00000001045ee1af libTKGeomBase.11.dylib`ProjLib_ProjectedCurve::ProjLib_ProjectedCurve(Handle_Adaptor3d_HSurface const&, Handle_Adaptor3d_HCurve const&, double) + 111
    frame #3: 0x00000001045f1f42 libTKGeomBase.11.dylib`GeomProjLib::Curve2d(Handle_Geom_Curve const&, double, double, Handle_Geom_Surface const&, double, double, double, double, double&) + 482
    frame #4: 0x00000001045f2819 libTKGeomBase.11.dylib`GeomProjLib::Curve2d(Handle_Geom_Curve const&, Handle_Geom_Surface const&) + 137
    frame #5: 0x0000000103089c92 libTKOffset.11.dylib`BRepOffset_Offset::Init(TopoDS_Vertex const&, TopTools_ListOfShape const&, double, bool, double, GeomAbs_Shape) + 2802
    frame #6: 0x00000001030890a1 libTKOffset.11.dylib`BRepOffset_Offset::BRepOffset_Offset(TopoDS_Vertex const&, TopTools_ListOfShape const&, double, bool, double, GeomAbs_Shape) + 161
    frame #7: 0x0000000103063db3 libTKOffset.11.dylib`BRepOffset_MakeOffset::BuildOffsetByArc() + 4083
    frame #8: 0x0000000103060eb8 libTKOffset.11.dylib`BRepOffset_MakeOffset::MakeOffsetShape() + 824
    frame #9: 0x00000001030796ec libTKOffset.11.dylib`BRepOffset_MakeOffset::MakeThickSolid() + 28
    frame #10: 0x00000001030163a0 libTKOffset.11.dylib`BRepOffsetAPI_MakeThickSolid::Build() + 32
    frame #11: 0x00000001030162e9 libTKOffset.11.dylib`BRepOffsetAPI_MakeThickSolid::BRepOffsetAPI_MakeThickSolid(TopoDS_Shape const&, TopTools_ListOfShape const&, double, double, BRepOffset_Mode, bool, bool, GeomAbs_JoinType) + 233
    frame #12: 0x000000012abcf7cb _BRepOffsetAPI.so`_wrap_new_BRepOffsetAPI_MakeThickSolid__SWIG_1(_object*, int, _object**) + 1163
    frame #13: 0x000000012abb666e _BRepOffsetAPI.so`_wrap_new_BRepOffsetAPI_MakeThickSolid(_object*, _object*) + 110
    frame #14: 0x0000000100089a82 python`PyCFunction_Call + 66
    frame #15: 0x000000010015d474 python`_PyEval_EvalFrameDefault + 47604
    frame #16: 0x00000001001508c9 python`_PyEval_EvalCodeWithName + 425
    frame #17: 0x00000001001602cc python`_PyFunction_FastCallDict + 364
    frame #18: 0x0000000100008f80 python`_PyObject_FastCallDict + 320
    frame #19: 0x00000001000305f8 python`method_call + 136
    frame #20: 0x00000001000105ce python`PyObject_Call + 62
    frame #21: 0x00000001000b15b5 python`slot_tp_init + 117
    frame #22: 0x00000001000b5af1 python`type_call + 241
    frame #23: 0x0000000100008ef1 python`_PyObject_FastCallDict + 177
    frame #24: 0x000000010015f718 python`call_function + 392
    frame #25: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #26: 0x00000001001508c9 python`_PyEval_EvalCodeWithName + 425
    frame #27: 0x000000010015fa8a python`fast_function + 362
    frame #28: 0x000000010015f67c python`call_function + 236
    frame #29: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #30: 0x000000010015f9dc python`fast_function + 188
    frame #31: 0x000000010015f67c python`call_function + 236
    frame #32: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #33: 0x00000001001508c9 python`_PyEval_EvalCodeWithName + 425
    frame #34: 0x00000001001ac249 python`PyRun_StringFlags + 185
    frame #35: 0x000000010014b1e4 python`builtin_exec + 532
    frame #36: 0x000000010008a068 python`_PyCFunction_FastCallDict + 200
[... cut ...]
    frame #38: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #39: 0x00000001001508c9 python`_PyEval_EvalCodeWithName + 425
    frame #40: 0x00000001001abad3 python`PyRun_InteractiveOneObjectEx + 1123
    frame #41: 0x00000001001a8791 python`PyRun_InteractiveLoopFlags + 433
    frame #42: 0x00000001001a85bb python`PyRun_AnyFileExFlags + 139
    frame #43: 0x00000001001cfb82 python`Py_Main + 4690
    frame #44: 0x0000000100000f59 python`main + 313
    frame #45: 0x00007fff75a063d5 libdyld.dylib`start + 1
(lldb)
```

rainman110, 2020-02-25:
We had a very similar issue in TiGL: https://github.com/DLR-SC/tigl/issues/651

It happened only on macOS together with the conda packages. We got the same error
```
python(44019,0x115ffb5c0) malloc: *** error for object 0x7f98ba307d68: pointer being freed was not allocated
python(44019,0x115ffb5c0) malloc: *** set a breakpoint in malloc_error_break to debug
```

The reason seemed to be mixing the conda compiler with the local macOS compiler. When I rebuild OCCT with the macOS compiler (not the conda one), the problem was fixed.

The error happens only, when you try to catch an exception, which does not  seem to work between the two different compilers.

adam-urbanczyk, 2020-02-25:
Thanks @rainman110 ! Which compiler version did you use in the end?

rainman110, 2020-02-25:
I used the command line tools that macos installs automatically.

jmwright, 2021-02-05:
Are any MacOS users from the community able to confirm if this issue has been fixed now that we're using the OCP bindings and running on OCCT 7.4.0+?