CadQuery/cadquery #209: Anisotropic scaling

yguel, 2019-11-11:
Dear cadquery team,

I cannot perform anisotropic scaling:

- cadquery version: commit a17c217803bfa8ee1252853722512fae7ba716de (HEAD -> master, origin/master, origin/HEAD) Date:   Fri Jul 12 07:29:02 2019 -0400
- ubuntu 18.04 as an Os
- Follow is the code tested:

`
import cadquery as cq

z_delta = -20
xs = 4
scaleM = cq.Matrix([
        [ xs, 0.0, 0.0, 0.0],
        [0.0, 1.0, 0.0, 0.0], 
        [0.0, 0.0, 1.0, 0.0]])
R = cq.Solid.makeBox(10,10,10)
R1 = R.transformGeometry(scaleM)
show_object(R1)
show_object(R.translate((50,0,0)))`

I was expecting to see a scaling only on the x axis, but instead the scaling was performed on all axis.
What did I do wrong ?

adam-urbanczyk, 2019-11-13:
I do confirm the issue, I think this is a legit bug @yguel .

jmwright, 2019-11-27:
@yguel @adam-urbanczyk 

The PythonOCC code below works, and is based on the forum post [here](https://pythonocc-users.gna.narkive.com/21Fpblka/problem-with-non-uniform-scaling).
```python
import cadquery as cq

from OCC.Core.gp import (gp_Mat, gp_Trsf, gp_GTrsf)
from OCC.Core.Precision import precision_Angular, precision_Confusion
from OCC.Core.BRepBuilderAPI import (BRepBuilderAPI_GTransform,
                                     BRepBuilderAPI_Transform)
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

z_delta = -20
xs = 4
#scaleM = cq.Matrix([
#[ xs, 0.0, 0.0, 0.0],
#[0.0, 1.0, 0.0, 0.0],
#[0.0, 0.0, 1.0, 0.0]])
R = cq.Solid.makeBox(10,10,10)
# R1 = R.transformGeometry(scaleM)
#R1 = cq.Shape.cast(BRepBuilderAPI_GTransform(R.wrapped, gp_GTrsf(scaleM.wrapped), True).Shape())
#show_object(R1)
#show_object(R.translate((50,0,0)))

#shape = BRepPrimAPI_MakeBox(10., 10., 10.).Shape()
xform = gp_GTrsf()
xform.SetVectorialPart(gp_Mat(
xs, 0, 0,
0, 1, 0,
0, 0, 1,
))
brep = BRepBuilderAPI_GTransform(R.wrapped, xform, False)
brep.Build()
fshape = brep.Shape()
R1 = cq.Shape.cast(fshape)
show_object(R1)

```

jmwright, 2019-11-30:
I'm still learning PythonOCC, but it's seems like gp_Trsf assumes that the scaling is the same in all directions. https://cdn.rawgit.com/tpaviot/pythonocc-core/804f7f3/doc/apidoc/0.18.1/OCC.gp.html?highlight=gp_trsf#OCC.gp.gp_Trsf

adam-urbanczyk, 2019-11-30:
@jmwright  indeed you are right. We are using `gp_Trsf` internally. Probably the simplest fix would be to  simply replace it with `gp_GTrsf` .

adam-urbanczyk, 2019-11-30:
That should be it. BTW: @yguel do you really need to perform such operations? They might have some weird side effects on the underlying geometry.

PR #225 (2019-11-30): Fixing Matrix and Shape to correctly support anisotropic scaling
This PR will resolve #209
closed 2019-11-30 by adam-urbanczyk