CadQuery/cadquery #211: debug and show_object problems

rvdb1, 2019-11-17:
I have several problems when  I try to run the following example

base_cube = cq.Workplane('XY').rect(1.0,1.0).extrude(1.0)
top_of_cube_plane = base_cube.faces(">Z").workplane()
debug(top_of_cube_plane, { 'color': 'yellow' } )
debug(top_of_cube_plane.center, { 'color' : 'blue' } )

circle=top_of_cube_plane.circle(0.5)
debug(circle, { 'color': 'red' } )

show_object( circle.extrude(1.0),{"color": "#aaaaaa" )

debug error:
Debug takes 1 positional argument but 2 are given

show_object produces a TypeError: unhashable type: 'dict'
I tried another syntax:
show_object(result, options={"rgba":(204, 204, 204, 0.5)})
But it has no effect in cq-editor.

I could change the name of an object shown in the cq-editor with the following syntax:
show_object(result, name="my_name")

I would like to set the color, alpha and visible parameters from my scripts, but all my trials were unsuccessful.

Could you provide a working example with all the available parameters ?

Thank you

jmwright, 2019-11-17:
@rvdb1 Your issue may be related to this: https://github.com/CadQuery/CQ-editor/issues/80

If you're not using CQ editor, but are instead running a script from the command line, I don't think those parameters will work there either. They're meant more for a CQGI-compliant execution environment. CQGI is our attempt at making scripts run in a similar way no matter where they are executed. However, support isn't fully implemented in the CQ 2.0 ecosystem yet.

So I would try dropping those parameters first to make sure your script runs, then work on from there, maybe chiming in on the discussion on that issue.

jmwright, 2021-02-05:
Closing as no more information was provided and this has been addressed (or is being addressed) in CQ-editor.

closed 2021-02-05 by jmwright