CadQuery/cadquery #214: exportStep and pythonOCC object

bragostin, 2019-11-21:
I created a pythonOCC Shape() and would like to save it as STEP file with CadQuery like this:
```python
cq_object = cq.CQ(pyOCC_Shape)
cq_object.val().exportStep('pyocc_shape.stp')
```
but I get the error message (I printed the pyOCC_Shape and cq_object variables to see their nature):
```pyOCC_Shape =  class<'TopoDS_Solid'; id:42083057>
cq_object =  <cadquery.cq.CQ object at 0x7f77717b5be0>
Traceback (most recent call last):
  File "test.py", line 197, in <module>
    cq_object.val().exportStep('pyocc_shape.stp')
AttributeError: 'TopoDS_Solid' object has no attribute 'exportStep'
```
I cannot find a solution in the docs. Any idea?

jmwright, 2019-11-21:
@bragostin I was able to reproduce the error, and here's how I fixed it. There might be a more elegant way, I'm not sure.
```
import cadquery as cq
from cadquery import *
import OCC.Core

# This is a stand-in for your pythonOCC Shape
box = cq.Workplane('XY').box(1, 1, 1).val().wrapped

cq_object = cq.Workplane('XY').newObject([Shape.cast(box)])

#cq_object = cq.CQ(pyOCC_Shape)
cq_object.val().exportStep('pyocc_shape.stp')
```

jmwright, 2019-11-21:
@bragostin Actually, this is cleaner and closer to your original code.
```
import cadquery as cq
from cadquery import *
import OCC.Core

box = cq.Workplane('XY').box(1, 1, 1).val().wrapped

cq_object = cq.CQ(Shape.cast(box))

#cq_object = cq.CQ(pyOCC_Shape)
cq_object.val().exportStep('pyocc_shape.stp')
```

bragostin, 2019-11-21:
@jmwright that works, thank you very much!

jmwright, 2019-11-21:
Great, glad to hear it. This ended up being the final version with all the extraneous imports removed.
```python
import cadquery as cq

box = cq.Workplane('XY').box(1, 1, 1).val().wrapped

cq_object = cq.CQ(cq.Shape.cast(box))

# cq_object = cq.CQ(pyOCC_Shape)
cq_object.val().exportStep('pyocc_shape.stp')
```

closed 2019-11-21 by jmwright