CadQuery/cadquery #216: Exporters cleanup

adam-urbanczyk, 2019-11-21:
Current public exporting interface is so-so. It would be at least good to expose `exportStep` in the fluent API.

Related to #212

marcus7070, 2019-12-22:
I'd get some use out of the AMF export option https://github.com/CadQuery/cadquery/blob/e69b2f83bde43ad6a0159a72bc058165e684e72e/cadquery/occ_impl/exporters.py#L148

The AMF format is really suited to 3D printers with dual printing heads, since it can have two (or more) objects within one file. AMF is well supported in the main open source 3D printing slicer, Slic3r. 

I haven't tried the AMF export implementation in CadQuery yet, just wanted to let you know my use case.

jmwright, 2020-05-03:
#246 is another related issue for the exporters

jmwright, 2021-02-05:
@adam-urbanczyk Hasn't this issue already been addressed by #415 and related issues?

adam-urbanczyk, 2021-02-05:
Actually it did. You can export AMF using `cq.exporters.export`

jmwright, 2021-02-05:
Ok, closing.

closed 2021-02-05 by jmwright