CadQuery/cadquery #217: Include "apply" function in the fluent API

abudden, 2019-11-25:
I've been modelling something with a lot of angle-iron sections.  I've got a little fed up with writing lots of bits like:

```python
myobj = (myobj
    .faces('>X')
    .workplane()
    .moveTo(corner_x, corner_y)
    .hLine(angle_size).vLine(-angle_thickness)
    .hLine(-(angle_size-angle_thickness)).vLine(-(angle_size-angle_thickness))
    .hLine(-angle_thickness).vLine(angle_size)
    .close()
    .extrude(angle_length)
    )
```

and it occurred to me it would be really useful if there was a way to define a function so that the code could look like this:

```python
def draw_angle_profile(obj, angle_size, angle_thickness, orientation):
    if orientation == 'southeast':
        return (obj
                .hLine(angle_size).vLine(-angle_thickness)
                .hLine(-(angle_size-angle_thickness)).vLine(-(angle_size-angle_thickness))
                .hLine(-angle_thickness).vLine(angle_size)
                .close()
        )

# elsewhere...

myobj = (myobj
    .faces('>X')
    .workplane()
    .moveTo(corner_x, corner_y)
    .apply(draw_angle_profile, angle_size, angle_thickness, 'southeast')
    .extrude(angle_length)
)

```

I know I could do this by breaking the flow, something like this:

```python
myobj = (myobj
    .faces('>X')
    .workplane()
    .moveTo(corner_x, corner_y)
    )
myobj = (draw_angle_profile(myobj, angle_size, angle_thickness, 'southeast')
    .close()
    .extrude(angle_length)
    )
```

but having an `apply` function seems a bit cleaner than leaving the string of functions with a last `moveTo`, then having to start a new set of functions.  Not the biggest issue in the world, I'll grant you, but it would presumably be very easy to implement...

adam-urbanczyk, 2019-11-25:
Are you maybe looking for those two methods: https://cadquery.readthedocs.io/en/latest/apireference.html#iteration-methods ?

abudden, 2019-11-26:
Hmm, that certainly looks interesting: I hadn't realised it could be used in this way.

Any chance you could have a look at this sample (sorry - it's a bit long: I tried to take out as much as possible from my model but wanted to be as representative as possible) and show me how I should be applying it as what I'm doing at the moment doesn't work.  The first block is just the definition of `draw_angle`, then there's some code making a frame out of box section.

The relevant section is in the code starting `contain_angle = `; I've added comments showing how I'd had it before and you can see the `.each` call where I've tried to call my `draw_angle` function.  The implementation that's commented out works; the `.each` call produces `AttributeError: 'Vector' object has no attribute 'hLine'`.

```python

#!/usr/bin/python3
# vim: set fileencoding=utf-8 :
import itertools
import cadquery as cq

def exportStep(object_list, filename):
    vals = list(itertools.chain(*[o.vals() for obj in object_list for o in obj.all()]))
    compound = cq.Compound.makeCompound(vals)
    compound.exportStep(filename)

def draw_angle(obj, angle_size, angle_thickness, orientation):
    if orientation.lower() == 'southeast':
        result = (obj.hLine(angle_size).vLine(-angle_thickness)
                .hLine(-(angle_size-angle_thickness)).vLine(-(angle_size-angle_thickness))
                .hLine(-angle_thickness).vLine(angle_size)
                .close()
                )
    elif orientation.lower() == 'southwest':
        result = (obj.hLine(-angle_size).vLine(-angle_thickness)
                .hLine((angle_size-angle_thickness)).vLine(-(angle_size-angle_thickness))
                .hLine(angle_thickness).vLine(angle_size)
                .close()
                )
    elif orientation.lower() == 'northeast':
        result = (obj.hLine(angle_size).vLine(angle_thickness)
                .hLine(-(angle_size-angle_thickness)).vLine((angle_size-angle_thickness))
                .hLine(-angle_thickness).vLine(-angle_size)
                .close()
                )
    elif orientation.lower() == 'northwest':
        result = (obj.hLine(-angle_size).vLine(angle_thickness)
                .hLine((angle_size-angle_thickness)).vLine((angle_size-angle_thickness))
                .hLine(angle_thickness).vLine(-angle_size)
                .close()
                )
    else:
        raise Exception("Unknown orientation")
    return result

box_size = 30.0
top_angle_size, top_angle_thickness = 30.0, 3.0
trolley_d, trolley_w, trolley_h = 460.0, 300.0, 500.0

# Make a frame to be used as an example
trolley = (cq.Workplane('XY').center(0, trolley_d/2.0)
        # Start with two rectangles to extrude the top frame
        .rect(xLen=trolley_w, yLen=trolley_d)
        .rect(xLen=trolley_w-2*box_size, yLen=trolley_d-2*box_size)
        .extrude(-box_size)
        # Now get the bottom face
        .faces('<Z').workplane()
        # Draw a construction rectangle to get the vertices at the middle of each upright
        .rect(xLen=trolley_w-box_size, yLen=trolley_d-box_size, forConstruction=True)
        .vertices()
        # Draw the box section (as square bar) for the uprights
        .box(box_size, box_size, (trolley_h-(3*box_size))/2.0, centered=(True, True, False))
        # Get the bottom face and draw another two rectangles for the bottom frame
        .faces('<Z').workplane()
        .rect(xLen=trolley_w, yLen=trolley_d)
        .rect(xLen=trolley_w-2*box_size, yLen=trolley_d-2*box_size)
        .extrude(box_size)

        # Repeat for a second frame
        .faces('<Z').workplane()
        # Draw a construction rectangle to get the vertices at the middle of each upright
        .rect(xLen=trolley_w-box_size, yLen=trolley_d-box_size, forConstruction=True)
        .vertices()
        # Draw the box section (as square bar) for the uprights
        .box(box_size, box_size, (trolley_h-(3*box_size))/2.0, centered=(True, True, False))
        # Get the bottom face and draw another two rectangles for the bottom frame
        .faces('<Z').workplane()
        .rect(xLen=trolley_w, yLen=trolley_d)
        .rect(xLen=trolley_w-2*box_size, yLen=trolley_d-2*box_size)
        .extrude(box_size)

        # Fillet the edges to represent the box form
        .fillet(4)
        )

contain_angle = (trolley
        # This is a really nasty way to select the correct face, but I can't get the '<Z[0]' thing working:
        # it just produces an error message
        .faces('-X').faces('>X').faces('not (>Z)').faces('>Z').workplane()
        # Need to convert this to use .edge or similar (ref github issue #218):
        .moveTo(trolley_d/2.0, -((trolley_h-(3*box_size))/4.0))
        # Try using .each to call a function to draw the angle
        .each(lambda x: draw_angle(x, top_angle_size, top_angle_thickness, 'northwest'), useLocalCoordinates=True)

        # The way I'd done it before...
        #.vLine(top_angle_size).hLine(-top_angle_thickness)
        #.vLine(top_angle_thickness-top_angle_size).hLine(top_angle_thickness-top_angle_size)
        #.vLine(-top_angle_thickness).hLine(top_angle_size)
        #.close()

        # And extrude... don't combine so that we can mirror it and produce two of them
        .extrude(trolley_w-(2*box_size), combine=False)
        )

# Produce a mirrored version so we can have the same feature on both sides
contain_angle_2 = (contain_angle
        .mirror('XZ', (0.0, trolley_d/2.0, 0.0))
        )
trolley = trolley.union(contain_angle).union(contain_angle_2)

# If in CQ-Editor GUI:
if 'show_object' in globals():
    show_object = globals()['show_object'] # Does nothing, but gets rid of a pylint error
    show_object(trolley)
else:
    # Otherwise just export the various objects
    exportStep([trolley], 'test_code.step')
```