CadQuery/cadquery #218: Support alternative workplane centre options

abudden, 2019-11-25:
The default centre option is Centre of Mass (which makes a lot less sense to me than Centre of Bounding Box, but that's by-the-by and at least the bounding box centre is also an option).  Often though, I have a face of a workpiece and I want to start drawing at one of the corners of that face.

At the moment, I do this by calculating the dimensions of the face (as I tend to do everything in the model by calculation so fundamental dimensions can be tweaked) and adding a `.center` call or a `.moveTo` call to get to a given corner.  This is a lot of boilerplate and it would be nice if there were options to `.workplane` that made the starting/centre point any of the 8 bounding box extremes (North, North-East, East etc).

adam-urbanczyk, 2019-11-25:
We have selectors for that. Check out the docs: https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors

NB: the default behaviour will change from "Center of Bounding Box" to "ProjectedOrigin"  cf. #202 

If you post a simple example I can offer more guidance.

abudden, 2019-11-26:
Thanks @adam-urbanczyk; I guess I need to better understand how the selectors work - I'd used them with faces all over the place and struggled a lot.  I found that e.g. '<Z[0]' produced error messages related to no objects found, so I've ended up with code like:

```
 .faces('+Z').faces('not (<Z)').faces('not (<Z)').faces('not (<Z)').faces('<Z').faces('not (<X)').faces('<X')
```

(that's a real example from my model).

I haven't got a "simple example" at the moment; I'll do some experimentation and try to find a simple demonstration of the sort of thing I'm trying to do.

As an aside, I'd be really really grateful if there was someone who could look at a not-very-simple example and give me general comments.  I'm fairly new at CQ and would love to have someone look at my full model and give any advice on better way of doing things.  However, the model is 396 lines long (of which about 100 lines are comments and about 90 lines are drawing angle-iron cross-sections over-and-over again), so I'd understand if no-one wants to read it!  The model is a personal one, so there's no problem posting it online if there is anyone willing to read it...

There are lots of things that feel wrong, for example:

* The long string of `.faces` that just seems like the wrong way to do it.
* The regular use of `moveTo` rather than using a selector to pick the corner of a face (ref this issue).
* I generally want `CenterOfBoundBox` as the workplane centre option so that I'm "sketching" in the middle of the selected face, but if I just get in the habit of passing it every time, I sometimes get error messages related to `tesselate` not being an attribute of `TopoDS_Face` (thankfully in every case this has happened, the current default 'CenterOfMass' has produced the same point as I'd expected so it's worked anyway).
* The repetitive code drawing particular cross-sections like the angle iron (ref #217).
* Better ways of selecting planes so that I can mirror objects around the plane through centre of another object (in a lot of cases at the moment I've just copied-and-pasted the code as it felt easier).

... so it would be great to have a more experienced viewpoint.

Regarding 'ProjectedOrigin' being the default, I don't think this is what I would naturally expect coming from a conventional CAD background (where you select a face and then sketch relative to the features in that face), but as long as the other options like Centre of Bounding Box are still available, it doesn't really bother me what the default is.  I'm struggling to think of a situation where I would want Centre of Mass (as opposed to Centre of Bounding Box or Projected Origin), so anything's got to be better than that.  I would expect face sketches based on Projected Origin would break if you go back and move the object that has the face on it, but perhaps that will be less of a problem if I figure out how these edge selectors work.

jmwright, 2019-11-26:
@abudden 

> The long string of .faces that just seems like the wrong way to do it.

Have a look at the [Combining Selectors](https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors) section of the docs to see if that helps.

> The repetitive code drawing particular cross-sections like the angle iron (ref #217).

There was some previous work done on DXF import as a way to help with things like this. It would be possible to have a library of profiles and import them directly to a CQ object. Of course, that doesn't help if your profiles are dynamically generated as-needed using a broad range of parameters.

> Better ways of selecting planes so that I can mirror objects around the plane through centre of another object (in a lot of cases at the moment I've just copied-and-pasted the code as it felt easier).

I'm not sure if this is what you're looking for, but some users save certain important workplanes in variables so that they can refer back to them more easily later.

> if I just get in the habit of passing it every time, I sometimes get error messages related to tesselate not being an attribute of TopoDS_Face (thankfully in every case this has happened, the current default 'CenterOfMass' has produced the same point as I'd expected so it's worked anyway).

Do you have an example that generates the `tesselate` error? If you haven't posted it already, that would probably be worth having its own issue.

abudden, 2019-11-26:
> @abudden
> 
> > The long string of .faces that just seems like the wrong way to do it.
> 
> Have a look at the [Combining Selectors](https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors) section of the docs to see if that helps.

That was where I started and I went through loads and loads of iterations trying (e.g.) `not (>Z) and >Y' and also trying every combination I could think of with '<Z[1]', '>Z[-3]' etc but all of them complained about empty lists or something like that (i.e. no faces found).  I've never got a single successful face out of the square-bracket notation.  The long and ridiculous string of '(not <Z).(not <Z).(not <Z).(<Z)' was the only way I found that worked.

> There was some previous work done on DXF import as a way to help with things like this. It would be possible to have a library of profiles and import them directly to a CQ object. Of course, that [...]

I've raised a separate issue on this one (#217); hopefully I'll get some advice there.
 
> I'm not sure if this is what you're looking for, but some users save certain important workplanes in variables so that they can refer back to them more easily later.

Yeah, I probably didn't explain that very well.  I had drawn on one edge of a face and wanted to do two extrudes: one where I'd drawn it and one mirrored on the opposite edge of the face.  I guess I wanted to say (very-pseudo-code):

```python
the_face_i_want = (object
    .faces().faces().faces() # whatever
    )
first_object = (the_face_i_want
    .workplane(centerOption='CenterOfBoundBox')
    .polyline(long_list_of_coordinates)
    .close().extrude(200, combine=False)
    )
# This is what I don't know how to elegantly do:
second_object = (first_object
    .mirror('XY', basePoint = middle of "the_face_i_want")
```

> Do you have an example that generates the `tesselate` error? If you haven't posted it already, that would probably be worth having its own issue.

I don't have a **short** example, but the full code (with comments showing where the errors would appear if the 'CenterOfBoundBox' option were selected) is in [this gist](https://gist.github.com/abudden/66b5d8fa66fb40d4e3a379559499d371).

dcowden, 2019-11-26:
@jmwright  @abudden @adam-urbanczyk  regarding the 'string of faces' selectors and selectors in general:

CQ is currently missing a key feature that would help a lot: operation selectors.

When an operation is performed, it should be possible to name it with a label it with a name ( or ID). Subsequently, CQ selectors should be able to select and filter  features that were created by and/or modified by that operation. 

The only framework I know of that's anything like CQ is Featurescript in Onshape.  Onshape supports this feature, and it quickly becomes the go-to method once you have it.   Its common, for example, to extrude a boss, naming the operation 'boss', and then immediately select its top face .faces(operation='boss', ">Z")

OCC has the ability to track which elements are modified or created for operations, but it is not built into the kernel.  

My original idea to implement this was to implement all CQ modelling functionality upon operation objects, which would encapsulate the ability to list modified/created entities. then, re-build the existing fluent api on top of those operations, so that selectors can use the operation id as filters. In some cases, fingerprinting and hashes may be needed to correctly determine that a feature was changed in an operation.

This is a captured in this issue https://github.com/CadQuery/cadquery/issues/49

michaelgale, 2019-11-26:
@dcowden Very interesting concept.  The CQ selector framework is one of CQ's best features.  I have gradually built up a "toolbox" of custom selector subclasses which really help when compositing complex objects.  Examples include "EdgesCommonWithVertex", "EdgesCommonWithCorner", "LengthSelector", etc. etc.  Even more powerful are the arithmetic and logical operators to really fine tune the selector behaviour.

If I understand your concept or operation selectors; it sounds like a mechanism to "tag" or "snapshot" a shape's state in "time"?  Is the idea that you can use this "tagged" state as a temporary "context" to perform an operation (like using a selector) on the shape in the "future"?  Have I got it right?  

In general, this concept of "tagging" or referencing a shape would be quite powerful.  Perhaps this state could be stored as an object variable rather than a string literal reference?  e.g. layercake1 = cake.state(),  and later on, cake = cake.state(layercake1).faces(">Z") ?  The "state" method is just a clumsy placeholder I've made up for the sake of example.

michaelgale, 2019-11-26:
p.s. Although, on reflection, I think there might be a risk with using a shape's state from the past.  Using the layer cake example, if an attempt is made to select the 1st layer top face in the future--it might not exist anymore (due to subsequent solid operations).  Then again, returning None is probably valid and ok since that is always a potential outcome of using selectors in any context.  Therefore, never mind my caution--as with all tools--you can always use them incorrectly!

dcowden, 2019-11-26:
@michaelgale yes, using previous geometry would be bad.  The approach is to attach metadata to entities ( faces, edges, and vertices), that traces their lineage to who modified them.

When you run a selector, you're always querying the latest object(s). However, you're query can include information based on the origin of the features.  Typically, this means that a surface was 'created by' a given operation, and this is 'modified by' one or more operations afterwards.  

Some tricky coding to implement this comes in when you need to make sure that each operation can identify which features it modified and created, but it is do-able

vlad0337187, 2020-06-23:
Probably I have similar problem.

Use CadQuery via `CQ-editor` GUI.
If I'll try to comment line 51 and will try to uncomment or line 53, or line 54 - it will throw error.

On line 53 I use `.center()` method to move origin to the corner of plane.
On line 54 I use `moveTo()` method to move origin to the corner of plane.

I attach all 3 screenshots:

<details>
<summary>was able center workplane during box creation</summary>
[image]
</details>

<details>
<summary>was unable to set box origin using `center()` method</summary>
[image]
</details>

<details>
<summary>was unable to set box origin using `moveTo()` method</summary>
[image]
</details>

marcus7070, 2020-06-23:
@vlad0337187, I think you might have misunderstood what `center` and `moveTo` do. 

The method `box` creates a box relative to the current `Workplane`. `center` creates a new `Workplane` that is in the same plane as the previous `Workplane`, but with a different center. So if your intention is to move the box you create in line 50, then you'll have to put the `center` call *before* you create the box. 

`moveTo` is mainly used when drawing lines and I don't think you want to use that method at all.

vlad0337187, 2020-06-23:
@marcus7070 , thank you, my bad.
Added note to my previous message that it's solved.