CadQuery/cadquery #219: Support radiusArc with relative coordinates

abudden, 2019-11-26:
Most of the line drawing functions exist in two forms.  For example `line` and `lineTo`.  Consider the code below that draws the cross section of some typical angle-iron.  The radiusArc function has to refer back to the drawing start point and work out how it relates to the position of the end of the arc.  With a relative version (shown in the comments) it would only need to know the radius.

```python
# Draws angle iron that looks a bit like
# this (with a radiused internal corner
# represented by the slashes):
#
# ┌────────┐
# │  /─────┘
# │ /
# │ │
# │ │
# └─┘

startPointX, startPointY = 12.3, 42.7
side = 75.0
thickness = 6.0
radius = 4.0
inside_length = side - (thickness + radius)

angle_section = (cq.Workplane('YZ', origin=(0, 0, 0))
        .moveTo(startPointX, startPointY)
        .vLine(-side)                       # Down the left-hand side
        .hLine(thickness)                   # Along the bottom edge
        .vLine(inside_length)               # Up the inside vertical edge
        # QUESTION: is there a way to do something relative with arcs?
        # i.e. the difference between line() and lineTo().  Would be cleaner
        # if this code was .radiusArc((radius, radius), radius)
        # rather than having to refer back to the origin of the drawing.
        .radiusArc((startPointX+thickness+radius, startPointY-thickness), angle_fillet_radius)
        .hLine(inside_length)               # Along the inside horizontal edge
        .vLine(thickness)                   # Up the right-hand side
        .hLine(-side)                       # Along the top
        .close()
        )
```

abudden, 2019-12-04:
Am I right in thinking this would be as simple as adding this to [cq.py](https://github.com/CadQuery/cadquery/blob/bb707347eaf21aea32bdfe06b247146594850b92/cadquery/cq.py#L1528)?

```python
    def radiusArc__NEWNAME__(self, endDist, radius, forConstruction=False):
        """
        Draw an arc from the current point to endDist with an arc defined by the sag (sagitta).
        :param endDist: distance from the current point to the end point of the arc 
        :type endDist: 2-tuple, relative to current point
        :param radius: the radius of the arc
        :type radius: float, the radius of the arc between start point and end point.
        :return: a workplane with the current point at the end of the arc
        Given that a closed contour is drawn clockwise;
        A positive radius means convex arc and negative radius means concave arc.
        """

        startPoint = self._findFromPoint(useLocalCoords=True)
        endPoint = (startPoint.x + endDist[0], startPoint.y + endDist[1])
        return self.radiusArc(endPoint, radius, forConstruction)
```

I guess part of the challenge is coming up with a good, non-breaking-backwards-compatibility name.  Logically, the new function would be called `radiusArc` and the old one `radiusArcTo` (to match with `line` vs `lineTo`), but that could be problematic...

I presume there's also the exercise of coming up with a test for it - I haven't dug into the test framework to work out what that might involve.

jmwright, 2019-12-04:
> I guess part of the challenge is coming up with a good, non-breaking-backwards-compatibility name. Logically, the new function would be called radiusArc and the old one radiusArcTo (to match with line vs lineTo), but that could be problematic...

Yeah, there are probably a lot of uses of `radiusArc` out in the wild that we wouldn't want to break. We'd probably need to add a deprecation warning to move users to `radiusArcTo`, and then make the switch. The only other option I can think of is adding `endDist` and `radius` to the current implementation and just set them to `None` by default. Then the function can check to see if they're being used to alter its behavior. This makes deprecation harder though because we're not getting rid of the function, just changing what it does. The warning would have to be directed towards the subset of users that use `radiusArc` as `radiusArcTo`. I've never done a deprecation warning based on what parameters you pass. I don't know if that's even a thing.

> I haven't dug into the test framework to work out what that might involve.

The tests for the current radiusArc function seem to be [here](https://github.com/CadQuery/cadquery/blob/bb707347eaf21aea32bdfe06b247146594850b92/tests/TestCadQuery.py#L985).

abudden, 2020-01-28:
I was going to suggest naming the two functions `arc` and `arcTo` to avoid the conflict with the existing `radiusArc`, but looking at the test code, I guess that would confuse people who use `sagittaArc` or `threePointArc`.  I presume the same issue (not having a relative coordinate version) applies to those two functions as well.

bernhard-42, 2020-03-15:
@abudden @adam-urbanczyk  

Now that we have `ellipseArc` (https://github.com/CadQuery/cadquery/commit/a499a4b5d353b023fc6a101150af7b438b4a69a1), would this help?

```
angle_section = (cq.Workplane('YZ', origin=(0, 0, 0))
        .moveTo(startPointX, startPointY)
        .vLine(-side)                       # Down the left-hand side
        .hLine(thickness)                   # Along the bottom edge
        .vLine(inside_length)               # Up the inside vertical edge
        .ellipseArc(x_radius=radius, y_radius=radius, 
                    angle1=90, angle2=180, sense=-1)
        .hLine(inside_length)               # Along the inside horizontal edge
        .vLine(thickness)                   # Up the right-hand side
        .hLine(-side)                       # Along the top
        .close()
        )
```

This creates an ellipse arc (`x_radius == y_radius` is allowed) with two angles `90°` and `180°`
```
    90°
   ⎛   ⎞
 180°   0°
   ⎝   ⎠
    270°
```
`sense == 1` means to go counter clockwise from `angle1` (start point) to `angle2` (end point)
`sense == -1` means to go clockwise from `angle2` (start point) to `angle1` (end point)
(this is how OCC treats the sense value)

The ellipse arc will then be added with its start point (taking `sense` into account) to the current point.

**Note:** If you would change the the direction you construct the wire, then the `ellipseArc` call gets slightly more intuitive:

```python
# startPointX, startPointY = 12.3, 42.7
side = 75.0
thickness = 6.0
radius = 4.0
inside_length = side - (thickness + radius)

angle_section = (cq.Workplane('YZ', origin=(0, 0, 0))
    .moveTo(startPointX, startPointY)
    .hLine(side)
    .vLine(-thickness)
    .hLine(-inside_length)
    .ellipseArc(radius, radius, 90, 180)
    .vLine(-inside_length)
    .hLine(-thickness)             
    .close()
)
```

marcus7070, 2020-03-15:
I've got a PR (#284) to add the method `.tangentArcEndpoint`, which draws an arc tangential to the previous line. I handled this issue by adding a `relative=True` keyword argument. 

I think the methods `line` and `lineTo` are justified because they get so much use, but if every method had a `foo` and a `fooTo` method it would get a bit crazy. If #284 gets merged as is, perhaps a keyword argument `relative` could be added to other methods as well?