CadQuery/cadquery #220: Is there a way to extrude a face without redrawing the profile?

abudden, 2019-11-26:
Sometimes it's useful to be able to "stretch" an object.  For example, consider this code that draws two lengths of angle-iron.  I'd like one angle-iron section to extend slightly at both ends to give clearance to some end-plates that have yet to be added.  I could go back to the original "sketch" and extrude it in both directions (1 mm in one direction and length+1 mm in the other), but it seems more representative of the thought process to create the second angle-iron section and just pull the two faces out a little.

```python
#!/usr/bin/python3
# vim: set fileencoding=utf-8 :
import itertools
import cadquery as cq

side = 75.0
thickness = 6.0
radius = 4.0
inside_length = side - (thickness + radius)

first_length = 750.0

first = (cq.Workplane('YZ', origin=(0, 0, 0))
        .vLine(-side)                       # Down the left-hand side
        .hLine(thickness)                   # Along the bottom edge
        .vLine(inside_length)               # Up the inside vertical edge
        .radiusArc((thickness+radius, thickness), radius)
        .hLine(inside_length)               # Along the inside horizontal edge
        .vLine(thickness)                   # Up the right-hand side
        .hLine(-side)                       # Along the top
        .close()
        .extrude(first_length)
        )

end_clearance = 1.0
second = (first
        .mirror('XZ')
        )
# What I'd like to be able to do, but can't figure out how to:
# second = (second
#         .faces('<X').extrude(end_clearance)
#         .faces('>X').extrude(end_clearance)
#         )
# I've also tried:
# second = (second
#         .faces('<X').edges().extrude(end_clearance)
#         .faces('>X').edges().extrude(end_clearance)
#         )
# I've also tried:
# second = (second
#         .faces('<X').wires().extrude(end_clearance)
#         .faces('>X').wires().extrude(end_clearance)
#         )

# As this is a simple extrusion, this nasty hack works, but it wouldn't work if any more
# complex features had been added beyond the simple extrusion.  I've added it here to
# make it clear what the end-result I'm trying to achieve should look like
second_x1 = (second
        .translate((-1.0, 0.0, 0.0))
        )
second_x2 = (second
        .translate((1.0, 0.0, 0.0))
        )
second = second.union(second_x1).union(second_x2)

# If in CQ-Editor GUI:
if 'show_object' in globals():
    show_object = globals()['show_object'] # Does nothing, but gets rid of a pylint error
    show_object(first)
    show_object(second)
```

Is there a way to do this that I'm missing?

jmwright, 2019-11-26:
@abudden Looks like your question might be related to this issue: https://github.com/CadQuery/cadquery/issues/148

adam-urbanczyk, 2019-11-26:
I think it is indeed a duplicate of #148 . If you disagree feel free to reopen @abudden .

closed 2019-11-26 by adam-urbanczyk

ulikoehler, 2024-05-18:
For everyone finding this ticket and trying to extrude a face, here's how you can do it:
```
result = face_to_extrude.extrude(2, taper=45)
```

Full example:
```python
import cadquery as cq

box = cq.Workplane("XY").box(1, 1, 1)

face_to_extrude = box.faces(">Z")

result = face_to_extrude.extrude(2, taper=45)
```
[image]

From my blog: https://techoverflow.net/2024/05/18/how-to-extrude-a-face-in-cadquery/