CadQuery/cadquery #222: AttributeError: 'TopoDS_Face' object has no attribute 'tesselate'

abudden, 2019-11-27:
As mentioned in #218, I've been having some issues with this error message.

> > if I just get in the habit of passing it (centerOption='CenterOfBoundBox') every time, I sometimes get error messages related to tesselate not being an attribute of TopoDS_Face (thankfully in every case this has happened, the current default 'CenterOfMass' has produced the same point as I'd expected so it's worked anyway).

> Do you have an example that generates the tesselate error? If you haven't posted it already, that would probably be worth having its own issue.

I'm afraid I don't have a small example that make this error happen, but I have a large example that makes it happen in 5 different places:

[https://gist.github.com/abudden/37e7e6a3c647b58a0646b047f40862d0](https://gist.github.com/abudden/37e7e6a3c647b58a0646b047f40862d0)

adam-urbanczyk, 2019-11-27:
Could you post a stack trace?

abudden, 2019-11-28:
```
Traceback (most recent call last):
  File "weldingtrolley_error.py", line 142, in <module>
    .workplane(centerOption='CenterOfBoundBox')
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\cq.py", line 358, in workplane
    center = Shape.CombinedCenterOfBoundBox(self.objects)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\occ_impl\shapes.py", line 414, in CombinedCenterOfBoundBox
    o.wrapped.tessellate(tolerance)
AttributeError: 'TopoDS_Face' object has no attribute 'tessellate'

```