CadQuery/cadquery #223: Struggling to sweep a circle along a wire

abudden, 2019-11-28:
I'm trying to model a set of drawers.  These drawers are intended to have a handle on the front, something looking [like this](https://media.rs-online.com/t_large/F7264181-01.jpg).

I've drawn the drawer and put a circle on the face where the handle will meet the drawer.  I've then added a curved path over which the circle should be swept.  However, I can't make the sweep itself work.

With `handle = handle_circle.sweep(sweep_path)` (full listing below), I get an error saying `BRepFill_Sweep::BuildEdge`.  If I add `sweepAlongWires=True`, I get an error saying `Sections must be all closed or all open`.

Can anyone help fix this issue?  I'm not finding the error messages tell me what I need to do.

```python
#!/usr/bin/python3
# vim: set fileencoding=utf-8 :
import os
import itertools
import cadquery as cq

obj_list = []

def exportStep(object_list, filename):
    """
    Simple worker function to take a list of objects and export a single step file with all objects in place.
    """
    vals = list(itertools.chain(*[o.vals() for obj in object_list for o in obj.all()]))
    compound = cq.Compound.makeCompound(vals)
    compound.exportStep(filename)

drawer_height, drawer_width, drawer_depth, drawer_thickness = 100.0, 300.0, 425.0, 0.8

handle_width = 100.0
handle_diameter = 6.0
handle_depth = 30.0
handle_corner_radius = 10.0

# Draw the drawer (five sides of a cuboid)
drawer = (cq.Workplane('XZ')
        .rect(xLen=drawer_width, yLen=drawer_thickness)
        .extrude(-drawer_depth)
        .faces('>Z').workplane(centerOption='CenterOfBoundBox')
        .rect(xLen=drawer_width, yLen=drawer_depth)
        .rect(xLen=drawer_width-(2*drawer_thickness), yLen=drawer_depth-(2*drawer_thickness))
        .extrude(drawer_height-drawer_thickness)
        )

# Draw a circle where the handle meets the drawer
handle_circle = (drawer
        .faces('>Y').workplane(centerOption='CenterOfBoundBox')
        .moveTo(-handle_width/2.0, 0)
        .circle(handle_diameter/2.0)
        )
# Draw a path over which we'd like to sweep the circle to make something that looks like a drawer handle
sweep_path = (drawer
        .faces('<Z').edges('>Y').workplane(offset=-(drawer_height/2.0), invert=True, centerOption='CenterOfBoundBox')
        .moveTo(handle_width/2.0, 0)
        .vLine(-(handle_depth-handle_corner_radius))
        .radiusArc(((handle_width/2.0)-handle_corner_radius, -handle_depth), handle_corner_radius)
        .hLine(-(handle_width-(2*handle_corner_radius)))
        .radiusArc((-handle_width/2.0, -(handle_depth-handle_corner_radius)), handle_corner_radius)
        .vLine(handle_depth-handle_corner_radius)
        .wire()
        )

# Add the drawer, the circle and the sweep path to the objects to be displayed / exported
obj_list.append(drawer)
obj_list.append(handle_circle)
obj_list.append(sweep_path)

# THIS IS WHERE IT GOES WRONG:
# Now try to sweep the circle along the path to make the handle
# If these lines are uncommented, error messages occur!
# handle = handle_circle.sweep(sweep_path)
# obj_list.append(handle)

# If in CQ-Editor GUI:
if 'show_object' in globals():
    show_object = globals()['show_object'] # Does nothing, but gets rid of a pylint error
    for obj in obj_list:
        show_object(obj)
else:
    # Otherwise just export the various objects
    outname = os.path.splitext(__file__)[0] + ".step"
    exportStep(obj_list, outname)
```

Error from the above with the commented out sweep lines put back in:

```
Traceback (most recent call last):
  File "drawer_unit_help.py", line 58, in <module>
    handle = handle_circle.sweep(sweep_path)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\cq.py", line 2373, in sweep
    transition)  # returns a Solid (or a compound if there were multiple)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\cq.py", line 2720, in _sweep
    isFrenet, transition)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\occ_impl\shapes.py", line 1431, in sweep
    builder.Build()
RuntimeError: Standard_ConstructionError
BRepFill_Sweep::BuildEdge
```

Error if I add `sweepAlongWires=True`:

```
Traceback (most recent call last):
  File "drawer_unit_help.py", line 58, in <module>
    handle = handle_circle.sweep(sweep_path, sweepAlongWires=True)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\cq.py", line 2373, in sweep
    transition)  # returns a Solid (or a compound if there were multiple)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\cq.py", line 2724, in _sweep
    thisObj = Solid.sweep_multi(sections, path.val(), makeSolid, isFrenet)
  File "c:\applications\development\languages\python\lib\site-packages\cadquery\occ_impl\shapes.py", line 1463, in sweep_multi
    builder.Build()
RuntimeError: Standard_DomainError
Sections must be all closed or all open
```

jmwright, 2019-11-28:
@abudden I poked around with your code a little bit, and don't see an obvious reason why this won't work. The sweep tests and the example all use a spline instead of assembling a wire as you have, but I can't see a difference between the way sweep assembles edges into a wire, and the way the `.wire()` call does it. I'd be curious if replacing your handle with a spline gives the same error. That wouldn't be a solution, it's just for testing.

adam-urbanczyk, 2019-11-28:
Confirmed. This does work in isolation:
```python
path = cq.Workplane("XZ").vLine(1).hLine(1)
swp = cq.Workplane("XY").circle(.1).sweep(path)
```
[image]