CadQuery/cadquery #224: Issue with Jupyter + extrude() and revolve() on Windows x64 10\7 and Python=3.7

Krande, 2019-11-29:
Hi, 

I am experiencing issues with the extrude and revolve using jupyter and python 3.7 on windows x64 10 & 7. I have tried this on 3 different machines and it is the same issue for all 3. 

Here are two examples that reproduces my issue:

s = cq.Workplane("XY").circle(.1).extrude(0.5)
s

and 

s = cq.Workplane('ZX').line(0,-r1).line(l1, 0).radiusArc((l1+l1_5,-r2), r)\
    .line(l2,0).line(0, r2).close().revolve(360, [0.0, 0.0], [1,0.0])
s

The process hangs and the memory usage by python keeps increasing until I manually stop the process.

Most of the other cadquery methods I have tried works as expected.
I have tried this with a clean Miniconda install using the cadquery version 2.0RC1. 

The same code works as expected using cq-editor

If anyone else experiences this and know a way to solve it, I would really appreciate any help! 

And also let me know what you need of additional information in order to debug this, and I will update and include my question with it. 

Kind Regards
Kristoffer

jmwright, 2019-11-29:
@Krande Are you using [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery)?

@bernhard-42 Any ideas on this?

bernhard-42, 2019-11-29:
@jmwright @Krande Yes, I know this from jupyter-cadquery.
The reason is that the dimensions are small (0.1 and 0.5). If you try 
```python
s = cq.Workplane("XY").circle(0.2).extrude(1.0)
s
```
or
```python
s = cq.Workplane("XY").circle(1.0).extrude(0.5)
s
```
it works. Somehow the ratio of base and extrusion height is relevant.

I would expect that r1, l1, ... in your second example are also small. For example
```python
f=4
r = r1 = l1 = l2 = r2 = l1_5 = 0.1 * f
s = cq.Workplane('ZX').line(0,-r1).line(l1, 0).radiusArc((l1+l1_5,-r2), r).line(l2,0).line(0, r2).close().revolve(360, [0.0, 0.0], [1,0.0])
```
works for `f=4` but hangs for `f=3`.

I have currently no idea why this happens. It definitely happens in my code, but I still don't know where. My guess is the tesselation during rendering, but I need to verify. 
My current workaround is to ensure that my object is "large enough".

@adam-urbanczyk Do you have any idea? Can it be the tesselation? My knowledge of the underlying occ is very limited.

bernhard-42, 2019-11-29:
I was able to reduce the problem to

```python
import cadquery as cq
from OCC.Core.Visualization import Tesselator

dim = 0.1
s = cq.Workplane("XY").circle(dim).extrude(0.5)

quality = 0.5
tess = Tesselator(s.objects[0].wrapped)
tess.Compute(uv_coords=False, compute_edges=True, mesh_quality=quality, parallel=True)
```
This works for `dim = 1` but hangs for `dim = 0.1`

@adam-urbanczyk It is the tesselation. what are you doing differently in cq-editor?

bernhard-42, 2019-11-29:
Side note: same happens in ipython in the terminal

Krande, 2019-11-30:
Thank you for answering so quickly! It makes a lot of sense this being related to the model-dimensions since I could have sworn I have gotten the same code working for other models in the past. I guess that's the penalty for working with meters not millimeters as my goto units for structural analyses :)

@jmwright I am not using jupyter-cadquery. I updated my original comment with the missing dimensions for my 2nd example and also added the import statement, so that each example is "complete".

Regarding the cq-editor, the version I tested and got the code working on was 0.1RC for windows.

I see that there's a 7.4.0 beta version of pythonocc-core using the latest opencascade release out now at https://github.com/tpaviot/pythonocc-core. Maybe there is some functionality in that release that could be of use?

I created a fork of the cadquery repo in a longshot attempt of seeing if I could port parts of the cadquery code to occ740, but I quickly realized that was easier said than done. I will give it a shot nonetheless, but I am still trying to grasp the concepts surrounding the whole python wrapped around c++ libraries (SWIG, CMAKE, etc.. are relatively new to me). So I am sure you guys will figure out a solution to this problem before I make any progress on the occ740 stuff.

As I tried to figure out a temporary way to visualize my model circumventing the memory issue with tesselation (or whatever the root cause) I thought I might be able to use the JupyterRenderer from OCC.Display.WebGl.jupyter_renderer just by converting the cadquery part stack to occ using the .toOCC() method. However I am seeing an error "'TopoDS_Compound' object is not subscriptable" error. Below is a minimum example. 

```
import cadquery as cq
from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer
s = cq.Workplane("XY").circle(.1).extrude(0.5)
JupyterRenderer(s.toOCC())
```

Although certainly unrelated, any chance you guys might know of a quick fix for the TopoDS_Compound issue, so that I can keep using cadquery in my daily work flow?

bernhard-42, 2019-11-30:
Try
```python
import cadquery as cq
from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer
s = cq.Workplane("XY").circle(2).extrude(5)

j = JupyterRenderer()
j.DisplayShape(s.toOCC(), update=True)
```
However it also fails for me with 0.1mm and 0.5mm

bernhard-42, 2019-11-30:
A little test of 

```python
from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeCylinder

cyl = BRepPrimAPI_MakeCylinder(0.1, 0.5).Shape()

j = JupyterRenderer()
j.DisplayShape(cyl, update=True)
```
shows that this fails for OCC 0.18.2 but is successful for 7.4.0 beta.

So I am not sure that opening an issue in pythonocc would make sense.

@adam-urbanczyk Are there any plans to update cadquery to OCC 7.4.0 soon.

adam-urbanczyk, 2019-11-30:
@bernhard-42  In the CQ-editor I am using internal visualization facilities of OCCT. The Jupyter render is an addition from PythonOCC (thus not a core functionality of OCCT). In the CQ code the `_repr_html_` method is implemented using `X3DExporter` from PythonOCC. From what @Krande writes it seems that the issue is triggered only for small models. I can suspect that the mesh quality parameter is defined in absolute and not relative terms in the version of PythonOCC we are using.

We are definitely going to use OCCT 7.4. It might be through the PythonOCC project or through our internal wrapper of OCCT.

Krande, 2019-12-01:
Okay, thanks for the help on this. 

It appears a working OCCT 7.4 port of cq will solve the root issue, and a temporary fix is to increase the modelling dimensions. 

For my part, feel free to close this issue.

closed 2019-12-03 by Krande