CadQuery/cadquery #226: Investigate PythonOCC based on OCCT7.4

adam-urbanczyk, 2019-11-30:


jmwright, 2019-11-30:
Related project on PythonOCC repo: https://github.com/tpaviot/pythonocc-core/projects/2

adam-urbanczyk, 2019-11-30:
Cf. https://github.com/tpaviot/pythonocc-core/issues/719

adam-urbanczyk, 2019-12-02:
Addons module (text related functions) is missing:

https://github.com/tpaviot/pythonocc-core/issues/723

Krande, 2020-01-30:
Hi, it seems that the RefineEdges method on BRepAlgoAPI_Fuse is missing in OCC7.4.

https://github.com/CadQuery/cadquery/blob/c53ad18b364d33059abf29338bac49d80348ee9c/cadquery/occ_impl/shapes.py#L587-L596

When using pythonocc-core=**7.4.0rc1**
Running the Example https://pythonhosted.org/cadquery/examples.html#making-lofts 

```
result = cq.Workplane("front").box(4.0, 4.0, 0.25).faces(">Z").circle(1.5) \
    .workplane(offset=3.0).rect(0.75, 0.5).loft(combine=True)
result
```
causes the following error
`AttributeError: 'BRepAlgoAPI_Fuse' object has no attribute 'RefineEdges'`

adam-urbanczyk, 2020-01-30:
It seems that `RefineEdges` and `FuseEdges`. I think you could replace it with `SimplifyResult`.

https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_algo_a_p_i___builder_algo.html#a5d258255785fcb04e4d1db72b36e3289

Krande, 2020-01-30:
Hi Adam, 

that worked like a charm! 
the edited function
```
    def fuse(self, toFuse):
        """
        Fuse shapes together
        """

        fuse_op = BRepAlgoAPI_Fuse(self.wrapped, toFuse.wrapped)
        # fuse_op.RefineEdges()
        # fuse_op.FuseEdges()
        fuse_op.SimplifyResult()
        # fuse_op.SetFuzzyValue(TOLERANCE)
        fuse_op.Build()

        return Shape.cast(fuse_op.Shape())
```

creates the expected result

[image]

Krande, 2020-03-02:
Hi, just a heads up (if you are not already on it). In the latest 7.4.0 release of pythonocc-core "Tessalator" has been renamed and moved. 

that means the lines:

https://github.com/CadQuery/cadquery/blob/3d9041e98bdd9312118c4e2abfc70317ee892f01/cadquery/occ_impl/shapes.py#L118

and

https://github.com/CadQuery/cadquery/blob/3d9041e98bdd9312118c4e2abfc70317ee892f01/cadquery/occ_impl/exporters.py#L2

should be replaced with 

```python
from OCC.Core.Tesselator import ShapeTesselator
```

Best regards
Kristoffer

Andrei-Pozolotin, 2020-03-19:
I must be missing the obvious, but:

is there a brunch
https://github.com/CadQuery/cadquery/branches

which builds with
https://github.com/tpaviot/pythonocc-core/releases/tag/7.4.0

adam-urbanczyk, 2020-03-19:
No, I don't have it. Feel free to try and report the results (e.g. failing tests) here.

Andrei-Pozolotin, 2020-03-20:
@adam-urbanczyk Adam:
* please create a branch for #300 
* or, lets forget about occ-0.18 and move straight to occ-7.4? :-)

jmwright, 2020-03-21:
> or, lets forget about occ-0.18 and move straight to occ-7.4?

OCP currently supports OCCT 7.4, as does PythonOCC. I don't think that any new work should target OCE 0.18 because it is so far behind, and the licensing issues that spawned OCE have been (at least largely) addressed.

adam-urbanczyk, 2020-03-21:
Thanks for the  work @Andrei-Pozolotin , I see a lot of test failing so let's leave it as PR..

closed 2021-01-04 by adam-urbanczyk