CadQuery/cadquery #229: Test: BRepOffsetAPI_MakeFilling implementation

bragostin, 2019-12-02:
I made an implementation of ```BRepOffsetAPI_MakeFilling``` into CadQuery in order to produce a plate surface with a thickness, enclosed by edge points, and going through interpolation points. To this end I added these 2 functions to shapes.py:
```python
from OCC.Core.GeomAbs import GeomAbs_C0 
from OCC.Extend.TopologyUtils import TopologyExplorer, WireExplorer 
from OCC.GeomAbs import GeomAbs_Intersection 
from OCC.BRepOffsetAPI import BRepOffsetAPI_MakeFilling 
from OCC.BRepOffset import BRepOffset_MakeOffset, BRepOffset_Skin

class Face(Shape):
    @classmethod
    def makeNSidedSurface(cls, edges, points, continuity=GeomAbs_C0, Degree=2, NbPtsOnCur=20, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        """
        makeNSidedSurface() creates a surface enclosed by a closed polygon defined by 'edges' and going through 'points'.
        """
        n_sided = BRepOffsetAPI_MakeFilling(Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments)
        for edg in edges:
            n_sided.Add(edg, continuity)
        for pt in points:
            n_sided.Add(pt)
        n_sided.Build()
        face = n_sided.Shape()
        return cls.cast(face).fix()

class Solid(Shape, Mixin3D):
    @classmethod
    def interpPlate(cls, surf_pts, surf_edge_pts, thickness, Degree=2, NbPtsOnCur=20, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        """
        interpPlate(): creates a plate surface that is 'thickness' thick, enclosed by 'surf_edge_pts' points,  and going through 'surf_pts' points.
        """
        # BOUNDARIES
        xmax, ymax, zmax = max(surf_pts[:,0]), max(surf_pts[:,1]), max(surf_pts[:,2]) 
        xmin, ymin, zmin = min(surf_pts[:,0]), min(surf_pts[:,1]), min(surf_pts[:,2])
        # POINTS CONSTRAINTS: eliminate edges points, otherwise crashes.
        pts_array = [gp_Pnt(*pt) for pt in surf_pts if pt[0]<xmax if pt[0]>xmin if pt[1]<ymax if pt[1]>ymin if pt[2]<zmax if pt[2]>zmin]   
        # EDGE CONSTRAINTS: build closed polygon
        e_array = [Vector(*e) for e in surf_edge_pts]
        wire_builder = BRepBuilderAPI_MakePolygon()
        for e in e_array:
            wire_builder.Add(e.toPnt())
        w = wire_builder.Wire()
        edges = [i for i in TopologyExplorer(w).edges()] 
        # MAKE SURFACE
        continuity = GeomAbs_C0 # Fixed, changing to anything else crashes.
        face = Face.makeNSidedSurface(edges, pts_array, continuity, Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments) 
        # THICKEN SURFACE
        solid = BRepOffset_MakeOffset()
        solid.Initialize(face.wrapped, thickness, 1.e-5, BRepOffset_Skin, False, False, GeomAbs_Intersection, True) #The last True is important to make solid
        solid.MakeOffsetShape()
        
        return cls(solid.Shape())
```
and this function to cq.py
```python
class Workplane(CQ):
    def interpPlate(self, surf_pts, surf_edge_pts, thickness, combine=True, clean=True, Degree=2, NbPtsOnCur=20, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        """
        interpPlate(): creates a plate surface that is 'thickness' thick, enclosed by 'surf_edge_pts' points,  and going through 'surf_pts' points.
        :Array of [x,y,z] Real coordinates: surf_pts
        :Array of [x,y,z] Real ordered coordinates: edges_pts
        :Real: thickness
        :Integer: Degree = 3 (OCCT default value, 2 works better for Gyroids)
        :Integer: NbPtsOnCur = 15 (OCCT default value, 20 works better for thickening)
        :Integer: NbIter = 2
        :Boolean: Anisotropie = False
        :Real: Tol2d = 0.00001
        :Real: Tol3d = 0.0001
        :Real: TolAng = 0.01
        :Real: TolCurv = 0.1
        :Integer: MaxDeg = 8
        :Integer: MaxSegments = 9
        :type combine: true to combine shapes, false otherwise.
        :param boolean clean: call :py:meth:`clean` afterwards to have a clean shape
        """
        
        def _makeplate(pnt):
            return Solid.interpPlate(surf_pts, surf_edge_pts, thickness, Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments)

        plates = self.eachpoint(_makeplate, True)

        # if combination is not desired, just return the created boxes
        if not combine:
            return plates
        else:
            # combine everything
            return self.union(plates, clean=clean)
```
As an example the attached script cq_InterpPlate_example.py, produces the Gyroid elementary plate of the attached screenshot through the use of ```cq.Workplane('XY').interpPlate(gxyz, xyz_max, AFT)```

The surface needs to be relatively simple, and the number of interpolation points limited, otherwise it crashes. Do you think this could be a useful addition to CadQuery?

[image]

[cq_InterpPlate_example.py.txt](https://github.com/CadQuery/cadquery/files/3911481/cq_InterpPlate_example.py.txt)

jmwright, 2019-12-02:
@bragostin Very nice, thanks for sharing. We've had a couple of users ask about modelling more "organic" shapes before, and it seems like this would be something those users would like to see.

> Do you think this could be a useful addition to CadQuery?

+1 from me. I'd like to hear any thoughts from @adam-urbanczyk and maybe @dcowden on this too.

> The surface needs to be relatively simple, and the number of interpolation points limited, otherwise it crashes.

I know that @adam-urbanczyk has been working on getting us on the latest OCC kernel (7.4). I'm wondering if that will help with the crashes.

If this ends up being integrated into CQ I would request that it come with [tests](https://github.com/CadQuery/cadquery/tree/master/tests). A [contributed example](https://github.com/CadQuery/cadquery-contrib/tree/master/examples) would be nice too if you're willing.

dcowden, 2019-12-02:
@jmwright  @bragostin very very nice! I'm a +1 to include this for sure. There need to be some tests for it to be included, for sure.

On a side note, the only concern I have with this isnt about the functionality.  Rather, its the nagging fact that CQ has too much coupling between the fluent api, and the nuts-and-bolts of doing the work.  

This contribution is a great example. Adding this code where it is proposed is the right answer given CQ's current architecture. However, it results in way too much code in the same (getting larger) files.

I dont know how to do it, but I believe that each operation should be its own object, and CQ should simply delegate to those operations. For example, cut, extrude, n-sided face, cbore, etc, shoudl all be seprate operations, which can be unit tested and contributed.

An cool advanced implementation would have all of the operations in a package, and automatically bind those by name into the cq namespace, so that they become available automatically when contributed.

I think we'll continue to see awesome contributions like this from users, and I think its important that they can be incorporated without churn on the core CQ classes.

bragostin, 2019-12-02:
@jmwright  @dcowden  yes, with 3D printing the use of organic like structures becomes more popular and the combination with all the great features of CadQuery can be powerful I think. The only limitation I see is that the generation of these organic structures with a CAD kernel is very resources demanding, I guess that's why usually softwares for 3D printing work directly on mesh structures like STL.

I can generate tests (any requirements for those?) and examples, I already notices some issues to fix by running a simple one.

jmwright, 2019-12-02:
@bragostin 

> I can generate tests (any requirements for those?)

We don't have any formalized requirements for tests. I normally just try to find something similar that already exists in the testing framework and adapt that. I'm not sure if you'll be able to find something similar or not. We can provide more direct guidance as you go if you want.

I do try to cover the major variations of the ways the operation can be called, varying the function parameters accordingly. It's really just a judgement call because it's hard to predict every failure mode.

> yes, with 3D printing the use of organic like structures becomes more popular and the combination with all the great features of CadQuery can be powerful I think.

Agreed

> The only limitation I see is that the generation of these organic structures with a CAD kernel is very resources demanding,

With CadQuery the model generation can be offloaded onto another machine (server) that has more resources. We haven't focused much on resource intensive models, but I think there's some room for experimentation and growth in that direction. I experimented a little bit with OpenFaas to create a function that would run on a server that could be optimized for model generation. The script text was sent from a low powered laptop to the server, and the result was returned. I got a proof-of-concept working, but never did any more with it. I could see some sort of architecture like that being useful in the future.

adam-urbanczyk, 2019-12-02:
@bragostin  that is cool! Take a look here for a related issue https://github.com/CadQuery/cadquery/issues/93 . I am not yet sure which of the multiple available OCCT functions (`BRepOffsetAPI_MakeFilling`, `GeomPlate_BuildPlateSurface` ? ) should we expose (first).

At the moment I'm busy with OCCT7.4 but I'll try your code soon.

bragostin, 2019-12-02:
@adam-urbanczyk  I tried at first with ```GeomPlate_BuildPlateSurface```, but apparently it is limited to 4 sided surfaces. OCCT offers a lot of different functions that seem to produce close results, and I am still trying to make sense of these.  ```BRepOffsetAPI_MakeFilling```  seems to be the most flexible one, with no limit on the number of sides (I needed 6 sides for the Gyroïd surface).

bragostin, 2019-12-03:
@dcowden yes, I can imagine that a system where users can simply add their contributions in a third-party folder and the CadQuery team can decide whether to make it part of the master release or not could be beneficial in the end and leverage the participative nature of CadQuery.

bragostin, 2019-12-04:
@jmwright here are some examples I generated to try to 'stress' the function, check if the optional arguments work, check if pushpoints works, if non co-planar edges works, and the last is a test in the sense that it is the only example I found in PythonOCC that uses ```BRepOffsetAPI_MakeFilling```. Would that be suitable?

[image]
[cq_BRepOffsetAPI_MakeFilling _examples.py.txt](https://github.com/CadQuery/cadquery/files/3923761/cq_BRepOffsetAPI_MakeFilling._examples.py.txt)

jmwright, 2019-12-04:
@bragostin Looks like a great variety. +1

adam-urbanczyk, 2019-12-06:
@bragostin do I understand correctly that only polygonal boundaries are supported? Any reason for not supporting other geometry types (e.g. splines)?

bragostin, 2019-12-06:
@adam-urbanczyk  That's right, the only reason is that I don't know how to do it otherwise.

Initially I wanted to use a wire created by CadQuery for the edges of the surface, but I ran into a problem I was not able to solve: ```BRepOffsetAPI_MakeFilling``` requires the surface boundaries coming from ```w = wire_builder.Wire()```  to be processed by ```edges = [i for i in TopologyExplorer(w).edges()]``` before building the surface.

The wires produced by CadQuery are not accepted by ```TopologyExplorer(w)```.  

But if you can explain me how to "unwrap" them so that they can be processed, I think I could improve the implementation so that CadQuery splines can be used instead of polygons. That would surely improve the final result quality.

bragostin, 2019-12-06:
@adam-urbanczyk one additional challenge also is that the surface edges may not be co-planar, like in the gyroïd example, which makes it difficult to use splines, unless we can stitch splines contained in different planes together to create a closed boundary.

jmwright, 2019-12-06:
@bragostin Some combination of `val().wrapped` or just `.wrapped` should extract the PythonOCC object, depending on what level you're at in the abstraction. The form is shown at the end of [this line](https://github.com/CadQuery/cadquery/blob/f9d9ae06406aac2feea4a2a94569de76d473a8d0/tests/TestCadQuery.py#L1333) for a face, but should work the same for a wire.

bragostin, 2019-12-07:
@jmwright  with this line ```wireSets = surf_edges.wire().ctx.pendingWires[0].wrapped``` now ```TopologyExplorer(w)``` accepts my wire, thank you!
Now I am faced with another problem:
``` python
import cadquery as cq
edge_wire = cq.Workplane('XY').polyline([(-7.,-7.), (7.,-7.), (7.,7.), (-7.,7.)])
edge_wire = edge_wire.add(cq.Workplane('YZ').workplane().transformed(offset=cq.Vector(0, 0, -7), rotate=cq.Vector(0, 45, 0)).polyline([(-7.,0.), (3,-3), (7.,0.)]))
show_object(edge_wire)
```
creates the polyline with one point NOT coplanar with the others, shown below
[image]

However it is not closed, and when I do 
``` python
edge_wire = edge_wire.close()
```
I get the closed line below
[image]

where the off plane point is not included.

Any idea how to create a closed wire that includes the off plane point? It's one of the advantages of ```BRepOffsetAPI_MakeFilling``` that it can create a surface from edges that are not in the same plane, but it requires a closed wire.

bragostin, 2019-12-08:
I have found the following solution. By adding these lines at the beginning of my ```interpPlate()``` function in ```cq.py```
```python
        surf_edges = [o.vals()[0] for o in surf_edges.all()]
        surf_edges = Wire.assembleEdges(surf_edges)
        wireSets = surf_edges.wrapped
```
this example
```python
import cadquery as cq
thickness = 0.1
edge_points = [[-7.,-7.,0.], [-3.,-10.,3.], [7.,-7.,0.], [7.,7.,0.], [-7.,7.,0.]]
edge_wire = cq.Workplane('XY').polyline([(-7.,-7.), (7.,-7.), (7.,7.), (-7.,7.)])
edge_wire = edge_wire.add(cq.Workplane('YZ').workplane().transformed(offset=cq.Vector(0, 0, -7),\
 rotate=cq.Vector(0, 45, 0)).spline([(-7.,0.), (3,-3), (7.,0.)]))
surface_points = [[-3.,-3.,-3.], [3.,3.,3.]]
plate_1 = cq.Workplane('XY').interpPlate(surface_points, edge_wire, thickness)
show_object(plate_1)
```
now works and produces the surface below
[image]
so it is possible to have polygons and splines together!
Do you think it would make sense to still leave the option to define the surface edges simply as a list of points used to create a polygon by pythonOCC like in the first version? For mathematically complex shapes like these gyroids it might still make sense.

jmwright, 2019-12-11:
> Do you think it would make sense to still leave the option to define the surface edges simply as a list of points used to create a polygon by pythonOCC like in the first version? For mathematically complex shapes like these gyroids it might still make sense.

I think that if you can leave the list-of-points option without complicating the interface too much, it would be a win.

bragostin, 2019-12-11:
@jmwright maybe a simple test, if the input is a list then make a polygon, else expect a cq object?

I am completely stuck on this now:

```python
import cadquery as cq

wire_list =  [[[3.54, 3.54], [1.77, 0.0], [3.54, -3.54]], [[-3.54, -3.54], [0.0, -1.77], [3.54, -3.54]], [[-3.54, -3.54], [0.0, -1.77], [3.54, -3.54]], [[-3.54, -3.54], [-1.77, 0.0], [-3.54, 3.54]], [[3.54, 3.54], [0.0, 1.77], [-3.54, 3.54]], [[3.54, 3.54], [0.0, 1.77], [-3.54, 3.54]]]
plane_list =  ['XZ', 'XY', 'YZ', 'XZ', 'YZ', 'XY']
offset_list =  [-3.5355339059327373, 3.5355339059327373, 3.5355339059327373, 3.5355339059327373, -3.5355339059327373, -3.5355339059327373]

edge_wire = cq.Workplane(plane_list[0]).workplane(offset=-offset_list[0]).spline(wire_list[0])
for i in range(len(wire_list)-1):
    edge_wire = edge_wire.add(cq.Workplane(plane_list[i+1]).workplane(offset=-offset_list[i+1]).spline(wire_list[i+1]))
 
edge_wire = [o.vals()[0] for o in edge_wire.all()]
edge_wire = cq.Wire.assembleEdges(edge_wire)
edge_wire = edge_wire.wrapped
```
returns the error
```
Traceback (most recent call last):
  File "issue.py", line 115, in <module>
    edge_wire = cq.Wire.assembleEdges(edge_wire)
  File "/usr/local/lib/python3.7/dist-packages/cadquery-0.5_SNAPSHOT-py3.7.egg/cadquery/occ_impl/shapes.py", line 828, in assembleEdges
    return cls(wire_builder.Wire())
RuntimeError: StdFail_NotDone
BRep_API: command not done
wrapper details:
  * symname: BRepBuilderAPI_MakeWire_Wire
  * wrapname: _wrap_BRepBuilderAPI_MakeWire_Wire
  * fulldecl: TopoDS_Wire const BRepBuilderAPI_MakeWire::Wire()
```
yet the wire looks closed in CQ-editor:
[image]

In other cases it works yet. Any idea what is the problem here?

jmwright, 2019-12-11:
> BRep_API: command not done

I feel like when I normally see that it's a CAD kernel error. Do you have a redundant vertex/endpoint or anything in what you're passing to `assembleEdges`?

bragostin, 2019-12-11:
@jmwright indeed there is a redundant point, to make sure that the loop is closed, otherwise ```BRepOffsetAPI_MakeFilling``` complains that the wire is not closed.

bragostin, 2019-12-12:
@jmwright I have found the problem: the points in wire_list are not in consecutive order because they were generated by another parametric function that cannot ensure this. Therefore I used the possibility that ```BRepBuilderAPI_MakeWire::Add(const TopTools_ListOfShape & L)``` offers to provide a list of shapes directly:
```Adds the edges of <L> to the current wire. The edges are not to be consecutive. But they are to be all connected geometrically or topologically. If some of them are not connected the Status give DisconnectedWire but the "Maker" is Done() and you can get the partial result. (ie connected to the first edgeof the list <L>)```
Following this I modified the assembledEdges in shapes.py like this
```python
   def assembleEdges(cls, listOfEdges):
        """
            Attempts to build a wire that consists of the edges in the provided list
            :param cls:
            :param listOfEdges: a list of Edge objects
            :return: a wire with the edges assembled
        """
        wire_builder = BRepBuilderAPI_MakeWire()
        
        # begin new
        edges_list = TopTools_ListOfShape()
        for e in listOfEdges:
            edges_list.Append(e.wrapped) 
        wire_builder.Add(edges_list)
        # end new
        
        # begin old
        #for edge in listOfEdges:
        #    wire_builder.Add(edge.wrapped)
        # end old
        
        return cls(wire_builder.Wire())
```
and now it works, I don't have to provide the edges in consecutive order, which makes parametric shape generation considerably easier in my opinion.
Do you think this could be made a permanent change of the ```assembleEdge``` function?

jmwright, 2019-12-12:
This might solve one or two other open issues as well. If you run the runtests.py script, do all the tests pass? @adam-urbanczyk What are your thoughts?

bragostin, 2019-12-12:
@jmwright looks like, output of ```python runtests.py``` is: 
```
Ran 178 tests in 13.304s
OK
```

jmwright, 2019-12-12:
@bragostin That's good news. I'm wondering if this will help with issues like https://github.com/CadQuery/cadquery/issues/125

adam-urbanczyk, 2019-12-12:
@bragostin nice catch! Your code looks much cleaner. Could you make a separate PR with just the `assembleEdges` change? I still need to dive into the parametric surface stuff.

bragostin, 2019-12-12:
@adam-urbanczyk  I will make the ```assembleEdges``` PR tomorrow. Regarding the parametric surface code I have just implemented the use of splines as you suggested, so I will post the new code soon.

bragostin, 2019-12-20:
I made some progress on the ```BRepOffsetAPI_MakeFilling``` front:
-  now splines and polylines are supported
- a simple list of points can still be supplied (more convenient sometimes, for example Shwartz D surfaces have 6 straight borders in 6 different planes; in such case the boundary can be defined simply by providing 6 points)
- 0 thickness is enabled to produce 2D surfaces

The new functions are the following. In cq.py in ```class Workplane(CQ)```:
```python
    def interpPlate(self, surf_edges, surf_pts=[], thickness=0, combine=True, clean=True, Degree=3, NbPtsOnCur=15, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        """
        Returns a plate surface that is 'thickness' thick, enclosed by 'surf_edge_pts' points,  and going through 'surf_pts' points.
        
        :param surf_edges
        :type 1 surf_edges: list of [x,y,z] float ordered coordinates
        :type 2 surf_edges: list of unordered CadQuery wires
        :param surf_pts = [] (uses only edges if [])
        :type surf_pts: list of [x,y,z] float coordinates
        :param thickness = 0 (returns 2D surface if 0)
        :type thickness: float >= 0
        :param combine: should the results be combined with other solids on the stack
            (and each other)?
        :type combine: true to combine shapes, false otherwise.
        :param boolean clean: call :py:meth:`clean` afterwards to have a clean shape      
        :param Degree = 3 (OCCT default)
        :type Degree: Integer >= 2
        :param NbPtsOnCur = 15 (OCCT default)
        :type: NbPtsOnCur Integer >= 15
        :param NbIter = 2 (OCCT default)
        :type: NbIterInteger >= 2
        :param Anisotropie = False (OCCT default)
        :type Anisotropie: Boolean
        :param: Tol2d = 0.00001 (OCCT default)
        :type Tol2d: float > 0
        :param Tol3d = 0.0001 (OCCT default)
        :type Tol3dReal: float > 0
        :param TolAng = 0.01 (OCCT default)
        :type TolAngReal: float > 0
        :param TolCurv = 0.1 (OCCT default)
        :type TolCurvReal: float > 0
        :param MaxDeg = 8 (OCCT default)
        :type MaxDegInteger: Integer >= 2 (?)
        :param MaxSegments = 9 (OCCT default)
        :type MaxSegments: Integer >= 2 (?)
        """
        
        # If thickness is 0, only a 2D surface will be returned.
        if thickness==0:
            combine=False
        
        # If a list of wires is provided, make a closed wire
        if not isinstance(surf_edges, list): 
            surf_edges = [o.vals()[0] for o in surf_edges.all()]
            surf_edges = Wire.assembleEdges(surf_edges)
            surf_edges = surf_edges.wrapped
        
        # Creates interpolated plate
        def _makeplate(pnt):
            return Solid.interpPlate(surf_edges, surf_pts, thickness, Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments)

        plates = self.eachpoint(_makeplate, True)

        # if combination is not desired, just return the created boxes
        if not combine:
            return plates
        else:
            # combine everything --> CRASHES, probably OCC 6 CAD kernel issue, better use pushpoints through each() instead
            return self.union(plates, clean=clean)
```
In shapes.py in ```class Solid(Shape, Mixin3D)```:
```python
 def interpPlate(cls, surf_edges, surf_pts, thickness, Degree=3, NbPtsOnCur=15, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        
        # POINTS CONSTRAINTS: List of (x,y,z) points provided
        pts_array = [gp_Pnt(*pt) for pt in surf_pts]
        
        # EDGE CONSTRAINTS: build closed polygon
        if isinstance(surf_edges, list): # List of (x,y,z) points provided
            e_array = [Vector(*e) for e in surf_edges]
            wire_builder = BRepBuilderAPI_MakePolygon()
            for e in e_array: # Create polygon from edges
                wire_builder.Add(e.toPnt())
            wire_builder.Close()
            w = wire_builder.Wire()
        else: # Closed wires provided
            w = surf_edges
        edges = [i for i in TopologyExplorer(w).edges()]
        
        # MAKE SURFACE
        continuity = GeomAbs_C0 # Fixed, changing to anything else crashes.
        face = Face.makeNSidedSurface(edges, pts_array, continuity, Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments) 
        
        if thickness>0: # Thicken surface
            solid = BRepOffset_MakeOffset()
            solid.Initialize(face.wrapped, thickness, 1.e-5, BRepOffset_Skin, False, False, GeomAbs_Intersection, True) #The last True is important to make solid
            solid.MakeOffsetShape()
            return cls(solid.Shape())
        else: # Return 2D surface only
            return face
```
And in shapes.py in ```class Face(Shape)```:
```python
def makeNSidedSurface(cls, edges, points, continuity=GeomAbs_C0, Degree=3, NbPtsOnCur=15, NbIter=2, Anisotropie=False, Tol2d=0.00001, Tol3d=0.0001, TolAng=0.01, TolCurv=0.1, MaxDeg=8, MaxSegments=9):
        
        n_sided = BRepOffsetAPI_MakeFilling(Degree, NbPtsOnCur, NbIter, Anisotropie, Tol2d, Tol3d, TolAng, TolCurv, MaxDeg, MaxSegments)
        for edg in edges:
            n_sided.Add(edg, continuity)
        for pt in points:
            n_sided.Add(pt)
        n_sided.Build()
        face = n_sided.Shape()
        return cls.cast(face).fix()
```
And finally a file with a few examples in attachment
[cq_InterpPlate_simple_examples.py.txt](https://github.com/CadQuery/cadquery/files/3988701/cq_InterpPlate_simple_examples.py.txt)
One thing still does not work: ```pushpoints``` makes the function crash when used directly with the new ```interpPlate``` function. I believe it is a CAD kernel issue because it comes from ```union``` when combine=True. But it works fine when ```pushpoints``` is used in conjunction with ```each```.

jmwright, 2019-12-20:
@bragostin This is awesome, thanks.

> One thing still does not work: pushpoints makes the function crash when used directly with the new interpPlate function. I believe it is a CAD kernel issue because it comes from union when combine=True.

It very well could be. @adam-urbanczyk has been working on getting us updated to OCC 7.3, which will hopefully help address crashes like this when that's ready.

bragostin, 2019-12-21:
@jmwright shall I do a pull request with this then?

jmwright, 2019-12-21:
Yes, please.

bragostin, 2019-12-27:
I close this issue since there is now a pull request "Implementation of BRepOffsetAPI_MakeFilling" #253

closed 2019-12-27 by bragostin