CadQuery/cadquery #230: Tutorials

jmwright, 2019-12-07:
I've started experimenting with creating tutorials using [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery). Thanks to @bernhard-42 the tutorials can be run on MyBinder.org without installing anything on your local computer. I'd like to get some feedback from the community before I sink too much time into creating more. Follow the instructions in the [readme here](https://github.com/CadQuery/cadquery-contrib/tree/master/tutorials) to get started and please let me know what you think.

geobruce, 2019-12-10:
I would be interested in some more tutorials.

jmwright, 2019-12-10:
@geobruce Thanks.

The [Google Group](https://groups.google.com/forum/m/#!forum/cadquery) is probably the best place to ask for examples, but if you can't access it (or just don't want to be on Google Groups), you can just post your question as a GitHub issue on this repo.

jmwright, 2019-12-10:
@geobruce I missed that you had requested access to the Google Group separately. You're approved now so you can post.

geobruce, 2019-12-11:
> @geobruce You're approved now so you can post.
Thank you very much!
this project looks promising, only I need to get some more understanding of the general workflow. 

You can find my question in the google group [here](https://groups.google.com/forum/#!topic/cadquery/yZF4eIfp_JU) 

Thanks!

closed 2019-12-16 by jmwright