CadQuery/cadquery #231: Example center() is not working

geobruce, 2019-12-10:
The example for the [center() command](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.center) is not working

```
#this workplane is centered at x=0.5,y=0.5, the center of the upper face
s = Workplane().box(1,1,1).faces(">Z").workplane()

s.center(-0.5,-0.5) # move the center to the corner
t = s.circle(0.25).extrude(0.2)
assert ( t.faces().size() == 9 ) # a cube with a cylindrical nub at the top right corner
```

Workplane() has no argument
and when I enter "XY" the example is still not showing the cylindrical nub at the corner.

jmwright, 2019-12-10:
@geobruce The `center` call returns a copy of the workplane, it doesn't edit it in place (notice the `s =` I added). The following code works fine for me in CQ-editor. You might want to have a look at the [Lego example](https://github.com/CadQuery/cadquery/blob/master/examples/Ex100_Lego_Brick.py), since it's similar to what you're doing in this code. There are certain situations where the normal of a workplane can get flipped so that an extrude operation goes down into the other object rather than up, making it look like the extrude operation had no effect. In that case you can pass `invert=true` to `workplane()`.
```python
import cadquery as cq

#this workplane is centered at x=0.5,y=0.5, the center of the upper face
s = cq.Workplane("XY").box(1,1,1).faces(">Z").workplane()

s = s.center(-0.5,-0.5) # move the center to the corner
t = s.circle(0.25).extrude(0.2)
#assert ( t.faces().size() == 9 ) # a cube with a cylindrical nub at the top right corner
show_object(t)
```
[image]

geobruce, 2019-12-11:
> @geobruce  (notice the `s =` I added). 

thanks, indeed this solves the problem. Does this mean the [example code](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.center) is wrong? Or that it needs to be interpreted for your specific application?

jmwright, 2019-12-11:
Ah, I had not noticed that before. Yes, it appears that the example code is wrong. The docs will need to be updated. Thanks for pointing that out.

jmwright, 2019-12-11:
The documentation has been fixed in #234

closed 2019-12-11 by jmwright