CadQuery/cadquery #232: Extracting edge / wire properties

osterwood, 2019-12-11:
Is there a way to get all edge (or wire) properties from a occ_impl.shapes.Edge?  

I see that I can get startPoint and endPoint, but is there a way to get the additional information needed to reconstruct circles and arcs? (like center, radius, start angle, end angle).  Is there a way to query if an edge is a straight line or an arc?

jmwright, 2019-12-11:
@adam-urbanczyk and @dcowden understand PythonOCC better than I do, but I'll offer up what I can. You can get access to the lower level PythonOCC object, [BRepAdaptor_Curve](https://www.opencascade.com/doc/occt-7.1.0/refman/html/class_b_rep_adaptor___curve.html), which will give you access to more properties. [Here's](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/shapes.py#L644) an example of where it's used in the source (follow _geomAdaptor down through the next few methods).

adam-urbanczyk, 2019-12-11:
@osterwood is the underlying geometry a circle?

osterwood, 2019-12-11:
Sometimes circles and sometimes arcs (which I guess are arc per https://www.opencascade.com/doc/occt-7.1.0/refman/html/_geom_abs___curve_type_8hxx.html#af25c179d5cabd33fddebe5a0dc96971c)

osterwood, 2019-12-11:
Looks like I can use `_geomAdaptor().GetType()` to determine if something is a circle (0 -> line, 1 -> circle), then:

```
circle = obj._geomAdaptor().Circle()
r = circle.Radius()
x,y,z = [circle.Location().Coord(i) for i in [1,2,3]]
```

And entry & exit angles can be determined by the relationship between the center (x,y,z) and the `startPoint()` & `endPoint()`.

adam-urbanczyk, 2019-12-11:
@osterwood indeed at the "geometry" level (as opposed to "topology" level) you should always get a `Geom_Circle`. You already figured out how to get the relevant values.

osterwood, 2019-12-11:
Thanks for the pointers here!

closed 2019-12-11 by osterwood