CadQuery/cadquery #240: Union doesn't always union

osterwood, 2019-12-13:
I'm not quite sure what's going on here, but code and image should be illustrative.

```
outside = 26
inside = 23
boss_distance = 20
edge_radius = (outside - bolt_distance)/2.0
height = 2.5

bosses = cq.Workplane("XY") \
    .rect(boss_distance, boss_distance, forConstruction=True) \
    .vertices().circle(edge_radius-0.1).extrude(height)
        
rib = cq.Workplane("XY").rect(outside, outside) \
            .rect(inner, inner).extrude(height) \
            .edges("|Z and (<X or >X or <Y or >Y)").fillet(edge_radius)
        
rib = rib.union(bosses)
```

[image]

See how the diameter of the bosses is fully visible?  It has not been combined with the outside rectangle.

An exported STEP file from this code has just the bosses, no rib part at all.

I've tried a number of different ways of joining these objects and the results are always similar to this.  Any ideas?

jmwright, 2019-12-14:
@osterwood If you're using CQ-editor and don't use `show_object` to show the model, CQ-editor will display all the separate models automatically, making it look like they're not unioned. You can either hide the bosses by unchecking their entry in the Object tree on the right side, or you can add `show_object`.
```python
import cadquery as cq

outside = 26
inside = 23
boss_distance = 20
edge_radius = (outside - boss_distance)/2.0
height = 2.5

bosses = cq.Workplane("XY") \
    .rect(boss_distance, boss_distance, forConstruction=True) \
    .vertices().circle(edge_radius-0.1).extrude(height)
        
rib = cq.Workplane("XY").rect(outside, outside) \
            .rect(inside, inside).extrude(height) \
            .edges("|Z and (<X or >X or <Y or >Y)").fillet(edge_radius)
        
rib = rib.union(bosses)

rib.val().exportStep('/home/jwright/Downloads/test.step')

show_object(rib)
```
The `exportStep` line works as expected as well. The whole, unioned object is in place and doesn't have any seams.

[image]

osterwood, 2019-12-15:
I was using the Jupyter notebook interface, but tried CQ-Editor and I get the same result using your code.

I am using a 2.0.0dev checkout of CadQuery.  What version of CadQuery are you running above?

jmwright, 2019-12-15:
@osterwood I'm running almost the latest master via the CQ-editor Anaconda environment.

https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml#L19

adam-urbanczyk, 2019-12-15:
@osterwood  I also cannot reproduce this error. Could you try if it works on linux?

osterwood, 2019-12-15:
I created a new Conda env from the CQ-editor yml file and that works on both Linux and MacOS.

It must have been some strange version mismatch in my previous Conda environment.

I'm not able to run CQ-Editor on MacOS (I get a "QWidget::repaint: Recursive repaint detected" error), but will report that issue either there on with PythonOCC (where the error seems to actually be occurring).

Thanks for trying this code out and figuring out the issue was with my install.

jmwright, 2019-12-15:
@osterwood I would start by reporting it on the CQ-editor repo. I think it's pretty well established that the viewer in the version of PythonOCC we're using works properly, and we don't have easy access to MacOS machines to do testing. It's probably something with CQ-editor. If you post there, I'll try to borrow a Mac sometime next week to troubleshoot this.

closed 2019-12-16 by osterwood