CadQuery/cadquery #242: CQ at FOSDEM 2020 - please share your success stories

adam-urbanczyk, 2019-12-15:
We will have a talk at FOSDEM 2020 in the Open Source Computer Aided Modeling and Design devroom. It would be great to show a slide with interesting models made by the users. Please share them here (preferably on white background)!

Peque, 2019-12-15:
Toy project, but public... :wink:

I used it in a couple of pieces for a robot: [rim](https://bulebule.readthedocs.io/en/latest/building.html#rim), [mount](https://bulebule.readthedocs.io/en/latest/building.html#mounting).

jmwright, 2019-12-15:
@adam-urbanczyk Thanks for taking this on. I hope they take a video of the presentation. I'm looking forward to seeing it.

@Peque Very cool! Thanks for sharing.

jmwright, 2019-12-15:
[This project](https://opendesignengine.net/projects/yavin-thruster/wiki/Wiki) is from 2015, but I think it's kind of interesting. It used equations from a Jupyter notebook to drive the generation of a cold gas thruster nozzle using CadQuery (I coded the CadQuery model generation). The nozzle was then 3D printed for testing. A presentation was given in 2015 at the Open Hardware Summit, and I can check with the organization (Mach 30) to get a full resolution version of the diagram shown there if you want to include it.

michaelgale, 2019-12-16:
I put this together a few weeks ago for the community's benefit.  Feel free to pull whatever assets you wish:
https://github.com/fx-bricks/fx-cad-notes

dcowden, 2019-12-21:
@adam-urbanczyk thanks for participating in fosdem.

Doesn't Kicad  use Cadquery to build its model repo? I think it still uses cq 1.0

https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/QFN_packages

jmwright, 2019-12-21:
> Doesn't Kicad use Cadquery to build its model repo? I think it still uses cq 1.0

Yes. It is still based on CQ 1.0 AFAIK.

jmwright, 2019-12-21:
@easyw Can you comment on the above, please?

bragostin, 2019-12-26:
I could share this Gyroïd surface
[image]
but this is done with the Implementation of BRepOffsetAPI_MakeFilling #253 pull request, so that's not part of the master branch of CadQuery.

adam-urbanczyk, 2019-12-26:
Cool @bragostin , could you share what is your application of CQ?

bragostin, 2019-12-27:
@adam-urbanczyk these are cooling fins. Triply periodic surfaces are supposed to offer a better ratio of heat transfer over pressure drop.

adam-urbanczyk, 2020-01-27:
I added some screenshots to the slides - thanks for your help @bragostin  @michaelgale @Peque @jmwright !

closed 2020-01-27 by adam-urbanczyk